arendst/Tasmota #163: MQTT_CLIENT_ID

kiralikbeyin, 2017-03-07:
I tried %012X for full mac adress to MQTT_CLIENT_ID but it didnt work.
returned like 00000q1w2e3
How can i put full mac adress?

davidelang, 2017-03-07:
On Mon, 6 Mar 2017, kiralikbeyin wrote:

> I tried %012X for full mac adress to MQTT_CLIENT_ID but it didnt work.
> returned like 00000q1w2e3
> How can i put full mac adress?

remember that the first three bytes of the mac address are the vendor ID, you 
are _extremely_ unlikely to have a collision using just the last half (which is 
what the software does by default)

kiralikbeyin, 2017-03-07:
I have to subscribe mqtt like this:
"topic":"users/18fe34bdaabf/#"
And didn't understand anything from your reply sorry.

davidelang, 2017-03-07:
a MAC address is 12 hex digits, of those 12 digits, the first 6 indicate the manufacturer of the device (or of the chipset in the device).

These first 6 digits are going to be the same on every ESP8266 module ever shipped. This limits them to "only" being able to ship 16M devices before they have to come up with a second prefix.

to change what is shown, you would have to make changes in the code to provide the full mac address not just the last half of it, the variable you are using only has 12 hex digits of the MAC, which is why changing the field to 12 digits didn't help

Why do you think you need to subscribe to mqtt with the full mac address?

kiralikbeyin, 2017-03-07:
OK I understand now.

My node server is designed for full mac address like "topic":"users/18fe34bdaabf/#"

You mean first 6 "5ccf7f" is the same for all sonoffs?
I will try "5ccf7f%06X"

closed 2017-03-08 by arendst