arendst/Tasmota #164: how to get temperature from MQTT: /DHTTemperature 23.2

Rudoxxxx, 2017-03-07:
how to get temperature from MQTT:
tele/sonoff/SENSOR	{"Time":"2017-03-03T22:05:36", "DS18B20":{"Temperature":25.4}, "DHT":{"Temperature":23.2, "Humidity":55.5}}

Majerdomo.ru server have PHP and Java script, but how to do it with easy way?
/DS18B20Temperature 25.4
/DHTTemperature 23.2
/DHTHumidity 55.5

MacWyznawca, 2017-03-07:
try {
				data = JSON.parse(message);
			}
			catch (e) {
			  that.log("JSON problem");
			}
			that.temperature -49.9;
			if (data === null) {
				that.temperature = parseFloat(message);
			} else if (data.hasOwnProperty("DS18B20")) {
				that.temperature = parseFloat(data.DS18B20.Temperature);
			} else if (data.hasOwnProperty("DS18x20")) {
				that.temperature = parseFloat(data.DS18x20.Temperature);
			} else if (data.hasOwnProperty("DHT")) {
				that.temperature = parseFloat(data.DHT.Temperature);
			} else if (data.hasOwnProperty("DHT22")) {
				that.temperature = parseFloat(data.DHT22.Temperature);
			} else if (data.hasOwnProperty("AM2301")) {
				that.temperature = parseFloat(data.AM2301.Temperature);
			} else if (data.hasOwnProperty("DHT11")) {
				that.temperature = parseFloat(data.DHT11.Temperature);
			} else if (data.hasOwnProperty("HTU21")) {
				that.temperature = parseFloat(data.HTU21.Temperature);
			} else if (data.hasOwnProperty("BMP280")) {
				that.temperature = parseFloat(data.BMP280.Temperature);
			} else if (data.hasOwnProperty("BME280")) {
				that.temperature = parseFloat(data.BME280.Temperature);
			} else if (data.hasOwnProperty("BMP180")) {
				that.temperature = parseFloat(data.BMP180.Temperature);
			} else if (data.hasOwnProperty(that.sensorPropertyName)) {
				that.temperature = parseFloat(data[that.sensorPropertyName].Temperature);
			} else {return null}

Rudoxxxx, 2017-03-08:
Thanks MacWyznawca.

Can you show step by step how to do it?

MacWyznawca, 2017-03-08:
I use this script in HomeBridge plugin, but I don't wnow Web developing.

Rudoxxxx, 2017-03-12:
Hombridge is for Apple?
We usingAndroid on  Galaxy S6, S7 edge. Its compatible?

MacWyznawca, 2017-03-12:
iOS and watchOS only.

MacWyznawca, 2017-03-12:
But You cann use my code in Your solution ex in Web.

davidelang, 2017-03-18:
This is a question about configuring your home automation software, not the sonoff. I'm not sure what software you are using.

The sonoff is sending the data, you need to do some sort of configuration in the home automation software that's pulling the data from mqtt to extract the json fields you want and format them in the way you are looking for.

davidelang, 2017-03-27:
reopen this ticket if you still have questions about configuring your software, although such questions are probably better directed at your home automation software folks

closed 2017-03-27 by davidelang

canis-ferrin, 2017-07-19:
How to realize the json reading in FHEM using a sonoff with dht22 sensor?