arendst/Tasmota #165: DS18S20 Negative Temperatures

dc75, 2017-03-07:
Thanks for providing this software!  I've just started playing around with it & flashed a NodeMCU board, with everything working great, except one issue that I can't seem to figure out.

I have two DS18S20 temperature sensors connected - one is measuring room temperature & the other is measuring the temperature inside a freezer.  Both sensors gave a correct reading via the Sonoff-Tasmota webpage when reading above 0 deg. C, but when I placed one sensor in the freezer I could watch the temperature reading falling until it reached below freezing.  Below freezing, where I would expect to see negative numbers, I'm seeing extremely high numbers, in the neighborhood of 30,000.  The sensor measuring room temperature continues to provide a correct reading at all times.

I've probably overlooked something simple, so would anyone have any ideas?

Thanks!

arendst, 2017-03-07:
I suppose you enabled the #define USE_DS18x20 in user_config.h?

dc75, 2017-03-07:
Thanks for the quick response!

Yes, I did enable the #define USE_DS18x20 in user_config.h.  It will be later tonight, but I'll try out your suggestion & report back.

dc75, 2017-03-07:
I made the change from uint8_t to int8_t on line 110 in file xsns_ds18x20.ino & the results were that it changed the temperature reading, but it is still not correct.  Now, instead of temp readings in the area of 30,000 from my freezer sensor, I'm seeing -122 deg C.  Any other suggestions?

Thanks!

arendst, 2017-03-08:
In addition to the change you already made could you also try the change in file xsns_ds18x20.ino starting at line 125 from
```
      if (data[1] > 0x80) sign = -1; // App-Note fix possible sign error
```
into
```
      if (data[1] > 0x80) {
        data[0] = (~data[0]) +1;
        sign = -1;  // App-Note fix possible sign error
      }
```

dc75, 2017-03-08:
I made the suggested change at line 125 & that works perfectly!  Temperature readings are now hovering around -8 deg C. & match up perfectly with my reference thermometer.

Thanks Theo!!

arendst, 2017-03-08:
Thnx for testing. Will be implemented in version 4.0.2 soon to be released.

closed 2017-03-08 by arendst