arendst/Tasmota #166: Using Sonoff Basic for controll a lamp manually

goncalvesdiogo, 2017-03-07:
Hi all,
I'd like to apologise by my bad english in advance.

What I need to do is to put a sonoff basic module inside the switch box in the walls of my home.
I did test and the sizes fits perfectly but I was wondering how can I hit the switch button once it is inside the wall?? The answer I found is to use a momentary switch(that is like a bell switch) to trigger the relay by soldering 2 wires on the sonoff button and so create a second way to control.

My question is,  If I decide to use the GPIO14 instead to solder the 2 wires, where else I need to plug the other wire? I mean should be 1 wire coming out the 3.3vcc and going to the momentary switch and another one from momentary switch to the GPIO14. Does not need to put some resistors?

Another question, the switch mode I need to use, may I keep it as TOGGLE?

I hope you guys understand what I am trying to do. I understand that the issues here should be about the TASMOTA firmware but I don't know where else I could get help.
Tks in advance.

arendst, 2017-03-07:
One wire to GPIO14 the other to Gnd

goncalvesdiogo, 2017-03-07:
Thank you so much Theo. Regards.

davidelang, 2017-03-16:
did your question get fully answered? if so, can this issue be closed?

goncalvesdiogo, 2017-03-16:
Yes the issue can be closed please. I'm newbie here. How can I do that?

closed 2017-03-16 by goncalvesdiogo