arendst/Tasmota #168: DS18B20 doesn't work on GPIO3

KonstantinMastak, 2017-03-07:
On most Sonoff boards it is possoble to solder to GPIO3 as it RX and it usually has wires on board. So I tried to plug DS18B20 to GPIO3 in Sonoff Basic and Sonoff S20, but it doesn't work - "Sensor busy" error in logs. However, DS18B20 on GPIO4 or GPIO14 works perfectly. 

I looked through oscillograph and I see that GPIO3 is always low while it gets high for 1 or 2 seconds when ESP tries to poll DS18B20 . However, as far as I understand, it should be in reverse - always high while low when polling.

I guessed that maybe it interferes with Serial output feature, but I set "Serial log level" to "None" and this doesn't help.

What may be the reason of problem?

arendst, 2017-03-07:
There is a reason I do not provide gpio3 as an user option...

With the ds18b20 it probably fails as the port init is overwritten by serial out. disabling serial happens to late after initialisation code.

KonstantinMastak, 2017-03-08:
I there any easy way to completely disable serial output?

KonstantinMastak, 2017-03-08:
Update: Actually, you DO provide gpio3 as an user option:
{ "Sonoff Basic",    // Sonoff Basic
     GPIO_KEY1,        // GPIO00 Button
     0, 0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 Optional sensor
     0, 0, 0, 0, 0, 0, 0,
     GPIO_REL1,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Green Led (0 = On, 1 = Off)
     GPIO_USER,        // GPIO14 Optional sensor
     0, 0
  }

And this is a good idea as some Sonoff modules make GPIO3 easily accessible for soldering, while GPIO4 and GPIO14 are not availabe at all. 

In order to make GPIO3 work with DS18B20 I tried to comment out everything connected with serial connection, both initialization and "serial" class usage everywhere. Unfortubately, this doesn't help, I still get "DSB: Sensor busy" in web log. 

Do you have any idea what else can prevent GPIO3 working with DS18B20?

arendst, 2017-03-08:
Stupid of me to provide access to GPIO03 ;-)

It works for me if I disable serial with ``seriallog 0`` and connecting the DS18B20 to Gnd, GPIO03 and 3V3 all available on the pin header of a sonoff basic. It also needs a resistor between 3V3 and GPIO03.

KonstantinMastak, 2017-03-08:
Resistor! This maybe the reason - I didn't put it because I thought DS18B20  is in active power mode, not in parasitic power. I will check it up tomorrow, thank you!

closed 2017-03-12 by arendst