arendst/Tasmota #169: Flashing firmware binary directly

ThomDietrich, 2017-03-08:
Did anyone successfully flash a fresh Sonoff module with the Tasmota firmware binary directly?

I've no factory firmware Itead Sonoff module left in my collection, so can't test myself. I'd like to add this option to the wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Upload because I recently learned that flashing a microcontroller using the Arduino IDE or PlatformIO is not as easy for non-programmers. I'd like to give easier instructions for these end users.
Which Software did you use, which pitfalls are there to look out for?

arendst, 2017-03-08:
I just did it like this on windows connecting a device to COM5:

1. Download the esptool.py program from https://github.com/espressif/esptool
2. Put device in firmware upload mode and clear flash with command ``esptool.py --port COM5 erase_flash``
3. Get the firmware file ``sonoff.ino.bin``
4. Flash a **Sonoff Touch** or **Sonoff 4CH** with command ``esptool.py --port COM5 write_flash -fm dout -fs 1MB 0x0 sonoff.ino.bin``
5. Flash all other Sonoffs with command ``esptool.py --port COM5 write_flash -fm dio -fs 1MB 0x0 sonoff.ino.bin``

ThomDietrich, 2017-03-08:
Thanks for your input! Looks like a good start.
All steps seem easy to further automate. I'm fluent in python but down to zero in time :sweat_smile: I'll leave this issue open for whoever might be interested. Maybe I'll code something the next time I'm in touch with some modules.

> You need to make sure you know the difference between Sonoff Touch/4CH and other Sonoffs

My guess is the flash will simply fail otherwise. One could hence first try dio, then dout!?

gajotnt, 2017-03-08:
Download the Bin file and flash using the esp flash tool that can be download from here:
https://github.com/letscontrolit/ESPEasy-platformio/releases/tag/v2.0.0-dev3

ThomDietrich, 2017-03-08:
I doubt this will work... https://github.com/BattloXX/ESPEasyFlasher/blob/master/FlashESP8266/Form1.cs#L57

gajotnt, 2017-03-08:
Used it last night, first flashed a 1024 all zero bin file to wipe the rom and then flash the tasmota bin. worked fine here :)

yann25, 2017-03-09:
I will give it a look even though I'm not fluent in python, but I have a lot of time :smile:

yann25, 2017-03-12:
I have written some code:

https://github.com/yann25/sonoff_autoflash 

It worked fine on Windows, I will try it on Raspbian tomorrow. Since I'm not a professional programmer comments are welcome. :)

ThomDietrich, 2017-03-13:
Nice one @yann25 ! I'll leave a few ideas over in your issue tracker ;)

dony71, 2017-05-22:
@gajotnt  What's the full command to do this with esptool?
"Used it last night, first flashed a 1024 all zero bin file to wipe the rom and then flash the tasmota bin. worked fine here :)"

gajotnt, 2017-05-24:
Hello there
here you go
https://drive.google.com/file/d/0BwVQa3PrL2G3aUF6NldsSV9QZzQ/view?usp=sharing

esptool, user GUI and the all zero bin file

gajotnt, 2017-05-24:
https://drive.google.com/open?id=0BwVQa3PrL2G3aUF6NldsSV9QZzQ

And in case you are worried about virus here you go:
`esptool.exe -vv -cd nodemcu -cb 115200 -cp COM%comport% -ca 0x00000 -cf reset.bin`

Where %comport% is the COM port of the programer :)

You just need to extract the reset.bin file from the above zip file.
Also the zip has the programming tool and a GUI for easier flashing.

ThomDietrich, 2017-05-24:
This might be a nice alternative: https://github.com/marcelstoer/nodemcu-pyflasher

jetema, 2017-08-02:
As well you could use a Raspberry to upload and flash a bin file to the sonoff, have a look at this link  
http://www.sigmdel.ca/michel/ha/sonoff/flashing_sonoff_en.html
I use this method with all my sonoff  ;-)

bionaught, 2018-04-10:
Managed to flash, but no temporary Access point popped up as I was expecting.
I got a serial connection 115200 and could see what is pasted below.
It say that it tries to connect to the two default APs, then goes on to WPS after not getting connected.
The board cycles power right away after saying WPS and the whole cycle repeats?
I've tried reflashing to no avail.

I've only tried by powering through the serial connection via PC & via a known 2 A USB charger, so I don't think it is because I have a brown out problem!?

I'd prefer not to try messing around on 240Vac,if it's not necessary. 

Any pointers will be appreciated. 

Graham

00:00:14 WIF: WPSConfigactive for 3 minutes
00:00:33 APP: Restarting
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld
   �
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_9AF46D, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-5229...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-5229...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 3 minutes
00:00:33 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld
   �
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_9AF46D, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-5229...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-5229...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 3 minutes

jboothe, 2019-01-10:
FYI...
I was not able to Flash Sonoff Basic with `-fm dio`. I had to use `-fm dout` instead.