arendst/Tasmota #170: ESP32

ionciubotaru, 2017-03-08:
Please consider to make the necessary changes to run Tasmota on ESP32.
When i tried I got a strange error:

```
sketch\support.h:8:28: fatal error: user_interface.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.

```

arendst, 2017-03-08:
Supply me with a sonoff using esp32 and I will have a look into it...

Btw I do not support your apple either

ionciubotaru, 2017-03-08:
Please send me your address and I will send you a new ESP32 - please send a private message to ionciubotaru@gmail.com

davidelang, 2017-03-09:
tl;dr summary, pull requests are king

@ionciubotaru I think that arendst is saying that he isn't going to bother with this until and unless there is a device that contains an ESP32. His focus is on sonoff devices, not general device control. When small additions can support other devices, he's willing to add some changes to support them (although in most cases, he's mostly accepting changes proposed by others rather than doing the development work himself)

so I don't think that shipping an ESP32 dev board to him will help (but if I'm wrong, I'll happily pay for one :-)

But if we can get things working on an ESP32, and the changes aren't that large, I think the odds are good that he'll accept the changes.

I have a few kickstarter power strips that contain horrific cloud-only-unencrypted controllers that I plan to replace the controllers on. I'm probably going to have to use an ESP32 as I don't think there are enough pins on the ESP8266, but I'd love to have it based on tasmota rather than having to duplicate the work.

ionciubotaru, 2017-03-09:
As you can see people love Tasmota.
Please add a donation mark on Tasmota's main page

arendst, 2017-03-10:
Had a look at the Esp32-Arduino package and must say that it could take a while to get Sonoff-Tasmota working on an ESp32; many libraries used by Tasmota like HTTPClient, OTAupgrade and Webserver have not been implemented yet.

ionciubotaru, 2017-03-10:
Of course, it's better to wait until all libraries are available.

I think that HTTPClient is here: [https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino)

Web server is here: [https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino)

and OTA has status To be implemented [https://github.com/espressif/arduino-esp32/issues/125](https://github.com/espressif/arduino-esp32/issues/125)

ionciubotaru, 2017-03-17:
OTA is now available [https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino)

arendst, 2017-03-17:
I received my esp32 (thnx) and were able to play with it using both Arduino IDE and PlatformIo. I must say it's a totally different ballpark with a lot of memory, features and new concepts (like flash partitioning). From what I see it needs a total rethinking of my current Sonoff-Tasmota.

Do not expect a Sonoff-tasmota-Esp32 any time soon....

davidelang, 2017-03-17:
On Fri, 17 Mar 2017, Theo Arends wrote:

> I received my esp32 (thnx) and were able to play with it using both Arduino 
> IDE and PlatformIo. I must say it's a totally different ballpark with a lot of 
> memory, features and new concepts (like flash partitioning). From what I see 
> it needs a total rethinking of my current Sonoff-Tasmota.

While it adds a lot of additional capability, could it be treated as just a 
slightly expanded ESP-8266 (just more pins available)?

Yes it wastes a lot of the capabilities of the esp32, but being able to run a 
feature-rich, known firmware like sonoff-tasmota is a huge start.

arendst, 2017-03-17:
@davidelang yes it is feature rich but the main problem is that many Arduino-esp8266 libraries have not been converted yet to the new esp32 environment; they all need (major) rewrites and from what I see there are not that many resources working on a quick release of Arduino-esp32.

davidelang, 2017-03-17:
On Fri, 17 Mar 2017, Theo Arends wrote:

> @davidelang yes it is feature rich but the main problem is that many 
> Arduino-esp8266 libraries have not been converted yet to the new esp32 
> environment; they all need (major) rewrites and from what I see there are not 
> that many resources working on a quick release of Arduino-esp32.

That's fair, if we just need to watch the list of libraries and wait for them to 
get checked off one-by-one, we can wait :-)

I suspect that for a lot of them, the esp8266 changes (which go from the 
arduino, "only one possibility", to "that can run on any pin and there can be 
more than one" will address many of the changes.

ionciubotaru, 2017-03-17:
Good news. Please take your time and develop Sonoff-tasmota-Esp32  when you feel comfortable.
The development board is a gift, you have no obligation.

davidelang, 2017-04-06:
reopening to keep tracking the packages

JacekDob, 2018-02-06:
My first approach of porting to ESP32 for Wemos32:
[Sonoff-Tasmota](https://github.com/JacekDob/Sonoff-Tasmota)

ionciubotaru, 2018-02-06:
great news @JacekDob 
As long as esp32 is for hardware integrators and is not present inside final products, I suggest to port the core version of Tasmota with wifi, configuration, web server, eeprom and some important sensors like ds18x, dht ....
Use only one standard template with all pins available.
After production release each developer will add it's own code and hopefully will contribute to your Tasmota's fork.

JacekDob, 2018-02-06:
This is exactly what I focused on:
# Working
- Wifi
- EEPROM/Flash (by NVS)
- Relays
- Webserver
- MQTT

Prepared for Wemos ESP32 (all pins configurable).

JacekDob, 2018-02-08:
Update:

## Working

- Wifi
- Flash (by NVS)
- Relays
- Webserver
- MQTT
- MDNS (Discovery)
- WS2812

## Not ported

- USE_MHZ19
- USE_ENERGY_SENSOR
- USE_SENSEAIR
- USE_ADC_VCC
- USE_ARILUX_RF
- USE_IR_REMOTE
- SNFBRIDGE

ionciubotaru, 2018-02-08:
great, please activate issues on your fork to communicate directly

JacekDob, 2018-02-08:
Activated

feel welcome to contribute:
https://github.com/JacekDob/Sonoff-Tasmota

mcsSolutions, 2018-03-06:
I captured this ESP32 fork for compilation using Atom/PlatformIO.  The platformio.ini contains only ESP8266 environments.  Much conditional compilation based upon ESP32 and ESP8266 definitions.  Without any edits several compile errors generated which I suspect are related to getting the environment setup.  Is there a platformio.ini for Wemos ESP32?  How does the ESP32 define get set?

JacekDob, 2018-03-06:
Tasmota for ESP32 currently compiles in Arduino IDE.

To work with PlatformIO needs adaptation. I guess for somebody with experience in it should take an hour.  Waiting for contribution.

PS.
Please raise issues in ESP32 Tasmota fork.

ascillato2, 2018-08-18:
Link to Tasmota ESP32 fork already in Tasmota Wiki

closed 2018-08-18 by ascillato2

GrendelWWU, 2020-01-06:
I don't have seen any changes since 2018. Is this project dead? I'm not a developer. SO, are there any working binaries for Devices like NodeMCU ESP32 WROOM? (The boards with Reset & Boot-Button)

meingraham, 2020-01-06:
Nobody is working on the project and there are no plans to port Tasmota to ESP32. There are no Tasmota binaries available for the ESP32.

GrendelWWU, 2020-01-06:
Why? The ESP32 is a great MCU. Sometimes, the 8266 has too few GPIOs.
I like Tasmota and I guess, there are no alternative firmware projects for the ESP32. Right?

meingraham, 2020-01-06:
@GrendelWWU 

There are at least two I/O expanders supported by Tasmota to add more GPIO to Tasmota. This is not a limitation when using Tasmota on an ESP82xx.

There is plenty of interest but nobody is available to work on porting Tasmota. It was already tried. But that effort died on the vine. As the Tasmota codebase is open source, nothing is stopping someone from doing giving it another go.

ESPHome, for one, supports the ESP32.

GrendelWWU, 2020-01-07:
Thank you for your hint to these MCP expanders. It seems, I'll have to learn, how to compile my own Tasmota, cause these chips are not enabled by default. At this moment, I've been just doing some C programs with gcc and some more smaller projects with Arduino Nano Boards. Maybe it would be the first step do learn porting Tasmota to ESP32? On the other hand.... the easiest way will be, just going there and using two WeMOS boards. ;-) I will think about it.
My project, for going into details, is, making this outlet smart:
https://www.bueroshop24.de/brennenstuhl-power-manager-pma-6-fach-steckdosenleiste-mit-%C3%BCberspannungsschutz-schwarz-328708?srpId=dd9eb1f577f5a2086a5da5abc39a8b8e&lkz=624763&obt=1&storeType=B2C&gclid=EAIaIQobChMIvfGb3Z_x5gIVT-h3Ch3clgEMEAQYASABEgJenfD_BwE

meingraham, 2020-01-07:
@GrendelWWU 

Give Gitpod a try. It is very simple to use. No software installation required.
https://tasmota.github.io/docs/#/installation/Prerequisites?id=online-compilers

mcsSolutions, 2020-01-07:
I have a Tasmota 5.1.2.0 spinoff for ESP32 upon which I include a trilateration location tracking application.  The source is available at http://mcsSprinklers.com/BLEScannerSource.zip.  This should be a reasonable base for Tasmota on ESP32.  There is a binary as *_BLE.bin in http://mcsSprinklers.com/mcsTasmota.zip that goes into basic ESP32 modules such as NodeMCU.  Documentation is contained as part of http://mcsSprinklers.com/mcsMQTT.pdf.

GrendelWWU, 2020-01-09:
@meingraham 
Thank you. I will see, which option is the best for me. But it'll take plenty of time.

@mcsSolutions 
Thanks a lot. Maybe this will help me, doing the first steps.