arendst/Tasmota #171: Sonoff Pow GPIO 1/3 TX/RX Pins

Filterer, 2017-03-09:
I have a few sonoff basics that I use GPIO14 wired to a hard light switch - this is working excellently and loving the firmware you have created.

I now want to do the same with the sonoff pow, however there are no easy header pins for adding an external switch. However on the esp's the serial tx/rx pins can be adjusted to also be general purpose io - and i can't quite work out if there is a specific reason you have not accounted for this option in the configure module page?

I have done a quick POC and added the appropriate flags to the sonoff_template.h file which set GPIO1 and GPIO3 as GPIO_USER which then allows them to be configured in the configuration page appropriately
[image]
once configured to be a switch I can then ground the rx/tx pins and the device switches perfectly, and setting back to 0 (None) means the serial port is back up and running after reboot.

Just wanted to check I am not doing something silly here? I can't see why it wouldn't have come up before as a suggestion as it is so trivial to implement? Assuming not I think it would make sense to make this available on all relevant sonoffs probably with some extra info on the configuration page so you know that these two GPIOs are special.

Let me know what you think I am happy to do the work and send over a pull request

arendst, 2017-03-09:
Several reasons why I did things:
- I do not support GPIO01 as user GPIO as there would be a lot of questions as why it doesn't perform as the user expected as serial input may come along.
- Sonoff Dual and (now obsolete) 4 Channel use the serial interface to control the relays
- Most important one for Sonoff Pow: THE GPIOs MIGHT BE CONNECTED TO AC POWER DEPENDING ON THE WAY YOUR POWER PLUG IS CONNECTED. I WILL NEVER SUPPORT GPIO ON SONOFF POW. You're on your own and it may work great as long as you do not touch the hardware while plugged in.

So no PR needed.

Filterer, 2017-03-09:
Thanks Theo, yes those reasons make sense. 

Just to make sure I understand you about the Pow are you saying the ground plane on LV side is tied to the ground/neutral HV input? So if accidentally neutral and line are reversed the ground plane could be live 240?

Seems like this is a bigger problem than just applying a PIN input/output - if you slightly incorrectly wire and leave the headers on the tx/rx you could be in for trouble?

arendst, 2017-03-09:
That is what I have tried to explain in the wiki. There you also find my optocoupler serial solution.

Filterer, 2017-03-09:
Thanks Theo I hadn't seen that one - for future reference for others if they come across this more details are here https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation.  I think we can close this ticket now.

Cheers

closed 2017-03-09 by Filterer

DNA99, 2017-06-29:
Hi Theo, thank you for the great work on the Tasmota. Sorry to jump on this thread. Do you know where I can find the GPIO pins for the Sonoff 4 CH? I only see GPIO1, GPIO2 and GPIO3 when I log into the module configurations page. I am trying to hook up the GPIO pins to a wall switch so that I can continue to use the existing wall switches I have. Is there possibility to configure 4 GPIO pins in the module configurations page instead of the 3?

arendst, 2017-06-29:
You're out of (accessable) GPIO on the Sonoff 4CH as can be seen in file ``sonoff_template.h``:
```
  { "Sonoff 4CH",      // Sonoff 4CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Optional sensor
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 Sonoff 4CH Red Led and Relay 3 (0 = Off, 1 = On)
     GPIO_REL2,        // GPIO05 Sonoff 4CH Red Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     GPIO_KEY3,        // GPIO10 Button 3
     0,
     GPIO_REL1,        // GPIO12 Red Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     GPIO_KEY4,        // GPIO14 Button 4
     GPIO_REL4,        // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)
     0, 0
  },
```

davidelang, 2017-06-29:
IIRC the 4ch has headers that match the built-in buttons. you could define the 
module as a wemos and use those GPIOs as switches.

davidelang, 2017-06-29:
question for @arendst, the pins for the flash are an SPI bus, is it possible to 
use a different GPIO as a chip select and attach other things to the SPI bus?

I have some power strips that I'm going to create a module for, and they have 
SPI based power measurement chips that I'd like to read.

arendst, 2017-06-29:
When I use SPI (for my SPI LCD display) I use GPIO pins like:
```
 * ESP-12F connections - TFT using ILI9340 with 320x240 display (MOD_TFT)
 *                     - TFT using ILI9481 with 480x320 display (MOD_TFT4)
 * 3V3   -------------
 *  |  -| Rst  GPIO01 |--- Tx
 *  |  -| Adc  GPIO03 |--- Rx
 *  |---| En   GPIO05 |- 
 *  |   |      GPIO04 |--- SPI_Rst (not connected) - Backlight
 *  |   |      GPIO00 |--- Button to Gnd (optional)
 *  |   |      GPIO02 |--- LED (ESP-12F)
 *  |   |      GPIO16 |--- SPI_DC
 *  |   |      GPIO14 |--- SPI_Clk
 *  |   |      GPIO12 |--- SPI_MISO
 *  |   |      GPIO13 |--- SPI_MOSI
 *  |   |      GPIO15 |-------|-- SPI_CS
 *  |---| Vcc     Gnd |---|  10k
 *       -------------    |   |
 *        | | | | | |    Gnd Gnd
```
declarations:
```
// TFT
#define SPI_miso               12           // GPIO 12 - Not used
#define SPI_mosi               13           // GPIO 13
#define SPI_sclk               14           // GPIO 14
#define SPI_cs                 15           // GPIO 15
#define SPI_dc                 16           // GPIO 16

#include <SPI.h>                // TFT
```
Hope that makes sense...

davidelang, 2017-06-29:
I was hoping to be able to use pins 10,13,14 on the esp-12f/esp-14

I'm trying to control 4 relays, 4 buttons, and if possible read 4 spi power 
chips (when we get the ability to use an esp-32 I'll add 2 more switches and 
relays)

this has me very short on pins, so I really want to be able to be able to share 
the SPI clock/miso/mosi pins

DNA99, 2017-06-29:
@arendst thanks for your reply! Sorry I'm a total newbie to this and its probably a wonder I havent killed myself yet trying to get this to work. Do you think I can use the GPIOs on the toggle buttons(I think its GPIO9, 10,  14)? And changing the code to allow changing these gpios to be configured as switches as I wont be using the toggle switches anyway.? Is this even possible?

davidelang, 2017-06-29:
@da99, you would need to not use the 4ch module type, as that defines them to be 
buttons, not switches (momentary only), you could use the wemos module type and 
then define them to be switch1-4 and then connect them to your external 
switches.

DNA99, 2017-06-29:
@davidelang thanks for your reply! how do I use this 4ch module type? Is it already in arendst code? And where do I find the code that defines these?

davidelang, 2017-06-29:
yes, it's already in the code. go to the configuration page and it lets you 
select the module type. arendst posted the definition for the 4ch module type a 
few posts back, that shows what's predefined for that type. You don't want to 
use that definition because you want to hook different things to the button 
inputs, so you can't use that module definition. You would have to use the wemos 
definition, which allows (forces) you to define every gpio

DNA99, 2017-06-29:
[image]
@davidelang do you mean that I should choose this in the configuration menu?

davidelang, 2017-06-29:
yes, after you switch to that module type, you will have a lot more gpios you 
can set

DNA99, 2017-06-29:
I saved that and I can't access my 4CH anymore... it doesn't even connect to my router.. probably something else I need to do?

davidelang, 2017-06-29:
hook it up to the serial cable and see what shows up (see the troubleshooting 
page on the wiki)

DNA99, 2017-06-29:
Pardon my noobness. what do you mean by connect to serial cable? like how I flashed the firmware?

davidelang, 2017-06-29:
yes

On Thu, 29 Jun 2017, DNA99 wrote:

> Pardon my noobness. what do you mean by connect to serial cable? like how I flashed the firmware?

DNA99, 2017-06-29:
Am I suppose to open a console or something? I only know how to flash the firmware.

DNA99, 2017-06-29:
OK I reflashed it with some modifications to sonoff_template.h. I changed the original sonoff 4CH to:
  { "Sonoff 4CH",      // Sonoff 4CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Optional sensor
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 Sonoff 4CH Red Led and Relay 3 (0 = Off, 1 = On)
     GPIO_REL2,        // GPIO05 Sonoff 4CH Red Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_USER,        // GPIO09 Button 2
     GPIO_USER,        // GPIO10 Button 3
     0,
     GPIO_REL1,        // GPIO12 Red Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     GPIO_USER,        // GPIO14 Button 4
     GPIO_REL4,        // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)
     0, 0
  },

and now after I log in to the IP and change the module config to Sonoff 4CH I have:
[image]
No to do some soldering and see if it works!

amitubale, 2017-06-30:
what software are you using to flash ?
"Am I suppose to open a console or something? I only know how to flash the firmware." --i have used ProgramIO windows app and you can flash and after the flash you can start Serial Monitor from eh menu to see the logs from teh TX/RX pins...take a look the video from QuickPi on youtube..its also in here in wiki

wavemaking, 2017-11-22:
Just as an additional thought to this issue: When the timeout required to put the module in reflashing mode is extended to a long duration, then I would find it accaptable to put a light switch on the same connections as the default button of the module. If the required time period is long enough (let's say 30 seconds), I would think that probability to accidentilly putting the module in reflashing mode is very  small. 

So my question: can the time period required to put the POW in reflashing mode be adjusted?

bphermansson, 2017-11-22:
@arendst I think the lack of Gpios on the Pow could be better explained in the Wiki. It took me a while to realize I was out of luck with a I2C-sensor and the Pow.

arendst, 2017-11-22:
@bphermansson Feel free to update the wiki with your comments.

arendst, 2017-11-22:
@wavemaking Scan the wiki to find commands to disable button hold and OTA initiation.

bphermansson, 2017-11-23:
@arendst Ok!

Deepakdsz, 2019-09-05:
> Thanks Theo I hadn't seen that one - for future reference for others if they come across this more details are here https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation. I think we can close this ticket now.
> 
> Cheers

This indeed helped in referencing, thank you for sharing the link.

corbolais, 2022-09-17:
> > Thanks Theo I hadn't seen that one - for future reference for others if they come across this more details are here https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation. I think we can close this ticket now.
> > Cheers
> 
> This indeed helped in referencing, thank you for sharing the link.

Link is dead meat by now, do you have a new link target?

Thanks.

barbudor, 2022-09-17:
Is this what you are looking for?
https://tasmota.github.io/docs/devices/Sonoff-Pow/

sfromis, 2022-09-17:
The old Hardware-Preparation page content is now part of
https://tasmota.github.io/docs/Getting-Started/#hardware-preparation