arendst/Tasmota #172: LWT configuration

sonic4002, 2017-03-09:
Hi,
i have mosquitto mqtt and sonoff
i want that the toggel button on HA will go "off" when i turn off the physical switch,
i found out that it can be done with LWT but i coulndnt figure how to config this option

please help me understand how to use the LWT option
thank you

davidelang, 2017-03-09:
what is LWT?

what is happening is that when you hit the button on the sonoff, it sends a mqtt 
message stat/name/POWER state (where state is ON or OFF) and then HA watches for 
that and changes the button status.

is LWT looking for stat/# messages?

On Thu, 9 Mar 2017, Omri Haviv wrote:

> Hi,
> i have mosquitto mqtt and sonoff
> i want that the toggel button on HA will go "off" when i turn off the physical switch,
> i found out that it can be done with LWT but i coulndnt figure how to config this option
>
> please help me understand how to use the LWT option
> thank you
>
>

sonic4002, 2017-03-09:
LWT - http://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament
i want that the HA will figure that the sonoff is off (i turned off the power from the physical switch)

1.i turned the physical sw on
2.i can turn the button on HA on and off and the sonoff reacts
3.HA switch is on ON state , light is physicly ON
4. i turn off the physical switch - sonoff turns off , HA switch stays on ON state
i want that HA will figure the sonoff lost power

a little search and i cant across LWT - it means that when a sonoff sign on MQTT it creates a "will"
when the MQTT tries to reach the sonoff (with keepalive) and it doesnt have any communication the MQTT apllies the "will" - and the "will" will say 
stat/sonoff_livingroom1/light  
payload = off

Hogesyx, 2017-03-10:
the default code already has a LWT, just listen to stat/<TOPIC NAME>/LWT for "offline" or "online".

sonic4002, 2017-03-10:
How? 
Tell me how to implement your solution

ozezra88, 2017-03-10:
Joining the request.

I can see only the LWT Online message when the sonoff is powered up. Cant see any "will" message. (Offline)

Hogesyx, 2017-03-10:
You won't be able to see will payload from the device itself, it is meant for the others to see. The MQTT server will automatically switch to Offline for that topic once the client disconnected.

ozezra88, 2017-03-10:
There is any mqtt configuration for this?

davidelang, 2017-03-10:
no, setup monitoring for every message and then turn a device off and after a 
while you will see the LWT message sent from the MQTT server

On Thu, 9 Mar 2017, ozezra88 wrote:

> There is any mqtt configuration for this?

Hogesyx, 2017-03-10:
Yes. Meaning for your device to turn on, monitor only POWER "ON", but for device to turn off, monitor both POWER "OFF" and LWT "Offline".

ozezra88, 2017-03-10:
I will appreciate if you will share mqtt configuration sample for this.

Thank you!

Hogesyx, 2017-03-10:
@ozezra88 we got no idea which software you are using, the configuration varies depending on your end point software.

ozezra88, 2017-03-10:
Home assistant with mosquitto mqtt.
Sonoff with the last tasmota firmware.

sonic4002, 2017-03-10:
We are using basic sonoff
Moquitto mqtt
And ardenst firmware

Hogesyx, 2017-03-10:
Moquitto is the Server
Sonoff is the MQTT publisher
You still need a MQTT subscriber <-- this is the place you need to code to listen to both payload.

davidelang, 2017-03-10:
This would be a configuration in Home Assistant, Not a tool I'm familiar with.

On Thu, 9 Mar 2017, ozezra88 wrote:

> Home assistant with mosquitto mqtt.
> Sonoff with the last tasmota firmware.

arendst, 2017-03-10:
And the wiki can't solve this?

ozezra88, 2017-03-10:
I could not find any reference for LWT 
:/

sonic4002, 2017-03-10:
each client should sign up the mqtt with its own "will"
so when it goes offline
the mqtt preforms the "will" - its the last thing to do after the sonoff has disconnected (or died)

arendst, 2017-03-10:
What do you know about LWT? From your questions I think it is not enough. Get informed by searching for MQTT Last Will and Testament. 

In the meantime start using the configurations as shown in the wiki - Home Assistant and enjoy your Sonoff.

davidelang, 2017-03-10:
On Fri, 10 Mar 2017, ozezra88 wrote:

> I could not find any reference for LWT
> :/

the message for the LWT is set on the sonoff. The MQTT server treats it just as 
another message to pass.

get something that lets you monitor the MQTT buss (on Android I use mqtt 
dashboard app) and connect to your mqtt server and subscribe to #, this will get 
you all messages. once you see that you can get the on/off messages, disconnect 
the sonoff from power and watch to see when you get the final message.

David Lang

davidelang, 2017-03-16:
did your question get answered? if so can we close this topic?

ozezra88, 2017-03-16:
Yes we did solve this issue

sonic4002, 2017-03-16:
Yes we managed to configure the ha to listen to the mqtt lwt message
It was a bit hard cause we couldn't find a proper example
The problem was on the HA side not the sonoff
Your firmware is amazing, thank you for your assistance

closed 2017-03-16 by davidelang

drfahod, 2017-05-12:
> sonic4002 commented on Mar 16:
Yes we managed to configure the ha to listen to the mqtt lwt message
It was a bit hard cause we couldn't find a proper example
The problem was on the HA side not the sonoff
Your firmware is amazing, thank you for your assistance

Hi, what is the configuration code of HA did you use? can you share it?
I have the same problem

sonic4002, 2017-05-12:
to figure how to catch the Last Will Testamet message we we signed to the MQTT with a regular PC (windows 10) , the software i used is mqttfx.
i signed to the MQTT and saw all topics - you need to sign to the mqtt with +/+/+ this means you'll see all messages on the network

now you need a code on the configurationg file that catches the "tele/sonoff_room/LWT"
and the payload is "Offline" when the sonoff disconnetcs from the network and "Online" when it reconnects. pay attention to the upper case "O"

so what we did was creating an automation section with the next code

copy this :
automation:
  hide_entity: True 
  trigger:
    platform: mqtt
    topic: "tele/sonoff_room/LWT"
    payload: "Offline"
  action:
     service: mqtt.publish
     data:
         topic : "stat/sonoff_room/POWER"
         payload : "OFF"

EXPLANATION : 

automation:     # new automation section in the configuration yaml
  hide_entity: True  # i dont want to see an on/off button to activate/deactivate the script so i hide it
  trigger:                 # what should be the trigger to activate the script
    platform: mqtt     # receive a message from the MQTT
    topic: "tele/sonoff_room/LWT"          # THIS IS THE TOPIC I WANT TO CATCH , i figured it with the mqttfx
    payload: "Offline"               # and this is the payload or the message 
  action:                                  # this is the action part - what will be done by activating the script
     service: mqtt.publish         # publish a message to the MQTT
     data:                              
         topic : "stat/sonoff_room/POWER"           # with this topic
         payload : "OFF"                                         # and this payload

we crafted an OFF message to the mqtt from HomeAssistant 
the MQTT sends the message to all the sonoffs
the relevant sonoff goes off

i'm sure you'll figure this out

hope i helped you, it took some time till we managed to do this :-)

good luck!

sonic4002, 2017-05-12:
one more thing
i can see that all the spaces are gone after posting the automation code
so i post another message 

`automation:
  hide_entity: True 
  trigger:
    platform: mqtt
    topic: "tele/sonoff_room/LWT"
    payload: "Offline"
  action:
     service: mqtt.publish
     data:
         topic : "stat/sonoff_room/POWER"
         payload : "OFF"
`

lance36, 2018-05-06:
just in case someone else googles tasmota lwt and end up in this thread, the solution of @sonic4002 is totally unnecessary or at least in my case.

just make sure to configure proper payload for availability for the switch, as the wiki suggests...
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

```
  - platform: mqtt
    name: "whatever"
    command_topic: "cmnd/<topic>/power"
    state_topic: "stat/<topic>/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: false
    availability_topic: "tele/<topic>/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
```

jnoxon, 2018-10-25:
Is there a way to reduce the lag when getting the Offline message? I assume the LWT Offline message is being generated by the broker somehow. I'm using Mosquitto and HA.

sfromis, 2019-01-27:
@jnoxon - The general way of getting the MQTT broker to discover dead clients faster would be to configure the client for a lower keepalive interval. Thats how often the client shouts that it is still alive, giving the broker the chance of noticing any absence. Beware that a shorter interval may adversely affect power usage and network traffic, and increase the risk of erroneous LWT messages being published. I'm running Tasmota on Sonoff, and while the status reveals a keepalive interval of 15 seconds, it does not appear to be configurable on the precompiled images.

lulol, 2022-05-14:
Using ```status/<TOPIC>/LWT``` to listen for LWT messages as shown in the above replies didn't work with the latest tasmota version.

Currently the right topic to listen to is: ```tele/<TOPIC>/LWT``` that with the mosquitto MQTT broker produces an output like:
```
$ mosquitto_sub -t "tele/tasmota_FABADA/LWT"
Offline
Online
```

The topic can be found in the tasmota console at boot:
```15:51:51.281 MQT: tele/tasmota_FABADA/LWT = Online (retained)```

This is nowhere shown in the [MQTT doc page](https://tasmota.github.io/docs/MQTT/) but in [Home-Assistant](https://tasmota.github.io/docs/Home-Assistant/) and [openHAB](https://tasmota.github.io/docs/openHAB/)

barbudor, 2022-05-14:
So you discovered the hard way that old and closed issues often includes outdated information whereas your device has the proper answer in the console.
Feel free to click on the pencil button on the top-right of the doc page to contributeto Tasmota documentation.
Thank you

lulol, 2022-05-14:
That is why I left the comment above and created a pull request. #986

And only could find the proper answer in the console after using a new device since the one where it was needed was cluttered with all the unrequested mqtt spam generated by the rules.

And I think that there is no need to be snarky when I only was trying to help others to avoid wasting the same time had to due to the lack of documentation and the erroneous replies. Thank you.