arendst/Tasmota #173: Can i make it to use with amazone echo also?

mp3llll, 2017-03-10:
Thanks.

davidelang, 2017-03-10:
Yes, if yout turn on the hue or wemo emulation, Amazon and Google should be able 
to find it.

David Lang

mp3llll, 2017-03-10:
Thanks david,

how i trun on the emulation?

Mp

davidelang, 2017-03-10:
On Fri, 10 Mar 2017, mp3llll wrote:

> Thanks david,
>
> how i trun on the emulation?

You can do it through the web management page, the console command line, or by 
sending the command via mqtt

see the commands page for the details.

David Lang

drbent15, 2017-03-10:
Hi, I have been try to get my Google Home to discover my sonoff basic for a while. When I turn on Wemo I see nothing in the console but when I turn on Hue I see this in the console even when I'm not running discovery it just keeps repeating:
Can you tell me What I'm doing worng?
Thanks
(192.168.1.16 is my Google Home and 192.168.1.5 is my Amazon Fire TV)

14:12:39 UPnP: HUE 3 response packets sent to 192.168.1.5:1290
14:13:04 UPnP: HUE 3 response packets sent to 192.168.1.16:57997
14:13:33 UPnP: HUE 3 response packets sent to 192.168.1.5:1290
14:14:26 UPnP: HUE 3 response packets sent to 192.168.1.5:1290
14:15:20 UPnP: HUE 3 response packets sent to 192.168.1.5:1290
14:16:04 UPnP: HUE 3 response packets sent to 192.168.1.16:57997
14:16:13 UPnP: HUE 3 response packets sent to 192.168.1.5:1290
14:16:59 MQTT: tele/sonoff/STATE = {"Time":"2017-03-10T14:16:59", "Uptime":11, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"NETGEAR2", "RSSI":64}}
14:17:04 UPnP: HUE 3 response packets sent to 192.168.1.16:57997
14:17:05 UPnP: HUE 3 response packets sent to 192.168.1.5:1290
14:18:00 UPnP: HUE 3 response packets sent to 192.168.1.5:1290
14:18:04 UPnP: HUE 3 response packets sent to 192.168.1.16:57997

davidelang, 2017-03-10:
that looks like both your Amazon Fire TV and your Google Home have found the 
sonoff and are talking to it once a minute or so to check it's status.

David Lang

drbent15, 2017-03-10:
Ok. But when i do a discovery they both say no device found. I don't know what I'm doing wrong. Do you have any ideas?

davidelang, 2017-03-10:
sorry, I've never used either system.

drbent15, 2017-03-11:
Okay thanks anyway, if you hear anything please let me know.

drbent15, 2017-03-11:
My research shows that Google home needs hue to be on port 80. Is there anyway i can get sonoff to advertise on port 80?

davidelang, 2017-03-19:
The webserver does run on port 80 by default, and the emulation is a subset of the webserver. so this should not be a problem.

The port numbers listed above were the ports used on the other end of the connection

davidelang, 2017-03-27:
take a look at #296 for additional discussion on emulation

closed 2017-04-02 by davidelang

khalifeserge, 2017-06-12:
This is my first post on Github and I am just starting to learn git and the process so excuse my comment instead of actual pull request and change.
Google Home send a request to the sonoff in json mode to the url /api
The code strips the first 4 chars before checking what to do next and end up calling hue_global_cfg(path);
A line should be added in xdrv_wemohue.ino to check for empty string and call the hue_auth(path) function.

This is what I did and tested on Google Home and now it works:

```
  if (path->endsWith("/invalid/")) {}                 // Just ignore
  else if (path->endsWith("/nouser/config")) hue_auth(path);       // New HUE App setup
  else if (path->length()==0) hue_auth(path); 
```
I can turn the relay off by saying Google home dim the light but I haven't figure out how to turn it back on.

shmrymbd, 2017-12-21:
hi.. my google home mini cant detect the tasmota sonoff.. already enable emulation. need help