arendst/Tasmota #174: Source for 4 pole plug (with 4 wire cable already attached) such as those used in the Sonoff TH?

mihalski, 2017-03-10:
I know I can get the 4 pole plugs and solder them myself but these are not as slim (and soldering very fiddly, more so even than SMD components) as the cables that come with the Sonoff modules.

Anyone know of a source for these?

I've searched aliexpress and eBay without any luck. The best I can figure out is to get the cheapest Sonoff modules on aliexpress and cut the cables to create my own. The enclosures that the Sonoff sensors use are readily available on aliexpress, but I guess in this way I would also get them in the bundle.

davidelang, 2017-03-16:
searching aliexpress for 2.5mm 4 pole jack plug

bulky solderless connectors

https://www.aliexpress.com/item/Video-AV-Balun-2-5mm-4-Pole-Stereo-Male-to-AV-Screw-Terminal-Stereo-Jack-2/32719169386.html?spm=2114.01010208.3.93.EDVsR6&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10140_10055_10054_10059_10099_129_10103_10102_10096_10052_10053_10050_10107_10051_10106_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_10078_10079_10073_10070_10122_10123_10126_10124,searchweb201603_5,afswitch_1,ppcSwitch_5,single_sort_0_price_asc&btsid=da9f9de8-d490-44c6-a2af-1291fee243a2&algo_expid=dfb047da-04e7-4885-aba1-453231803d23-10&algo_pvid=dfb047da-04e7-4885-aba1-453231803d23

right-angle solder connectors

https://www.aliexpress.com/item/2-piece-3-5mm-4-Pole-black-L-shaped-right-angle-Male-stereo-Plug-for-2/32760838770.html?spm=2114.01010208.3.11.EDVsR6&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10140_10055_10054_10059_10099_129_10103_10102_10096_10052_10053_10050_10107_10051_10106_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_10078_10079_10073_10070_10122_10123_10126_10124,searchweb201603_5,afswitch_1,ppcSwitch_5,single_sort_0_price_asc&btsid=da9f9de8-d490-44c6-a2af-1291fee243a2&algo_expid=dfb047da-04e7-4885-aba1-453231803d23-1&algo_pvid=dfb047da-04e7-4885-aba1-453231803d23

searching for 2.5mm 4 pole cable found

https://www.aliexpress.com/item/Car-AUX-4-pole-audio-cable-MP3-MP4-or-mobile-phone-to-car-CD-AUX-90/32460319440.html?spm=2114.01010208.3.27.hT4Fa2&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10140_10055_10054_10059_10099_129_10103_10102_10096_10052_10053_10050_10107_10051_10106_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_10078_10079_10073_10070_10122_10123_10126_10124,searchweb201603_5,afswitch_1,ppcSwitch_5,single_sort_0_default&btsid=bacc4e55-09d6-465d-9f03-993fde2b2a20&algo_expid=c536cef7-1708-4e01-8354-984153f566dc-3&algo_pvid=c536cef7-1708-4e01-8354-984153f566dc

https://www.aliexpress.com/item/High-Quality3Ft-1M-2-5mm-male-to-male-plug-4-pole-3-ring-extension-Audion-Aux/32484863475.html?spm=2114.01010208.3.77.zd6648&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10140_10055_10054_10059_10099_129_10103_10102_10096_10052_10053_10050_10107_10051_10106_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_10078_10079_10073_10070_10122_10123_10126_10124,searchweb201603_5,afswitch_1,ppcSwitch_5,single_sort_0_price_asc&btsid=3753cc49-601f-4974-b405-73e234f2a7fa&algo_expid=55a0b718-6453-4893-b1a6-d30221455101-9&algo_pvid=55a0b718-6453-4893-b1a6-d30221455101

does this get you started? can this issue b

mihalski, 2017-03-16:
I've ordered some of those items and many more similar ones but am doubtful any of them will be as easy to use as the ones the Sonoff sensors comer with so I ordered a few of those at $4 as well.

I guess there's no point leaving this open as it'll be weeks before I can comment on any of the items I ordered.

closed 2017-03-16 by mihalski