arendst/Tasmota #176: Can Sonoff LED response jason in 1 line?

hamwong, 2017-03-10:
Hi, @arendst,

Sorry for spam you again, can you do a little adjust on Sonoff LED response?

cmnd/sonoff-LED/dimmer 2
stat/sonoff-LED/RESULT {"POWER":"ON"}
stat/sonoff-LED/POWER ON
stat/sonoff-LED/RESULT {"Dimmer":2}

cmnd/sonoff-LED/dimmer 0
stat/sonoff-LED/RESULT {"POWER":"ON"}
stat/sonoff-LED/POWER ON
stat/sonoff-LED/RESULT {"Dimmer":0}

cmnd/sonoff-LED/power off
stat/sonoff-LED/RESULT {"POWER":"OFF"}
stat/sonoff-LED/POWER OFF

Is there anyway that RESULT in one line like stat/sonoff-LED/RESULT {"POWER":"ON", "Dimmer":2}?
also when dimmer 0, can it be POWER OFF?

hamwong, 2017-03-11:
Dear @arendst ,
If you will do modify your code base on LED,
Beside Jason output that Power ON with Dimmer,
Please also let Power off with Dimmer value,

Also I feel difficult on adjust LED color (temperature), will it be able to add a slider for adjust color, I think that would more useful than dimmer slider

Thank you very much for you attention and best regards.

hamwong, 2017-03-15:
Dear @arendst,

Huge thanks for promote update,
command dimmer and color is in 1 line now on response except on Power command
cmnd/sonoff-LED/Dimmer ON
stat/sonoff-LED/RESULT {"POWER":"ON"}
stat/sonoff-LED/POWER ON
stat/sonoff-LED/RESULT {"POWER":"ON", "Dimmer":1, "Color":"0200"}
cmnd/sonoff-LED/Dimmer OFF
stat/sonoff-LED/RESULT {"POWER":"OFF"}
stat/sonoff-LED/POWER OFF

davidelang, 2017-03-27:
the power stat output is being sent in the legacy format to keep the simple on/off switching working for people.

however, from your sample,it looks like it is being sent twice when it's turned on. can you confirm this?

hamwong, 2017-03-27:
in fact it's not double msg, 
when in power on /off command,
there is 2 output, 1 in normal output and one in json,
it's just not output in standard json output, it somehow output with single Json data output
as I mention above, I don't care simple onput:  POWER ON
but in Json format, it better be standard output
which means in RESULT  should always like below dimmer 0, not just a single power status output in json

02:43:31 CMND: dimmer 0
02:43:31 MQTT: stat/sonoff-LED/RESULT = {"POWER":"OFF", "Dimmer":0, "Color":"0000"}
02:43:37 CMND: power off
02:43:37 MQTT: stat/sonoff-LED/RESULT = {"POWER":"OFF"}
02:43:37 MQTT: stat/sonoff-LED/POWER = OFF

Also there is 1 more bug with jade, jade from 0-100 is fine,
but when fade 100-1 is fine, but not 0, 0 will instantly turn dark,
I guess it should due to logic involve above.

closed 2017-09-08 by arendst