arendst/Tasmota #178: add gpio 0 in user test module

PitPaul, 2017-03-11:


arendst, 2017-03-11:
I'll drop User test in the next release as the Wemos D1 mini has exactly what you want; all GPIOs are user selectable

PitPaul, 2017-03-11:
Thanks ( vielen dank im voraus )

closed 2017-03-11 by PitPaul

davidelang, 2017-03-11:
On Sat, 11 Mar 2017, Theo Arends wrote:

> I'll drop User test in the next release as the Wemos D1 mini has exactly what you want; all GPIOs are user selectable

Please keep user test (unless all features are now selectable)

arendst, 2017-03-11:
See the wemos defines. All is user selectable so the user test will be gone in next release.

joann235, 2017-04-29:
The Wemos support looks great so far, unless I am mistaken.  I loaded the firmware onto my D1 Mini and configured it using GPIO12 as a relay output.  The problem I am experiencing, one that I have seen before on other micro-controllers, is that if the unit power cycles so does the relay.  In my target application as a remote garage door controller this is bad because if the unit experiences a power failure then the door will open unexpectedly when power is restored.  If I am away this could be a huge problem.  Is this a problem that can be addressed in a future update to the software?

davidelang, 2017-04-29:
not really, the relay click you are seeing is happening before the software is 
up and running.

the best that you could do is to setup the relay to be inverted, instead of 
going from the pin to ground to drive your relay, try going from 3.3v to the pin

other than that, you need to setup some way to require a longer pulse before 
triggering the garage door, you could probably do something with a capacitor on 
the output of the relay that would absorb the power for a bit (possibly with a 
pull-down resistor to slow it's charging), it depends on how sensitive the 
garage door is.

David Lang

joann235, 2017-04-29:
I came to the same conclusion as you did and with similar solutions..  I was heading down the path of having a re-settable  latching circuit that you would latch on first startup of the device.  After that, in the even of a power failure, the latch would open interrupting the output of the relay till it is manually reset.  This is not a perfect solution but I think it is a workable solution.  Thanks for responding so quickly to my post.