arendst/Tasmota #181: Long passwords does not work for MQTT

choeflake, 2017-03-11:
Using a password of 50 characters does not work, a password of 25 will work. I have not searched for longest possible password.

I use Mosquitto as the MQTT server. When using a password of 50 characters I can connect with other clients, but the ESP shows the error (rc 5 = auth failed):
MQTT: Attempting connection...
MQTT: CONNECT FAILED, rc 5. Retry in 10 seconds

I'm not sure if this has to be registered as a bug. Documenting the max password length is also viable.

choeflake, 2017-03-11:
In settings.h is mttt_pwd defined as 
`char          mqtt_pwd[33]`

arendst, 2017-03-11:
It's 32 indeed.

closed 2017-03-12 by arendst

khcnz, 2017-03-13:
I have added some documentation here for others in the future https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview

knutbert, 2019-01-08:
Hi there,
I stumbled across this limitation while trying to connect my tasmota-enabled device (OBI socket) to QNAPs QIoT suite, which generates/requires usernames of 36 charcters and passwords of 34 characters.
This seemed to be easy enough to fix by increasing the respective array sizes in settings.h. To move things around in struct SYSCONFIG as little as possible I reduced the size of the unused free_1D5 array by the amount of additional bytes required:

```
struct SYSCFG {
.....
// byte          free_1D5[20];              // 1D5  Free since 5.12.0e
byte          free_1D5[4];              // 1D5  Free since 5.12.0e --> Reduced by 16 bytes needed to increase mqtt_user and mqtt_pwd
.....
// char          mqtt_user[33];             // 22D
  char          mqtt_user[41];             // 22D --> 21D (position shifted by 16 bytes) ; added 8 bytes
  // char          mqtt_pwd[33];              // 24E
char          mqtt_pwd[41];              // 24E --> 246 (position shifted by 8 bytes); added 8 bytes
.....
}
```

With the new firmware the device now connects successfully to the MQTT broker in QNAPs QIoT suite.

Unfortunately when connecting via MQTT to the server the device forgets the configured type (OBI socket). Also web-server response is really sluggish then. This behavior is reproducible.

If I inhibit the data transmission by setting a non-existent MQTT server address everything is fine. But as soon as I re-enable MQTT by setting the right server IP the device acts strange again.

Any idea on what's going on?
Was it a bad idea to increase the array sizes and/or move things around in struct SYSCONFIG?
Or is there a known limitation in one of the libraries used?

Any help is appreciated!

Regards,
knutbert