arendst/Tasmota #182: Suggestion: Support for i2c GPIO port expanders

mihalski, 2017-03-11:
This would be most useful for Sonoff modules which don't expose many GPIO's.

arendst, 2017-03-11:
#149

davidelang, 2017-03-11:
Are there any such port expanders that are standardized/common enough to bother 
with? Are they cheaper than just getting an ESP-8266 and hooking up all the 
GPIOs to sensors instead?

On Sat, 11 Mar 2017, mihalski wrote:

> This would be most useful for Sonoff modules which don't expose many GPIO's.

mihalski, 2017-03-11:
Yes. The MCP23008/MCP23017 are probably the most common and cheap.

Wouldn't even need to take the Sonoff TH apart to use.

davidelang, 2017-03-11:
On Sat, 11 Mar 2017, mihalski wrote:

> Yes. The MCP23008/MCP23017 are probably the most common and cheap.
>
> Wouldn't even need to take the Sonoff TH apart to use.

Actually, you would, the TH has a single-wire connection, i2c needs two wires 
(data + clock)

The adafruit library for this chip would need to be re-written to work with the 
ESP-8266, is there another library that would work better? And how large would 
the resulting code be? We are running tight on both storage and RAM.

I don't know if arendst will be interested in adding this or not (given that 
this is almost as expensive as a ESP8266 module that would give you far more 
GPIOs, and would use the stock code, it doesn't seem like it's worth a lot of 
effort, but if the code is contributed, and can be 'ifdef'ed out by default, he 
may accept a PR

David Lang (user, not project owner)

mihalski, 2017-03-11:
Hmm. Well that puts a damper on this idea. I know the plug supports 4 wires, I didn't realise it was only using 3. I still have a Sonoff TH that refuses to work correctly so I guess I may be looking into the details a lot more closely at some stage.

I did consider the code size problem but I also though that it should be a compile time option as it won't be used by most.

I wish I had more time to better expand my knowledge and ability to contribute :/ I have plenty of ideas but not enough ability to execute on them or even evaluate their feasibility.

Thinking about it a little more I think I should curb my desire to have Tasmota as the ultimate do everything firmware for esp8266 as that's clearly not what Theo is aiming for.. Still, I would love if he added the Mitsubishi/Toshiba IR controls now available in the IR library ;P

davidelang, 2017-03-11:
you could power another ESP8266 from the TH plug, this one is <$2 including 
shipping (but will take a while to arrive)

https://www.aliexpress.com/item/1PCS-ESP-12F-ESP-12E-upgrade-ESP8266-Remote-Serial-Port-WIFI-Wireless-Module/32714088769.html?spm=2114.01010208.3.37.lPg8dZ&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10056_10055_10054_302_10059_10099_10103_10102_10096_10052_10053_10050_10107_10051_10106_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_5000000_10078_10079_10073_10070_10122_10123_10126_10124,searchweb201603_3,afswitch_1,ppcSwitch_5,single_sort_0_price_asc&btsid=96ce8eaa-2afa-4183-9cf9-f2d1aecb7f2b&algo_expid=4e8272df-722e-42c5-ba47-5f3f7c1d3245-4&algo_pvid=4e8272df-722e-42c5-ba47-5f3f7c1d3245

it will give you at least a dozen GPIOs (if not all 16)

mihalski, 2017-03-11:
That's insane. I'm ordering 5 just based on principle..
To add to the 5 that haven't even arrived yet.

If you could find the plugs + cables for a similar price that would be great :P

arendst, 2017-03-11:
Why do you need I2C expansion? I only support one type of I2C devices each (HTU21, BMP280 and BH1750) as the code is non re-entrant and you can connect several I2C devices to the same I2C bus. In fact I have both HTU21 and BMP280 happily connected to one I2C bus on GPIO4/GPIO14.

I would not buy ESP12F anymore as there is now WeMos D1 mini with USB for serial connection. Once flashed all you need is an external 5V power supply and you can stick them anywhere.

mihalski, 2017-03-11:
I was hoping for a pathway to support rotary encoders at the same time as buttons and i2c sensors.

ie. https://blog.squix.org/2016/05/esp8266-peripherals-ky-040-rotary-encoder.html

closed 2017-03-12 by arendst

mihalski, 2017-03-13:
@davidelang There is a problem with having temp sensors close to the esp8266. The temperature readings tend to be WAY off the actual ambient temperature.