arendst/Tasmota #183: ElectroDragon and more GPIOs

MacWyznawca, 2017-03-11:
Can You add more GPIOs do ElectroDragon config?
Board has pins for:
GPIO4 (I tested, working)
GPIO5 (I tested, working)
GPIO14 (I tested, working)
GPIO15
RX (I tested, working)
TX (I tested, working)
and
ADC

I test in User and ESX Relay mode

MacWyznawca, 2017-03-11:
OK, I tested this, and working (GPIO15 don't work as switch, but sensors and WS2812 LET - OK)

{ "ElectroDragon",   // ElectroDragon IoT Relay Board (ESP8266)
     GPIO_KEY2,        // GPIO00 Button 2
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_KEY1,        // GPIO02 Button 1
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 Optional sensor
     GPIO_USER,        // GPIO05
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_REL2,        // GPIO12 Red Led and Relay 2 (0 = Off, 1 = On)
     GPIO_REL1,        // GPIO13 Red Led and Relay 1 (0 = Off, 1 = On)
     GPIO_USER,        // GPIO14 Optional sensor
     GPIO_USER,        // GPIO15 (don't working as switch, but WS2812 OK, DHT11 and other sensors OK)
     GPIO_LED1         // GPIO16 Green/Blue Led (1 = On, 0 = Off)
  },

arendst, 2017-03-11:
Will be released in version 4.0.4

closed 2017-03-12 by arendst