arendst/Tasmota #184: Enhancement: Adding Switching behaviour / Grouping

felixekman, 2017-03-11:
A suggestion would be to add the functionality to set how switching behaviour works on on/off/toggle commands. For example, with usage for devices with multiple relays controlling one device. Usage example, controlling blinds or motors with two relays for up/down or on/off with the option to set either latching or pulsating command. Currently it’s possible to do via MQTT with separate commands but toggle not working as one device.

**Usage Example:**

A Sonoff Dev/PSB-04 connected to a blinds system with two relays controlling up and down. Currently the system would require the ‘down’ command to be sent to one relay whilst ‘up’ is sent to another. Whilst toggle would not work. The ability to say that a down/on command turns relay 1 on and an up/off command turns relay 1 off and relay 2 on. Ideally the ability to set latching or pulse command for this would cover most use cases. In cases where for example blinds are set to down with one short 100 ms pulse on relay 1 and turned up with a 100 ms pulse on relay 2.

**Useful for the following hardware:**

- Sonoff 4CH
- Sonoff Dual
- PSB-04
- Sonoff DEV
- WeMos D1 mini

arendst, 2017-03-11:
See motor clockwise / anti clockwise device from itead. Link in wiki.

felixekman, 2017-03-11:
I actually already have this unit and whilst standard from factory it switches between the two and can't have both simultaneously off/on. I have made a hardware mod to it by removing one of the control chips and re-routing a wire from a spare GPIO on the PSA-01 module attached (which has more than one output pin free). 
But the problem remains that the current firmware does not make it easy to change the behaviour of on/off and or pulse when combining two relays for one device that needs to have relays not always on. I've done a custom mod of your firmware for my device but it only applies to this limited use case and I think a broader integration would be beneficial for more people.

aferraz, 2017-03-11:
@arendst at the moment I use nodered for what @felixekman describes.
I use electrodragon relay board for covers. With nodered I listen on the state of POWER and POWER1 and after some second (when the cover is totally closed/opened) nodered give the OFF to te relay.
In the same time nodered control that only 1 relay is ON at the same time.

Integrate this scenario on the firmware would be great.

Thanks
Alessandro

davidelang, 2017-03-16:
just a thought on this, trying to use two relays to control direction can run into grief (bad things tend to happen if both get turned on at the same time, look at h-bridge documentation for an example of why)

now that you can hook additional relays to the GPIO pins, I think the best approach would be to have one DPDT relay to control direction and a second one to turn power on/off, then have your central software manage it. This way you can even setup additional switches to detect your limits

davidelang, 2017-03-18:
space is tight on these devices, we can't add every control scenario to the devices.

The intent of this firmware isn't to put lots of control logic onto the device, but to put enough onto the device to allow them to talk to a home automation server that can have anything you can dream up added to it.

closed 2017-03-18 by davidelang