arendst/Tasmota #185: Upload problem

gemitro, 2017-03-11:
I can't upload Tasmota 4.0.3 (with static ip) to Sonoff basic using Arduino IDE.
It compiles but when i try to upload it gives me the following errors.

[Sonoff.txt](https://github.com/arendst/Sonoff-Tasmota/files/835707/Sonoff.txt)

lucwuyts, 2017-03-11:
I guess: wrong com-port

davidelang, 2017-03-11:
On Sat, 11 Mar 2017, lucwuyts wrote:

> I guess: wrong com-port

or tx/rx reversed

gemitro, 2017-03-11:
Both com port and tx/rx are ok.
I have successfully upload Tasmota more than 10 times when suddenly this happened.

lucwuyts, 2017-03-11:
The error message is :  espcomm_open failed
Maybe the port is in use by another program?
Try to open the port with a terminal program?

gemitro, 2017-03-11:
Maybe I damaged the usb to serial TTL when accidentally reversed the 3.3V to the ground for three minutes.

macadamer, 2017-03-12:
On your USB/Serial dongle, connect together TX and RX and use the serial monitor tool in the Arduino IDE to send some characters, which should appear in the monitor main window if the dongle is working.

gemitro, 2017-03-12:
Thanks
I connected together TX and RX on the USB/Serial dongle and sent a few characters with the serial monitor in the Arduino IDE as you suggested, but nothing appeared in the monitor main window.

macadamer, 2017-03-12:
Ho hum!

closed 2017-03-12 by arendst