arendst/Tasmota #186: Updateing to new firware

Slip7000, 2017-03-11:
Hi All,

Please share a small guide how and where to get the updated firmware and how to please.

davidelang, 2017-03-11:
On Sat, 11 Mar 2017, Slip7000 wrote:

> Please share a small guide how and where to get the updated firmware and how to please.

The wiki goes through the steps that you need to use to compile the firmware and 
flash it. Is there something you have specific questions about?

David Lang

Slip7000, 2017-03-11:
The option on the Sonoff itself gives an URL option or "Upgrade by file upload"

how can I use them?

TX

Slip7000, 2017-03-11:
[image]

davidelang, 2017-03-11:
On Sat, 11 Mar 2017, Slip7000 wrote:

> The option on the Sonoff itself gives an URL option or "Upgrade by file upload"
>
> how can I use them?

compile the new firmware, and then either put it on a webserver (and use the URL 
option) or select it with the 'choose file' option and hit upgrade.

If you are using the arduino IDE, you can select the OTA option and have it 
compile and upload the file to the webserver (if you have the webserver setup 
for it)

decologne, 2017-03-11:
Try the next url for update:
http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin

davidelang, 2017-03-16:
did your question get answered? if so, can we close this issue?

Slip7000, 2017-03-16:
Hi David I did not use the URL, but thank you very much for all your time and trouble. 
I reloaded the Sonoff with the new firmware via Atom as I received my new once and they also needed firmware for out of the box.

Thank you

closed 2017-03-16 by Slip7000