arendst/Tasmota #187: Touch w/ TouchScreen

pimvanpelt, 2017-03-11:
I am interested in modifying the physical Sonoff Touch module to include a 2.4" TouchScreen. This way, the user can configure icons that turn on/off the local relay, or send power on/off/toggle messages to remote devices. 

I have two options:
1) If I were to create the code in Tasmota that implements this, would you accept that, or are you interested in supporting stock Sonoff devices only? 

2) I can create a custom project that does this, and interface with Tasmota purely via pubsub. It would be unfortunate, as I'd then be re-implementing various bits and pieces to expose the local relay on pubsub, and a lot of excellent plumbing (webserver, ota, flash config, etc).

I'd really appreciate it if you were OK with (1), probably hidden behind a master #define in user_config.h so that other users are not bothered by the code that implements the screen driver.

Could you let me know your thoughts, Theo?

groetjes uit Zwitserland,
Pim

arendst, 2017-03-11:
Pim, take option 1 using a define as I'm afraid memory is tigh so people can choose what to select. Send a PR and I'll see if it fits.

trusten, Theo.

pimvanpelt, 2017-03-11:
Will do! Thanks for the opportunity. I'm going to play around with DigoleSerial for the moment, although I'm seeing a lot of issues from people using it with ESP8266 platform family. It may take a while, but I'll follow your main branch and see if I can send you a demo somewhere in Q2'17.

closed 2017-03-11 by pimvanpelt