arendst/Tasmota #188: what is the license on this code?

davidelang, 2017-03-11:
I may be just overlooking it, but I'm not seeing any license listed for this codebase.

arendst, 2017-03-11:
Only the one you see at the top of some the files. Other than that there is no license.

davidelang, 2017-03-11:
In many parts of the world "no license" means that technically we aren't allowed 
to use it at all. If you mean "no license" to be "do anything you want with this 
code", (and looking at the files that do have license text in them, and how the 
project is used, this seems like what you mean), please declare the rest of the 
code to be under the same license (it looks like your intent is BSD)

> Only the one you see at the top of some the files. Other than that there is no license.

benjamincburns, 2017-05-13:
@davidelang & @arendst I haven't done a full review on all of the libraries referenced by this project, but I believe that since at least some of these libraries are LGPL licensed (e.g. `core_esp8266_timer.c`), it wouldn't matter if this project were BSD licensed, as the "static linking" rule from the LGPL would apply. I suppose you could distribute unlinked binaries to get around this, but that wouldn't be particularly useful to anyone.

davidelang, 2017-05-13:
well, arendst only distributes source, so it really doesn't matter (except 
possibly for the licenses that are forked, but since they are distributed as 
source as well, that complies with any licenses involved)

benjamincburns, 2017-05-13:
Yeah, I don't mean to say he's not in compliance, just that since the project statically links to LGPL code, the distribution restrictions of that license would need to apply to others looking to distribute this software or derivative works.

benjamincburns, 2017-05-13:
But my comments distract from your previous point, which I strongly agree with. @arendst if you haven't done so already, I strongly recommend reading https://choosealicense.com/no-license/ (Don't worry, it's a short read). The most pertinent detail from this is that by explicitly declaring that there is no license, you're declaring that nobody may use your software. Worse yet, if others contribute to this work under similar terms, technically _nobody_ may use this software, as no permission for use of the complete work has been granted to anyone.

I'm working on a different PR today, but if you're willing to declare a license I'd be happy to submit a PR w/ the various license documents, headers, etc, as well.

arendst, 2017-05-13:
Thnx for the license info. Looked at the link supplied by @benjamincburns and now have to decide between the MIT or LGPL license. The latter one needs "an" email and mail address. Looking at the arduino esp8266 project files this seems to be the default yoyodine info. Is that enough?

I prefer the LGPL.

closed 2017-08-09 by arendst

neographikal, 2017-09-05:
Please add the license info to the readme.md