arendst/Tasmota #191: Different topic for different buttons.

MacWyznawca, 2017-03-12:
Whether it is or would be possible to define different "topic" for different buttons? So that from one device to manage several others?

Now I am able to control one device with another. However Electrdragon has the ability to connect up to 6 buttons and could be better to use it with to control few others devices.

davidelang, 2017-03-12:
when you hit a button it sends a stat/* message

you need a cmnd/* message to switch a device on or off

these need to be different or you end up with a loop where the sonoff sends a 
message to the mqtt hub, and it sends it back to the sonoff, which responds to 
it and sends a message to the mqtt hub.....

so you are always going to need something besides the sonoffs and the mqtt hub 
in the loop. Whatever that something is, it can handle stat/node1/power[1234] 
just as well as it would handle stat/note[1234]/power

so it really doesn't make a functional difference which is used.

or am I missing something?

David Lang

MacWyznawca, 2017-03-12:
Yes, You miss 😉

Now I use ButtonTopic  | <topic>      | Set MQTT button topic
for set topic of other device.
First (controling) have "dragon001" secont (controled) "dragon002"
I sed in first ButtonTopic "dragon002" If I press button defined to switch 1, first devise send cmd/dragon002/POWER ON (etc.) this veru useful, but I can control one device from other becouse I cann set only one special divverend topic for button use only ;-)

arendst, 2017-03-12:
Well if you set buttontopic to dragon002 it would send `cmnd/dragon002/POWER1 TOGGLE` for button 1 and `cmnd/dragon002/POWER2 TOGGLE` for button 2. At least that is what my Sonoff 4CH does. Using MQTT you can act on these messages to whatever you like.

MacWyznawca, 2017-03-12:
OK, this work.
But I want to make one button of dragon001 to control the dragon002 and the second button of dragon001 to control dragon003.

davidelang, 2017-03-12:
so either modify the code yourself, or use some control software in or talking 
to the mqtt server to do this.

> But I want to make one button of dragon001 to control the dragon002 and the second button of dragon001 to control dragon003.

MacWyznawca, 2017-03-12:
I can do it in HomeBridge (HomeKit), but I wanted to ask, because as already is the solution in  Tasmota firmware to control by the MQTT one on one, it could and would be one of many.

davidelang, 2017-03-12:
you are asking tasmota to do things beyond it's configuration. HomeBridge is the 
right place to do this sort of thing.

On Sun, 12 Mar 2017, Jaromir Kopp wrote:

> I can do it in HomeBridge (HomeKit), but I wanted to ask, because as already is the solution in  Tasmota firmware to control by the MQTT one on one, it could and would be one of many.

MacWyznawca, 2017-03-12:
Buddy, if it were, it would not do Arends special solutions and additional topic for the button, but it added. Truth?
So it was worth it to ask for more. OKAY?

davidelang, 2017-03-12:
the reason for buttontopic is to make it so that buttons don't change the local 
relays, but instead send to the central system without changing things locally.

I'm not the one running this project, arendst is, and he has explained the same 
thing multiple times (you are far from the first to ask about this). I'm just 
trying to assist him by answering questions so he can spend more time coding.

David Lang

On Sun, 12 Mar 2017, Jaromir Kopp wrote:

> Buddy, if it were, it would not do Arends special solutions and additional topic for the button, but it added. Truth?
> So it was worth it to ask for more. OKAY?
>
>

closed 2017-03-12 by arendst