arendst/Tasmota #192: #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"

Slip7000, 2017-03-12:
Morning All,

Please assist I'm trying to load the latest firmware onto a Sonoff via Arduino and getting this error:

 #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"

Thank you 
@Slip7000

davidelang, 2017-03-12:
The error message is straightforward, you need to edit the file PubSubClient.h 
and change the value of MQTT_MAX_PACKET_SIZE to be at least 427.

Have you tried to do this?

On Sun, 12 Mar 2017, Slip7000 wrote:

> Please assist I'm trying to load the latest firmware onto a Sonoff via Arduino and getting this error:
>
> #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"

Slip7000, 2017-03-12:
Where can I find the this PubSubClient.h file ?

Slip7000, 2017-03-12:
This is the Complete error:

Arduino: 1.6.12 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

sonoff:148: error: #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"
   #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"
    ^
/Users/SlipPro/Box Sync/Sonoff/Sonoff-Tasmota-master/sonoff/sonoff.ino:155:75: fatal error: ArduinoJson.h: No such file or directory
 #include <ArduinoJson.h>                    // WemoHue, IRremote, Domoticz
                                                                           ^
compilation terminated.
exit status 1
#error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

davidelang, 2017-03-12:
as the error message says, you find it in libraries/PubSubClient/src

This is where you installed the pubsub library as you were setting things up.

On Sun, 12 Mar 2017, Slip7000 wrote:

> Where can I find the this PubSubClient.h file ?

davidelang, 2017-03-12:
This is showing two error messages, first you set the variable larger, second 
you need to install the arduinoJson library.

On Sun, 12 Mar 2017, Slip7000 wrote:

> Date: Sun, 12 Mar 2017 01:59:33 -0800
> From: Slip7000 <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f6906735d9644ba49d5ed8d2bd73caf47d7ee470a92cf0000000114dcde0
>     592a169ce0cbb2dad@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: davidelang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] #error
>     "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubCli
>     ent.h, increase it to at least 427" (#192)
> 
> This is the Complete error:
>
> Arduino: 1.6.12 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"
>
> sonoff:148: error: #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"
>   #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"
>    ^
> /Users/SlipPro/Box Sync/Sonoff/Sonoff-Tasmota-master/sonoff/sonoff.ino:155:75: fatal error: ArduinoJson.h: No such file or directory
> #include <ArduinoJson.h>                    // WemoHue, IRremote, Domoticz
>                                                                           ^
> compilation terminated.
> exit status 1
> #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 427"
>
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>
>
>

Slip7000, 2017-03-13:
ok I'm still having a problem, found the file and it was already at 512 like below

#define MQTT_MAX_PACKET_SIZE 512

khcnz, 2017-03-13:
I had this problem as well - are you on windows and using Arduino IDE? I found the same issue but Arduino IDE was using a copy in Documents/Arduino/Libraries/ and not Documents/Arduino/Sonoff-Tasmota/Libraries/

I ended up creating a junction from Documents/Arduino/Libraries/ to Documents/Arduino/Libraries/

Slip7000, 2017-03-13:
Thank you, I'll try tonight. I'm using Mac and did change the root Libraries in the options of Arduino but still getting the error. I also reinstalled the app completely, but will try on windows tonight. 

Will let you know...

davidelang, 2017-03-13:
IF you changed it and are still getting the error, then there is another copy of 
the library around somewhere on your system and the one you changed is not what 
the system is actually using.

On mac, you can do

find / -name PubSubClient.h

and have it report back all the places the library exists.

  On Mon, 13 Mar 2017, Slip7000 wrote:

> Thank you, I'll try tonight. I'm using Mac and did change the root Libraries 
> in the options of Arduino but still getting the error. I also reinstalled the 
> app completely, but will try on windows tonight.

Slip7000, 2017-03-14:
Thank you, got it working. I tried Atom and open the Sonoff file there changed the config and uploaded successful.

Thank you for everybody help

closed 2017-03-14 by Slip7000