arendst/Tasmota #193: How to display status in Openhab2

andrestrauss, 2017-03-12:
Hi There.
I have a basoc sonoff working perfectly with arendst firmware. It connect fine to MQTT and i can see status in console of sonoff and also on the mosquitto_sub

How do I display those statusses in Openhab2 UI. I just want to see if the sonoff is online, and iff it has power on/off

I tried using this item
this item

default.items
Number Test1 “Status [%s]" (GF_Living) { mqtt="<[mosquitto:stat/sonoff2/power:state]" }

and this sitemap

sitemap default label="My first sitemap"
{
    Switch item=Light1 label="Light11 Switch"
    Text item=Test1
 }

But it doesnt work at all

Please help

davidelang, 2017-03-12:
On Sun, 12 Mar 2017, andrestrauss wrote:

> default.items
> Number Test1 “Status [%s]" (GF_Living) { mqtt="<[mosquitto:stat/sonoff2/power:state]" }

I use something like:

Switch sw03 "LR Light" { 
mqtt=">[openhab-mqtt:cmnd/sw03/POWER:command:OFF:off],>[openhab-mqtt:cmnd/sw03/POWER:command:ON:on],<[openhab-mqtt:stat/sw03/POWER:state:MAP(binary.map)],<[openhab-mqtt:tele/sw03/POWER:state:MAP(binary.map)]", 
autoupdate="false"

and then in the sitemap

Switch item=sw03

andrestrauss, 2017-03-12:
Thanks davidelang
The switch works, but it still doesnt change the state when I manually change the stat using the button on the sonoff. 

I would like to know in OpenHab2 what the power state is without switching the switch
Is it possible?

You refer to binary.map, does that reside in tranform folder, and what does it look like?

thanks in advance

davidelang, 2017-03-13:
you should have gotten error messages about binary.map being missing

root@weather:/etc/openhab/configurations/transform# cat binary.map
1=ON
0=OFF
On=ON
Off=OFF
ON=ON
OFF=OFF

On Sun, 12 Mar 2017, andrestrauss wrote:

> Date: Sun, 12 Mar 2017 09:08:58 -0700
> From: andrestrauss <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f09e95bbdc22ec7ad5226480be81cf23b0d52847892cf0000000114dd349
>     a92a169ce0cbb61f7@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: davidelang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] How to display status in Openhab2 (#193)
> 
> Thanks davidelang
> The switch works, but it still doesnt change the state when I manually change the stat using the button on the sonoff.
>
> I would like to know in OpenHab2 what the power state is without switching the switch
> Is it possible?
>
> You refer to binary.map, does that reside in tranform folder, and what does it look like?
>
> thanks in advance
>
>

davidelang, 2017-03-16:
did this answer your question? or are there still problems?

andrestrauss, 2017-03-16:
Hi david
you have helped me a lotm but I cant get it to work properly. I basically want the status in OpenHab to change if I manually toggle the relay with a button. At the moment, the OpenHab switch owrks fine, but I cannot update the status

See Items file:
//Sonoff Basic

Switch Light1 "Living Room Light" (LR,gLight){ mqtt="<[mosquitto:stat/sonoff2/powe:state:default]" }

String Test1 "State [%s]" { mqtt="<[mosquitto:stat/sonoff2/power:state]" }

Switch sw03 "LR Light" { mqtt=">[mosquitto:cmnd/sonoff2/POWER:command:OFF:off],>[mosquitto:cmnd/sonoff2/POWER:command:ON:on],
<[mosquitto:stat/sonoff2/POWER:state:MAP(binary.map)]", autoupdate="false" }

// Sonoff Pow

Switch BA_TV "Home Entertainment" { mqtt=">[mosquitto:cmnd/sonoffpow1/power:command:*:default],<[mosquitto:stat/sonoffpow1/POWER:state:default]" }

Number BA_TV_Power "Home Entertainment Power [%.1f W]" { mqtt="<[mosquitto:tele/sonoffpow1/ENERGY:state:JSONPATH($.Power)]" }

and Sitemaps File
sitemap default label="My first sitemap"
{
    Switch item=Light1 label="Light11 Switch"
    Text item=BA_TV_Power
    Switch item=BA_TV label="Home Entertainment Switch"
    Switch item=sw03

 }

And screenshot
http://prntscr.com/ektkcj

The relay is currently switched off, and I need the UI to show an inactive switch

davidelang, 2017-03-17:
you have a lot of things here, what happens when you switch the module? can you 
watch the mqtt bus and verify that the stat message is sent as expected? (I use 
the mqtt dashboard on my android phone for this purpose)

ThomDietrich, 2017-03-19:
Did you look at https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB ?

davidelang, 2017-03-27:
are you still having issues with this?

closed 2017-04-02 by davidelang