arendst/Tasmota #194: Sonoff in an in-line light switch?

oenc, 2017-03-12:
Hi, not sure how to mark this as "help wanted" so apologies, and it might be more an electrician question.

I've been trying to connect the sonoff basic into a circuit with my light switch as shown in the picture: 
[image]

This is because my wife isn't keen on using her phone etc.. to control the light, so I want the switch to work as normal (its a press once to turn on / off switch). 

I haven't yet attached the GPIO's in this setup as I wanted to test it first before soldering.

The sonoff connects to wifi and I can access it in the browser to config etc... however when it is "off" the light bulb is flickering very dimly, but maybe about 2x per second.  When I click "toggle" to turn the sonoff on (or use the button on the sonoff) the light bulb comes on, but then immediately the sonoff turns off and so does the bulb.  The wifi connection is maintained and the webpage shows ON, toggle again and the same thing happens but the page changes to OFF.

I'm guessing maybe it doesn't like the sonoff being in the switch circuit and I should have put it between mains and light instead.

For info: I've flashed your firmware successfully and tested a similar concept on a kettle lead with no issues (including a light switch on the GPIOs).

Any thoughts?

Cheers

Owen

davidelang, 2017-03-13:
you are showing that the line output of the sonoff is shorted out, did you mean 
that the lamp is there?

if you take the lamp switch and re-wire things so that the switch is not 
connected to line voltage at all. If you wire it to GPIO4 and ground, you should 
be able to hook it up and set that gpio to switchmode follow things should work 
the way you want them to.

The fact that it's loosing power when you turn it on indicates that you have 
something wrong in your basic wiring

David Lang

oenc, 2017-03-13:
Hi David, 

(forgetting about the GPIO for the moment as that does work in other circuits I've done)

Yes, its a shorted connection which was supposed to simulate a closed light switch, essentially replacing the light switch completely with the sonoff.  Even putting the original light toggle switch where the shorted connection is has the same behaviour (when the toggle is closed) so I guess I'm putting the sonoff in the wrong part of the circuit.

I'll keep looking around, 

Thanks

davidelang, 2017-03-13:
if you short the output of the sonoff, you will pop your circuit breaker in your 
house. It's just like putting a short across an outlet.

you wouldn't put a switch across the wires of an extension cord would you?

David Lang

On Mon, 13 Mar 2017, oenc wrote:

> (forgetting about the GPIO for the moment as that does work in other circuits I've done)
>
> Yes, its a shorted connection which was supposed to simulate a closed light switch, essentially replacing [...]

arendst, 2017-03-13:
It does turn off all lights though...

oenc, 2017-03-13:
Thanks guys, it wasn't popping the fuse box or anything, just immediately switching the light off.  I think I was struggling to understand the difference between how when the lightswitch completes the circuit it is fine, but when the lightswitch is replaced with the sonoff which completes the circuit in what seems like the same way it is a short.  

I'll try it further up the circuit near the bulb itself instead of the switch.

Cheers

closed 2017-03-13 by oenc

davidelang, 2017-03-13:
If you are trying to put the sonoff in place of the existing switch, that won't 
work.

It sounds like you are trying to do

prong1 --- sonoff --- light ---- prong2

what you need to do is

prong1 --- sonoff line ----\
                           light
prong2 --- sonoff neutral--/

David Lang

On Mon, 13 Mar 2017, oenc wrote:

> Thanks guys, it wasn't popping the fuse box or anything, just immediately switching the light off.  I think I was struggling to understand the difference between how when the lightswitch completes [...]