arendst/Tasmota #195: Strange behaviour with COMMANDS1-4 repeating on every POWER1/2 switch

mihalski, 2017-03-13:
This is with Switch1 on GPIO2 and Switch2 on GPIO14.
```
00:00:07 Wifi: Connected
00:00:07 mDNS: Initialized
00:00:07 HTTP: Webserver active on upstairs-1557.local with IP address 192.168.1.93
13:07:23 MQTT: Attempting connection...
13:07:25 mDNS: Query done with 1 mqtt services found
13:07:25 mDNS: Service found on homepi ip 192.168.1.64 port 1883
13:07:25 MQTT: Connected
13:07:25 MQTT: tele/upstairs/LWT = Online (retained)
13:07:25 MQTT: cmnd/upstairs/POWER = 
13:07:25 MQTT: tele/upstairs/INFO1 = {"Module":"WeMos D1 mini", "Version":"4.0.4", "FallbackTopic":"DVES_040615", "GroupTopic":"sonoffs"}
13:07:25 MQTT: tele/upstairs/INFO2 = {"WebserverMode":"Admin", "Hostname":"upstairs-1557", "IPaddress":"192.168.1.93"}
13:07:25 MQTT: tele/upstairs/INFO3 = {"Started":"External System"}
13:07:33 MQTT: tele/upstairs/STATE = {"Time":"2017-03-13T13:07:33", "Uptime":0, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":74}}
13:07:42 MQTT: cmnd/upstairs/POWER1 = OFF
13:07:42 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
13:07:42 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
13:07:42 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
13:07:42 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
13:07:42 MQTT: cmnd/upstairs/POWER1 = ON
13:07:42 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
13:07:42 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
13:07:43 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
13:07:43 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
13:09:51 MQTT: cmnd/upstairs/POWER1 = OFF
13:09:51 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
13:09:51 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
13:09:51 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
13:09:52 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
13:09:52 MQTT: cmnd/upstairs/POWER1 = ON
13:09:52 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
13:09:52 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
13:09:52 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
13:09:52 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
13:10:34 MQTT: cmnd/upstairs/POWER1 = OFF
13:10:34 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
13:10:34 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
13:10:35 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
13:10:35 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
13:10:40 MQTT: cmnd/upstairs/POWER1 = ON
13:10:40 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
13:10:40 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
13:10:40 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
13:10:40 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
13:11:23 MQTT: cmnd/upstairs/POWER1 = OFF
13:11:23 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
13:11:23 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
13:11:23 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
13:11:23 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
13:12:33 MQTT: tele/upstairs/STATE = {"Time":"2017-03-13T13:12:33", "Uptime":0, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":80}}
```
Any ideas what could be causing this?

arendst, 2017-03-13:
Lack of relays.

arendst, 2017-03-14:
While finding the `New issue` button doesn't seem to be a problem for some people demonstrating the use of the `Close issue` button is also highly appreciated be me and others...

mihalski, 2017-03-14:
I'm not sure what the benefit of reporting the commands each time there is a switch toggle when there is a lack of relays, but OK.

Also, I've never closed an issue because I didn't realise it was my place to do so. Lesson learnt.

closed 2017-03-14 by mihalski