arendst/Tasmota #197: Has anyone tried connecting two PIR's to a Wemos D1 mini?

mihalski, 2017-03-13:
I've got two SR-HC501's facing in opposite directions and connected to GPIO2 and GPIO14 and powered by 5v going nuts and I can't figure out what the hell is going on. I don't have this problem on other Wemos D1 pro's that only have one PIR connected.

```
19:37:57 MQTT: cmnd/upstairs/POWER1 = ON
19:37:57 MQTT: cmnd/upstairs/POWER2 = ON
19:38:00 MQTT: cmnd/upstairs/POWER1 = OFF
19:38:02 MQTT: cmnd/upstairs/POWER2 = OFF
19:38:05 MQTT: cmnd/upstairs/POWER1 = ON
19:38:07 MQTT: cmnd/upstairs/POWER2 = ON
19:38:08 MQTT: cmnd/upstairs/POWER1 = OFF
19:38:11 MQTT: cmnd/upstairs/POWER2 = OFF
19:38:16 MQTT: cmnd/upstairs/POWER1 = ON
19:38:16 MQTT: cmnd/upstairs/POWER2 = ON
19:38:19 MQTT: cmnd/upstairs/POWER1 = OFF
19:38:21 MQTT: cmnd/upstairs/POWER2 = OFF
19:38:25 MQTT: cmnd/upstairs/POWER1 = ON
19:38:26 MQTT: cmnd/upstairs/POWER2 = ON
19:38:27 MQTT: cmnd/upstairs/POWER1 = OFF
19:38:31 MQTT: cmnd/upstairs/POWER2 = OFF
19:38:35 MQTT: cmnd/upstairs/POWER1 = ON
19:38:35 MQTT: cmnd/upstairs/POWER2 = ON
19:38:37 MQTT: cmnd/upstairs/POWER1 = OFF
19:38:39 MQTT: cmnd/upstairs/POWER2 = OFF
19:38:42 MQTT: cmnd/upstairs/POWER2 = ON
19:38:43 MQTT: cmnd/upstairs/POWER1 = ON
19:38:46 MQTT: cmnd/upstairs/POWER1 = OFF
19:38:46 MQTT: cmnd/upstairs/POWER2 = OFF
19:38:54 MQTT: cmnd/upstairs/POWER1 = ON
19:38:54 MQTT: cmnd/upstairs/POWER2 = ON
19:38:57 MQTT: cmnd/upstairs/POWER1 = OFF
19:38:58 MQTT: cmnd/upstairs/POWER2 = OFF
19:39:21 MQTT: cmnd/upstairs/POWER1 = ON
19:39:21 MQTT: cmnd/upstairs/POWER2 = ON
19:39:24 MQTT: cmnd/upstairs/POWER1 = OFF
19:39:26 MQTT: cmnd/upstairs/POWER2 = OFF
19:39:30 MQTT: cmnd/upstairs/POWER1 = ON
19:39:30 MQTT: cmnd/upstairs/POWER2 = ON
19:39:32 MQTT: cmnd/upstairs/POWER1 = OFF
19:39:34 MQTT: cmnd/upstairs/POWER2 = OFF
19:39:39 MQTT: cmnd/upstairs/POWER1 = ON
19:39:39 MQTT: cmnd/upstairs/POWER2 = ON
19:39:42 MQTT: cmnd/upstairs/POWER1 = OFF
19:39:52 MQTT: cmnd/upstairs/POWER2 = OFF
19:39:56 MQTT: cmnd/upstairs/POWER1 = ON
19:39:56 MQTT: cmnd/upstairs/POWER2 = ON
19:39:59 MQTT: cmnd/upstairs/POWER1 = OFF
19:40:01 MQTT: cmnd/upstairs/POWER2 = OFF
19:40:05 MQTT: cmnd/upstairs/POWER1 = ON
19:40:05 MQTT: cmnd/upstairs/POWER2 = ON
19:40:07 MQTT: cmnd/upstairs/POWER1 = OFF
19:40:09 MQTT: cmnd/upstairs/POWER2 = OFF
19:40:12 MQTT: cmnd/upstairs/POWER2 = ON
19:40:14 MQTT: cmnd/upstairs/POWER1 = ON
19:40:16 MQTT: cmnd/upstairs/POWER1 = OFF
19:40:16 MQTT: cmnd/upstairs/POWER2 = OFF
19:40:20 MQTT: cmnd/upstairs/POWER2 = ON
19:40:23 MQTT: cmnd/upstairs/POWER1 = ON
19:40:25 MQTT: cmnd/upstairs/POWER2 = OFF
19:40:25 MQTT: cmnd/upstairs/POWER1 = OFF
19:40:28 MQTT: cmnd/upstairs/POWER2 = ON
19:40:32 MQTT: cmnd/upstairs/POWER1 = ON
19:40:32 MQTT: cmnd/upstairs/POWER2 = OFF
19:40:35 MQTT: cmnd/upstairs/POWER1 = OFF
19:40:35 MQTT: cmnd/upstairs/POWER2 = ON
19:40:40 MQTT: cmnd/upstairs/POWER2 = OFF
19:40:47 MQTT: cmnd/upstairs/POWER1 = ON
19:40:47 MQTT: cmnd/upstairs/POWER2 = ON
19:40:50 MQTT: cmnd/upstairs/POWER1 = OFF
19:40:51 MQTT: cmnd/upstairs/POWER2 = OFF
```
It shows no sign of stabilising even 10 minutes later.

davidelang, 2017-03-13:
do you have pull-up (or pull-down as appropriate) resisters on these 
pins?

mihalski, 2017-03-13:
The HC-SR501 PIR module does not need these are all the necessary circuitry is already on the module.

Three other Wemos modules with one PIR attached are working fine.

davidelang, 2017-03-13:
> The HC-SR501 PIR module does not need these are all the necessary circuitry is already on the module.

well, if the connection to the cpu is drifting between on and off, there 
probably isn't a pull-up/down resister in place, even if the diagrams say there 
should be.

the circuit on the CPU is simple, it checks if the voltage on the pin is above a 
given figure, if so it returns a 1, if not it returns a 0. The only reason for 
it to drift from one to the other is if the voltage is drifting, and that is 
exactly what happens if it is not connected to anything (i.e. high impedence, 
not pulled either up or down)

the other possibility is if you have a pull-up on the sonoff side and a 
pull-down on the PIR side cancelling each other out.

> Three other Wemos modules with one PIR attached are working fine.

connected to the same pins?

We have seen some reports of devices with manufacturing defects resulting in 
cold solder joints that make them a bit unstable.

mihalski, 2017-03-13:
The others are all connected to GPIO14 which has no pull-up or pull-down resistors.

One of the PIRs is connected to GPIO14 while the other is connected to GPIO2 which also has a 10k Pull-up and an led connected.

I'll be delving deeper into the problem in a couple of days when I have a bit more time. I'll post here if I identify the problem.

mihalski, 2017-03-14:
Are the IO pins (GPIO2 and GPIO14) supposed to be outputting 3.3v all the time? Because that's what their doing. They're set to switch1 and switch2.

davidelang, 2017-03-14:
On Tue, 14 Mar 2017, mihalski wrote:

> Are the IO pins (GPIO2 and GPIO14) supposed to be outputting 3.3v all the time? Because that's what their doing. They're set to switch1 and switch2.

If they have pull-up resisters (which I have read is how they are configured), 
then yes, they will show 3.3v all the time, and you can switch them to ground 
with a simple switch.

what do the output pins os your PIRs show?

mihalski, 2017-03-14:
I have 2 identical Wemos and 3 identical PIRs (I have more of both).
The PIRs are fine. By themselves they're 3.3v when active and 0v when inactive.
On the working Wemos GPIO14 is 0v with no resistors. On the other Wemos using the same GPIO14 with no resistors I get 3.3v with nothing even connected. Not only that but GPIO2 with a 10k pull-up resistor is also 3.3v.

I'm going to see if reconfiguring switches makes any difference to this. And I'm going to grab a 3rd Wemos to check.

davidelang, 2017-03-14:
On Tue, 14 Mar 2017, mihalski wrote:

> I have 2 identical Wemos and 3 identical PIRs (I have more of both).
> The PIRs are fine. By themselves they're 3.3v when active and 0v when inactive.
> On the working Wemos GPIO14 is 0v with no [...]

in sonoff.ino

line 1983 sets GPIO_KEY* to have pullup resistors configured

line 1993 sets GPIO_SWT* to have pullup resistors configured

check your firmware versions on the wemos, it may be old that had something 
configured differently.

David Lang

mihalski, 2017-03-14:
I'm just trying to get the new wemos setup but for some reason it has defaulted to the "indebuurt1" SSID and is ignoring any changes I make in the user_config.h.. Could it be caching the firmware rather than rebuilding it using the changes?

mihalski, 2017-03-14:
Changing CFG_HOLDER and re-uploading fixed it.

mihalski, 2017-03-14:
Interestingly without any additional resistors GPIO14 works perfectly.. With a 10k pull-up resistor GPIO2 behaves strangely. It's a bit jittery and high is 3.3v as it should be but low is 0.6v.

mihalski, 2017-03-14:
Yep.. I think it only works without any resistors. I tried GPIO12 and it works..

davidelang, 2017-03-14:
On Tue, 14 Mar 2017, mihalski wrote:

> I'm just trying to get the new wemos setup but for some reason it has defaulted to the "indebuurt1" SSID and is ignoring any changes I make in the user_config.h.. Could it be caching the firmware rather than rebuilding it using the changes?

hold down the button for four seconds to force it to overwite the config from a 
prior image with what's in the user_config.h

or you can change CFG_HOLDER in user_config.h to make the new firmware image 
consider any prior config corrupt.

since the common case is expected to be updating an image, it tries to keep as 
much as possible of the old config so that you don't have to reconfigure 
everything every time you update. If CFG_HOLDER in the saved config matches 
what's compiled, it assumes that it should use the old config.

holding the button down for 4 seconds tells it to overwrite the config with 
what was compiled in from user_config.h

changing the CFG_HOLDER value makes it consider the config corrupt/missing and 
causes it o write the user_config.h values to the config.

David Lang

mihalski, 2017-03-14:
I'm at a loss.. The only thing I can think of is that the Weomos is somehow fried and the IO is behaving erratically.

davidelang, 2017-03-14:
I saw one mailing list thread where the person found that there was a capacitor 
that had a bad solder joint and that was causing similar odd behavior

On Tue, 14 Mar 2017, mihalski wrote:

> I'm at a loss.. The only thing I can think of is that the Weomos is somehow fried and the IO is behaving erratically.

mihalski, 2017-03-14:
I'm actually getting 4.3v on one of the GPIO's now.. That can't be good.

davidelang, 2017-03-14:
sounds like a bad power supply on that module, what is vcc going to the chip?

On Tue, 14 Mar 2017, mihalski wrote:

> I'm actually getting 4.3v on one of the GPIO's now.. That can't be good.

mihalski, 2017-03-14:
3.3v to the chip. 4.7v available from USB. 3.3v on the GPIO's with nothing plugged in. I'm starting to wonder if the problem is 2 PIR's connected at once..

davidelang, 2017-03-14:
are you sure the PIRs are 3.3v, not 5v based systems?

On Tue, 14 Mar 2017, mihalski wrote:

> 3.3v to the chip. 4.7v available from USB. 3.3v on the GPIO's with nothing plugged in. I'm starting to wonder if the problem is 2 PIR's connected at once..

mihalski, 2017-03-14:
They're can be powered by 4.5-20v but they're 3.3v based. There's a trick to power them with 3.3v but since the Wemos have the 5v broken out there is no need to do so.

davidelang, 2017-03-14:
On Tue, 14 Mar 2017, mihalski wrote:

> They're can be powered by 4.5-20v but they're 3.3v based. There's a trick to power them with 3.3v but since the Wemos have the 5v broken out there is no need to do so.

if they're being powered by 5v, are the pull-up resisters on the PIRs pulling up 
to 5v?

mihalski, 2017-03-14:
https://www.mysensors.org/dl/57c41fdd4d04abe84cd93e12/design/31227sc.pdf

mihalski, 2017-03-14:
I think one of my problems may be my crappy Dupont crimping tool. Unreliable connections might be an issue.

mihalski, 2017-03-14:
Yes.. Seems that inexplicably one of the wires/connectors/crimps doesn't conduct electricity. The one that carries the signal from one of the PIR's. I had tested it for continuity but I guess it was working intermittently at the time. I'm going to order myself a better crimping tool right now.

Hopefully conductivity will help this work as expected.

davidelang, 2017-03-16:
since it looks like you found the problem, can this issue be closed now?

mihalski, 2017-03-16:
I had time to rewire and rebuild the whole thing tonight but I'm still seeing the constant triggering. To make it worse another unit which I updated too the latest version a couple of days ago is now exhibiting the same problem and it has just one PIR connected.

So I'm still trying to figure it out. I don't know what/where the problem is for sure.

mihalski, 2017-03-21:
For the time being I am closing this. I've ordered various models of PIRs and will create a new issue if I can't get multiple PIRs to behave in the same manner as one can.

closed 2017-03-21 by mihalski