arendst/Tasmota #198: Wall Switch Flickering Issue

scottdarby93, 2017-03-13:
Hello, First off I have to say thankyou for amazing software for the sonoff, half of my house is now controlled by your software!

However I am having a problem when using in-line switches, I am finding that when I connect up the switch to GPIO14 and GND every now and again the lamp I am controlling will come on and go off straight away, I am unsure if this is due to my wiring on the sonoff or if it is that sensitive that it detects even a slight drop?
Would love to know if anyone else has had this problem.

I was thinking would the sonoff be able to wait ever so slightly when it detects the connection change to make sure it is a genuine toggle? Just a thought.

davidelang, 2017-03-13:
On Mon, 13 Mar 2017, scottdarby93 wrote:

> Hello, First off I have to say thankyou for amazing software for the sonoff, half of my house is now controlled by your software!
>
> However I am having a problem when using in-line switches, I am [...]

Even if the switch bounced, it would turn on, go off, and then turn on again. 
That doesn't seem like it's likely to cause the problem (unless tasmota is 
depending on interrupts as opposed to checking the state of the switch and an 
interrupt is being lost)

hamwong, 2017-03-15:
I have same issue, while place a 10K resistor at PGIO14 end, flicking will lot lower, but still around once every 5 mins, finally I change to MQTT mode and problem mostly solved and it's too quick for OH than back to sonoff to response in action

davidelang, 2017-03-18:
Are you still having problems with this?

rk125, 2017-03-19:
I had similar issue, was trying resistor to VCC, but did not help. Ended up putting 1uF tiny tantalum capacitor from switch pin to ground. Do not experience unexpected on/offs. Btw, have long lines between wall switches (reused 220V wires for ESP switching) in some places - no unexpected on/offs there.

davidelang, 2017-03-27:
It sounds as if the problem has been identified. Please re-open this issue if you are still having problems.

closed 2017-03-27 by davidelang