arendst/Tasmota #200: Button multipress

ionciubotaru, 2017-03-14:
When the button is press one time the relay toggle and a MQTT message is published
When the button is pressed two times sonoff do the same. (Except Sonoff Dual)

Sugestion:

```
    if (multipress != 2 or sysCfg.module == SONOFF_DUAL) {
      switch (state) {
      case 0: { // Off
        power &= (0xFF ^ mask);
        break; }
      case 1: // On
        power |= mask;
        break;
      case 2: // Toggle
        power ^= mask;
      }
      setRelay(power);
    } else {
//  publish a nice mqtt message
    }
```

Can you add this in the future release ?
I use this code to control a second less accesibile Sonoff from a Sonoff Touch.

arendst, 2017-03-14:
When the button is pressed once AND buttontopic is set it will execute a MQTT command. In that case a
button press twice will execute the command that otherwise was executed when button was pressed once.

See the cryptic message in wiki - Button usage (https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage).

crazynet24, 2017-03-14:
I also would appreciate ionciubotaru's suggestion. I've test several times the buttontopic/switchtopic/switchmode command's, but I don't get it. I'm not able figuring out what the parameters are doing. I'm also looking for a possibility, to switch a fan in the bathroom, by tapping the sonoff touch twice.
For me, unfortunately, the commands mentioned are to confusing. I don't know what's the difference between switch and button(-topic), and also the parameters toggle/follow. 

But Theo,  Tasmota is the most fascinating development for ESP's at the moment. Thank you very much for your project!

ionciubotaru, 2017-03-14:
Yes, Tasmota is the best software for Sonoff. Looking inside the code I'm impressed of Theo's programming skills..

I set buttontopic to 'second_sonoff' and when the button is pressed once sonoff send the mqtt message:
`cmnd/second_sonoff/POWER TOGGLE`
When the button is pressed twice sonoff send the status mqtt message and toggle the relay.

Is it possible to swap ?
- when the button is pressed once to toggle the relay
- when the button is pressed twice to send the mqtt message to buttontopic

davidelang, 2017-03-16:
you have run across a bug that's due to the special casing of the first button. The "correct" behavior afte you have set buttontopic is to not change any local relays, only to send the mqtt messages

ionciubotaru, 2017-03-16:
For all Sonoffs, except Touch this behavior is ok, but for Touch is better to turn on the light on the first press - is hard to teach all family members to toggle light with double click, when for many year they did it with a single one :)

davidelang, 2017-03-16:
the fixed code would not toggle the relays at all with switchtopic set, it would send one message if you single-tap and a different message if you doubletap

The idea is that once you set the switchtopic, it should no longer control the relays directly, only send messages

ionciubotaru, 2017-03-16:
It's OK if everything works fine (wireless + MQTT), but if something is wrong nobody can turn on the light and this will be frustrating in a smart house (bad thinks always happened - Murphy's law)

davidelang, 2017-03-16:
well, if you don't want the button to not turn on the local relays, why are you setting switchtopic? that's what it does.

ionciubotaru, 2017-03-16:
I'm sorry `davidelang` but switchtopic has no effect on double click
Only ButtonTopic has effect, but in a reverse order:

Excuse me for insitence, please tell me how to do:
- when the button is pressed once to toggle the relay
- when the button is pressed twice to send the mqtt message to buttontopic

If this is not possible it's ok, i will change the code manually

davidelang, 2017-03-17:
On Thu, 16 Mar 2017, Ion wrote:

> I'm sorry `davidelang` but switchtopic has no effect on double click
> Only ButtonTopic has effect, but in a reverse order:

that's what I've been saying, and it's a bug

> Excuse me for insitence, please tell me how to do:
> - when the button is pressed once to toggle the relay
> - when the button is pressed twice to send the mqtt message to buttontopic
>
> If this is not possible it's ok, i will change the code manually

This is not possible in the current code.

davidelang, 2017-03-19:
Ok, after re-reading this umpteen times I think I understand what @arendst  was saying way up above.

double tapping button0 will do one of two things.

If you are on a sonoff dual, it will switch the second relay

If you are on anything else, it will switch the first relay (as long as it's not in the middle of connecting to a mqtt server), this provides a back-door to switch the first relay if you are disconnected from the network and have buttontopic set.

At this point, I think it would be less confusing to have the double-tap do nothing on devices other than the dual, or always act as if button2 was pressed (no matter what type). But I suppose legacy consistency applies here.

crazynet24, 2017-03-19:
I also got this issue like you. At the moment, it is only a workaroud to archieve a double tap function, and it's really confusing. Furthermore I would appreciate a more easier way to control other things by tapping twice a sonoff touch.

davidelang, 2017-03-19:
@arendst

would it make sense to change it so that the double-tap always did what button2 
is configured for? (either switch relay 2 or send the power2 mqtt command 
depending on buttontopic)

arendst, 2017-03-19:
@davidelang if I make that change switching won't work anymore when no MQTT server can be reached. That is what you explained just above correctly. That was also the reason of having the double tap available.

I do not see a relation between a sonoff touch double tapping and a sonoff dual.

Changing it's functionality makes sonoffs useless when there is no MQTT server connection.

I need more convincing...

davidelang, 2017-03-19:
well, I think it's wrong that if you have buttontopic set, any manipulation of 
the button changes the local relay :-)

I see two logical config scenarios, and both are valid

1. you set buttontopic and want the local relays to do nothing if there isn't a 
command from a mqtt server.

2. you set buttontopic but want a fallback to local control if the mqtt server 
is not reachable.

If you want #2, you really want it for all buttons (including the sonoff dual's 
'fake' button from the doubletap)

If you want #1, it's not possible right now, there's always a backdoor for the 
local relay (except for on the dual)

So I think it makes more sense to have this as a config option 
(buttonfallback??) than to have it be just a feature of a single-button/relay 
device.

remember that tasmota is being used more and more with devices that have 
multiple switches/relays on them, the stock single-relay sonoffs are still the 
common case, but far from the only case.

joshuagti, 2017-03-21:
I'd like to do something similar with the Sonoff Touch.

For example, it is on the bedroom wall controlling the ceiling light. I have two bedside lights each with basic Sonoffs inside. A single press to the Touch would turn ceiling light on but two short presses would send out an MQTT message to my OpenHAB which would then turn on the lights. 

From what I am reading, it isn't possible in the current build but it should be possible to implement?

ionciubotaru, 2017-03-21:
Please take a look to the first item in this issue.
You can modify the code and Sonoff Touch will work as you wish.

rhodopsik, 2017-03-23:
Hi ion,
do you have a short explanation where to apply your code snippet? I which file/row?
This would be great to get in to it...

Thanks!

ionciubotaru, 2017-03-23:
In sonoff.ino search for the block inside if and replace it with the new one.

terba, 2017-06-02:
I would also use double tapping on the Touch to control other things (through MQTT). So it would be good to have this as an option.

ThomDietrich, 2017-06-04:
I'd also appreciate the described feature. It should be possible to have double tap only send an MQTT message while single tap is unmodified.

arendst, 2017-06-05:
By trying to get this issue solved I need to know what you want by keeping in mind the original functionality.
```
| Situation | Buttontopic | Single Press                  | Double Press                  | Description
|-----------|-------------|-------------------------------|-------------------------------|-------------------------------------
| Current   | 0           | Toggle relay                  | Ditto                         | Ditto used for Dual relay 2
|           |             |                               |                               |
|           | = topic     | Send MQTT msg:                | Toggle relay                  | Double press backup for MQTT failure
|           |             | cmnd/<topic>/POWER ON|OFF     |                               |
|           | <> topic    | Send MQTT msg:                | Toggle relay                  | Double press backup for MQTT failure
|           |             | cmnd/<non-topic>/POWER TOGGLE |                               |
|-----------|-------------|-------------------------------|-------------------------------|-------------------------------------
| Wish 1    | 0           | Toggle relay                  | Ditto                         | Ditto used for Dual relay 2
|           |             |                               |                               |
|           | = topic     | Toggle relay                  | Send MQTT msg:                | Single press for MQTT failure
|           |             |                               | cmnd/<topic>/POWER ON|OFF     |
|           | <> topic    | Toggle relay                  | Send MQTT msg:                | Single press for MQTT failure
|           |             |                               | cmnd/<non-topic>/POWER TOGGLE |
|-----------|-------------|-------------------------------|-------------------------------|-------------------------------------
```
I could make situation Wish 1 a selectable option if that solves all your problems although I expect an answer where someone wants to send a MQTT message when pressed once and another message when pressed twice bypassing any local relay control...

terba, 2017-06-05:
For me Wish 1 makes what I would like to do now.

But for a long-term solution I think it would be better to invest into a more generic solution where any combination could be configured. I think it would be easier to maintain and maybe is less LOC. There is also an issue for a Long Press Event (#489), which could be integrated as well (I would use that also). And there is SwitchMode too, which I was not able to use last time I tried. It is and gets even more complicated if you try to implement all wishes with the current model.

I think it would be more versatile and easier to understand if it could be configured somehow like this:

```
PressAction = <Nothing | RelayToggle <N> | arbitrary MQTT topic + data (with template variable for ON/OFF)>
ReleaseAction = <Nothing | RelayToggle <N> | arbitrary MQTT topic + data (with template variable for ON/OFF)>
DoubleClickAction = <Nothing | RelayToggle <N> | arbitrary MQTT topic + data (with template variable for ON/OFF)>
LongPressAction = <Nothing | RelayToggle <N> | arbitrary MQTT topic + data (with template variable for ON/OFF)>
```

Where ReleaseAction is if no immediate press follows (ie this is the SingleClickAction but can be used in Press-Release scenarios like a doorbell button).

With these 4 settings all the combinations could be achieved: default operation, MQTT failsafe mode, SwitchMode, SwitchTopic and ButtonTopic things, all the wishes and the multi-relay models too.

For example for Wish 1 I could configure this:

```
PressAction = Nothing (0)
ReleaseAction = RelayToggle 1 (1)
DoubleClickAction = "/somewhere/something/dothis asdasd"
LongPressAction = Nothing (0)
```

Default on a Dual:
```
PressAction = Nothing (0)
ReleaseAction = RelayToggle 1 (1)
DoubleClickAction = RelayToggle 2 (2)
LongPressAction = Nothing (0)
```

Or a momentary switch mode:
```
PressAction = RelayToggle 1
ReleaseAction = RelayToggle 1
DoubleClickAction = Nothing (0)
LongPressAction = Nothing (0)
```

But one can connect other switches to the GPIOs so it can be made even more generic... :)

davidelang, 2017-06-05:
I was getting reading to make such a request as I read the chart :-)

> I expect an answer where someone wants to send a MQTT message when pressed once and another message when pressed twice bypassing any local relay control...

arendst, 2017-06-06:
@terba this issue is about the **push button** on Sonoff and NOT about externally connected push buttons (configured as switches using command SwitchMode ).

In version 5.1.6 I will implement hold button functionality for both the **push button** AND the external pushbutton. If a ButtonTopic (and if SetOption1 (=ButtonRestrict) is On) or SwitchTopic (and SwitchMode 5|6) has been defined and a button is pressed longer than define KEY_HOLD_TIME (in 0.1 seconds. default 4 seconds) a MQTT message like ``cmnd/sonoff/POWER HOLD`` will be send. The HOLD text can be changed with command StateText4.

Another command (SetOption11 On|Off) allows for swapping the functionality of the **push button** as mentioned before as Wish 1.  

These changes result in below table
```
| Set     | Set      | Button   |                               |                                |                             |                                      |
| Option1 | Option11 | Topic    | Single Press                  | Double Press                   | Press Hold                  | Description                          |
|---------|----------|----------|-------------------------------|--------------------------------|-----------------------------|--------------------------------------|
| 0       | 0        | 0        | Toggle relay                  | Ditto                          | Reset 1                     | Ditto used for Dual relay 2          |
|         |          |          |                               |                                |                             |                                      |
|         |          | = topic  | Send MQTT msg:                | Toggle relay                   | Reset 1                     | Double press backup for MQTT failure |
|         |          |          | cmnd/<topic>/POWER ON|OFF     |                                |                             |                                      |
|         |          | <> topic | Send MQTT msg:                | Toggle relay                   | Reset 1                     | Double press backup for MQTT failure |
|         |          |          | cmnd/<non-topic>/POWER TOGGLE |                                |                             |                                      |
|---------|----------|----------|-------------------------------|--------------------------------|-----------------------------|--------------------------------------|
| 1       | 0        | 0        | Toggle relay                  | Ditto                          | Discarded                   | Ditto used for Dual relay 2          |
|         |          |          |                               |                                |                             |                                      |
|         |          | = topic  | Send MQTT msg:                | Toggle relay                   | Send MQTT msg:              | Double press backup for MQTT failure |
|         |          |          | cmnd/<topic>/POWER ON|OFF     |                                | cmnd/<topic>/POWER HOLD     |                                      |
|         |          | <> topic | Send MQTT msg:                | Toggle relay                   | Send MQTT msg:              | Double press backup for MQTT failure |
|         |          |          | cmnd/<non-topic>/POWER TOGGLE |                                | cmnd/<non-topic>/POWER HOLD |                                      |
|---------|----------|----------|-------------------------------|--------------------------------|-----------------------------|--------------------------------------|
| 0       | 1        | 0        | Toggle relay                  | Ditto                          | Reset 1                     | Ditto used for Dual relay 2          |
|         |          |          |                               |                                |                             |                                      |
|         |          | = topic  | Toggle relay                  | Send MQTT msg:                 | Reset 1                     | Single press for MQTT failure        |
|         |          |          |                               | cmnd/<topic>/POWER ON|OFF      |                             |                                      |
|         |          | <> topic | Toggle relay                  | Send MQTT msg:                 | Reset 1                     | Single press for MQTT failure        |
|         |          |          |                               | cmnd/<non-topic>/POWER TOGGLE  |                             |                                      |
|---------|----------|----------|-------------------------------|--------------------------------|-----------------------------|--------------------------------------|
| 1       | 1        | 0        | Toggle relay                  | Ditto                          | Discarded                   | Ditto used for Dual relay 2          |
|         |          |          |                               |                                |                             |                                      |
|         |          | = topic  | Toggle relay                  | Send MQTT msg:                 | Send MQTT msg:              | Single press for MQTT failure        |
|         |          |          |                               | cmnd/<topic>/POWER ON|OFF      | cmnd/<topic>/POWER HOLD     |                                      |
[... cut ...]
|         |          |          |                               | cmnd/<non-topic>/POWER TOGGLE  | cmnd/<non-topic>/POWER HOLD |                                      |
|---------|----------|----------|-------------------------------|--------------------------------|-----------------------------|--------------------------------------|
```

For the external (Push)Button the command SwitchMode has been extended with options 5 and 6 providing both PushButtonRelease and PushButtonHold (or Inverted) functionality. SwitchMode options 3 and 4 stay the same as being PushButtonPressed (or inverted) functionality.

terba, 2017-06-12:
@arendst: Yes, I know it is about the "internal" push button not the external ones, but I tried to combine the features of the two to make it even more standardized and easily understandable and programmable way. But never mind, it was just a quick idea, may be worth considering before a major release in the future.

I flashed the new release on all my sonoffs, and configured single, double click and hold on some of them and it works like a charm. Thank you very much!

The only problem I saw is that the default 4 seconds for hold is a lot. I flashed them with 2 seconds initially, but it is also a bit too much. So I was searching for the MQTT command to alter this value, but I didn't found a command for that. Checked in the source, I saw it is hardcoded (defined). Not a big problem, I will reflash them later, but I think this should be configurable.

arendst, 2017-06-12:
While writing the code I think: How long will it take that someone wants to be able to change the hold timeout online instead of using the default of 4 seconds. I guessed it has always been there as being 4 seconds so a define would have been a great improvement...

Any way, in the next release there will be a command to change its value from 0.1 to 20 seconds ;-)

TRIROG, 2017-06-15:
How can i enable this on Touch?

terba, 2017-06-16:
@TRIROG: Use the table above to find the required settings. I assume you are using MQTT. Configure your automation software to listen to "cmnd/<topic or buttontopic>/POWER" topic with a payload "HOLD". This is sent in case of long press.

TRIROG, 2017-06-16:
I figured that much, but i can't seem to find the correct way to do it.
Would you care to tell me which command i should change?

terba, 2017-06-16:
@TRIROG: Something like that:
```
Topic livingroom_ceiling
ButtonTopic 1
SetOption11 1
SetOption1 1
```

This is the second variation in the last line of the table:

- Single click: toggles the relay (and sends a status message)
- Double click: send a command to toggle the relay (and sends a status message after the relay is toggled): `cmnd/livingroom_ceiling/POWER ON|OFF`
- Long press: sends a command message to the same topic but with a payload "HOLD": `cmnd/livingroom_ceiling/POWER HOLD`

TRIROG, 2017-06-17:
Is there any way to give feedback to user on Sonoff Touch that the HOLD event was triggered? Maybe blink a led or something?

PeggyFree, 2017-06-20:
Hello,

Thanks for all your work @arendst ! Can you tell me how you coded the multipress and longpress in domoticz mqtt messages? I would like to suggest simulating a selector switch using levels, just like the xiaomi button works : 

10 = single 
20 = double
30 = hold
40 = release

This would give a MQTT message like this (this is an example, I need to check tonight the right syntax)
```
command:switchlight
idx:145
switchcmd:Set Level 40
```

closed 2017-08-31 by ionciubotaru

davidelang, 2017-08-31:
if you set buttontopic, by default one press will send the remote message, two 
presses will change the local relay as if buttontopic wasn't set.

I believe that this is now configurable.

ionciubotaru, 2017-08-31:
`SetOption11`  Show current button single and double press swap state

RobertSzabo, 2017-11-12:
Hi,

Can the button configuration options (table) work with T1 with more than 1 switches, i.e., configuring both options and topics individually for 2 or 3 switches?

Say, with a double tap (or long press), I would like to have a delayed toggle; is it possible to configure?

Thanks,
/r

davidelang, 2017-11-13:
On Sun, 12 Nov 2017, Robert wrote:

> Can the button configuration options (table) work with T1 with more than 1 switches, i.e., configuring both options and topics individually for 2 or 3 switches?

no buttontopic works for all buttons, switchtopic affects all switches.

multipress only works on the first button, it was created to work around the 
hardware limit of only one button on the stock sonoff devices.
k

ksya, 2017-12-13:
I have a T1 3ch and have it configures like this:

```
ButtonTopic othertopic
SetOption11 1
SetOption1 1
```
Now, I get "ON/OFF" when I both press and doublepress the button. If I set SetOption11 to 0, I get for both single and doublepress "TOGGLE". This is not expected. What can be an affecting option?

Aa26300, 2018-02-15:
Hello,

I'm just starting to use the sonoff.
And I find great what you do
I loaded the last version on sonoff basic.
I would like if it is possible to use a sonoff basic from the button to order a second (or more) sonoff basic.
After I want to connect a switch to the first Sonoff, a ligth to the second switch and use the first sonoff to remote the second.
I have done several tests but not conclusive for the moment
I don't know if I was very clear

Thank

pmarq2008, 2018-03-01:
Can this be used on one of the GPIO's set as a button? I have tried it but it doesnt seem to work.

cooper2751, 2018-03-03:
hello could some one answer me what i hope is a simple question. in the table for button multi presses. lets take the last 3 row option 1 and 11 are both set to on, button topic is set to what i require it to be called (button2) but how do you change between the different results, ie if i double press i get cmnd/button2/power = toggle and i would like cmnd/button2/power =on/off   .   
its the only bit of that table i can get my head around.. 
many thanks

mattpurnell, 2018-03-17:
What does "reset 1" mean, within the actions table, under "Press Hold"?

Alestrix, 2018-06-18:
Is it possible to distinguish between different kinds of button presses of the 433 MHz button of the Sonoff RF? This way I could control a second Sonoff (Basic) from the 433 MHz button of the Sonoff RF.