arendst/Tasmota #201: Flash write count

drbent15, 2017-03-14:
Is the Flash write count supposed to be incrementing every time you sent a MQTT command. Because my 4.0.3  sonoff basic is.

davidelang, 2017-03-14:
It's probably updating because your mqtt commands are switching it on or off and 
it's set to remember it's state after a power outage. This requires writing to 
the flash each time you switch it.

modern flash can be written to something like 100k or 1m times, so it's not 
likely to be that big a problem (your relay is probably rated for a similar 
number of switches)

If you are worried about it powerretain and poweronstate let uou change this

  On Tue, 14 Mar 2017, drbent15 wrote:

> Is the Flash write count supposed to be incrementing every time you sent a MQTT command. Because my 4.0.3  sonoff basic is.
>
>

arendst, 2017-03-14:
Which commands do you send all the time?

drbent15, 2017-03-15:
On and off , as he suspected. Sorry for not stating it.

drbent15, 2017-03-15:
My setting are {"PowerOnState":3} , {"PowerRetain":"OFF"} I see from the wiki these are both set to default. What do I need to change ? And will it have an affect on something else? Not really worried about it but would like know the trade off for having it not update.

davidelang, 2017-03-15:
If it doesn't update the flash, then when it powers up, it has no idea if it was 
on or off before it lost power.

closed 2017-03-17 by drbent15

lulol, 2019-06-01:
Depending on the kind of use, with hundreds of writes per day you can't expect the flash to last very long even for the ones claiming 100k write cycles endurance which are the more common nowadays. In a couple of years could be well past that, and then the controller misteriously failing.

Just for reference for anyone landing here, despite being already well documented in the wiki, the related default options to look for to disable flash writes when acting on "relays" are:
<a href="https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#setoption-overview">setoption0</a> and <a href="https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration">PowerOnState</a>