arendst/Tasmota #202: Feature request: Timeswitching

f0m3, 2017-03-14:
First: This project is great! Thanks for ur work! 
It would be cool to have something like a time setting for a delayed action that is executed when the sonoff powers up. Or even maybe 2 actions with the possibility to repeat. So the sonoff could be used as a fan control with the ability to switch it on or off by wifi.

crazynet24, 2017-03-14:
You can archieve this by using mqtt an node-red, for instance.
That's the use-case I'm running at my home. 
With that combination, you can hit all cases you can imagine and much more.

davidelang, 2017-03-14:
On Tue, 14 Mar 2017, f0m3 wrote:

> First: This project is great! Thanks for ur work!
> It would be cool to have something like a time setting for a delayed action that is executed when the sonoff powers up. Or even maybe 2 actions with [...]

This sort of thing is probably better done in your central software, it can 
detect when a system starts up and then set a timer and take whatever actions 
you want

khcnz, 2017-03-14:
Agreed - the general idea with these devices is that they are "dumb" on/off switches and sensors and you control complicated logic remotely.

davidelang, 2017-03-16:
did you have any other questions on this topic, or can this issue be closed?

f0m3, 2017-03-18:
Since i am not a developer of this i don't want to be impolite... 
But i think, that it is more failsafe when u can tell a device a time to switch and maybe control the success by a server. It will switch even when the Network connection fails.

Rudoxxxx, 2017-03-19:
The problem is Sonoff dont have inside clock module.

I have The some 3 problem,  not working from server, when mobile network problem.
I will start:
1. my swimming pool circulation pump with timer and
2. ventilation heat exchanger (recuperation) with humidity probe and
3. heating with tmperature probe.
I have one idea, its cheap server in my swimming pool house, but better if Sonoff can work without stabil network.
Today I tray original China Sonoff timer, humidity and temperature relay without network.

davidelang, 2017-03-19:
The sonoff is not intended to be used when there is not a local server. It's not 
setup to be secure across the Internet (even enabling the mqtt TLS mode isn't 
really enough for that, but it's a fair start and better than most IoT things)

There is really very little space for more code without disabling stuff in this 
firmware, according to arendst, the next version includes enough features that 
if you enable them all, you can no longer to an OTA (over the air) update.

As you say, the sonoff devices do not have a clock module in them, so if they 
loose power for any reason, their clock becomes worthless until they can get the 
time from the network again. Once more a local server can trivially solve the 
problem.

note that a Raspberry Pi (with real-time clock addon) makes a perfectly good 
server, so it's not like you need something expensive.

davidelang, 2017-03-27:
I'm going to go ahead and close this, if someone disagrees, go ahead and re-open it

closed 2017-03-27 by davidelang

FeepingCreature, 2017-12-13:
I'm planning to use a S20 to implement fail-safe on a 3D printer, such that it interrupts power when an attached RPi fails to give an all-clear signal every so-and-so seconds, which seems similar.

Can I use Backlog in the current version to do this? Ie. send `Backlog Delay 600; Power 0` once per 30 seconds.