arendst/Tasmota #208: MQTT domoticz, issues.

eriktwotoe, 2017-03-15:
How do I enable domoticz MQTT publishing, it's just not doing anything whatever flag I set. only MQTT tele is sent to broker. Do I need to enable webserver also? it's been disabled no use for it. I'm using a SonOff Pow and want it to publish to domoticz energy consumption.

ionciubotaru, 2017-03-15:
Read [https://github.com/arendst/Sonoff-Tasmota/wiki/Domoticz](https://github.com/arendst/Sonoff-Tasmota/wiki/Domoticz)

Check `user_config.h` for 
`//#define BE_MINIMAL `
This line ust be a comment

eriktwotoe, 2017-03-16:
Hello, I Tried with and without comment BE_MINIMAL. 
Weird thing is it works now with webserver enabled, without webserver it doesnt work at all.
but with Webserver enabled DomoticzUpdateTimer flag does not work, it gets result set, but its like 300-400s interval, cannot change/override nothing happens. :(

davidelang, 2017-03-16:
mqtt and domoticz are different beasts, if you are using mqtt messages with domoticz, then the DomoticzUpdateTimer isn't relevant as that only applies to the domoticz protocol

if you are only seeing updates every 300 seconds, that sounds like you are looking at the tele/ messages, which are only sent periodically (defaulting to every 300 seconds), and not the stat/ messages which are sent immediatly after a change.

arendst, 2017-03-16:
Just tested with or without webserver and Domoticz works great as expected.

Stop fooling around with enabling/disabling defines in user_config.h as the default settings, apart from STA_SSID1 and STA_PASS1 are ok.

NEVER enable BE_MINIMAL and expect anything to work as it is just a shortcut for future memory problems to load a minimal interim version before a final version. I've tried to explain that in the file too but people tend to fool around first, then ask questions and finally agree they had to read first.

See wiki - Domoticz for further information.

eriktwotoe, 2017-03-16:
David Lang: There is no such thing as domoticz protocol. Ehh, I don't understand what your trying to say.

Arendst: Yes sir, never fooling around with that again, what made everything working for me, was using Reset=2 after firmware update. if someone else having the same problem. Thanks for the help , and the great coding & work you are providing.

davidelang, 2017-03-18:
since it seems as if this is fixed, I'll close the issue. reopen it if you are still having problems

closed 2017-03-18 by davidelang

jeylites, 2017-05-08:
What needs  to be change if I need the POW messages sent immediately?

davidelang, 2017-05-08:
On Mon, 8 May 2017, jeylites wrote:

> What needs  to be change if I need the POW messages sent immediately?

you would need to change the hardware so that the pow has the ability to measure 
power immediately. The existing hardware cannot do so, it needs to accumulate 
power information over time.

It doesn't have hardware that reports "I am using X power now", instead what it 
does is every time a specific amount of power is used, it toggles a bit, and the 
software has to count how many times this bit is toggled over several seconds to 
calculate how much power is being used.

David Lang