arendst/Tasmota #209: Adding a PCF 8591 ADC/DAC  Support

tombartosz, 2017-03-15:
Please consider adding  PCF 8591 ADC/DAC module support to this project. It would enable possibility to connect analog sensors.

ionciubotaru, 2017-03-16:
I think that instead of adding different peripherals support is better to add user defined functions.
Add `user.ino` with the following empty functions:
```
void user_before_setup() {
// code to run before Tasmota's setup
}

void user_after_setup() {
// code to run after Tasmota's setup
}

void user_loop() {
// code to run inside loop
}

// other functions to allow to connect different peripherals
```

In this way the upgrade process will be easier for users who use their own code and in the same time allow other users to develop interfaces without affecting Tasmota's core

arendst, 2017-03-16:
Sounds like espeasy...

davidelang, 2017-03-19:
There is a fair bit of code (with lots of ifdef protection) that is conceptually just walking through the sensor data and calling the appropriate mqtt_present or web_present code.

The proposal that I had under #217 to make there be an array that contained function pointers into each sensor would allow us to simplify the main code (simple loop over the array functions without all the ifdefs and different names) while making it easier to add in user code like this

khcnz, 2017-03-19:
Its been a while since i have had to worry about function pointers in c - but a quick refresher http://stackoverflow.com/questions/840501/how-do-function-pointers-in-c-work if i understand correctly what you are suggesting is we could basically define a "interface" for modules so they could all have an config() init() loop() webpresent() mqtt() etc

tombartosz, 2018-09-13:
> Hi,
>
> There is still interest on supporting this chip?
>
> Anyone is working in a PR ?

Of cource I'm still interested.

ascillato, 2018-09-13:
Great, have you done any progress on making a driver for it?

ascillato, 2018-09-14:
I'm just looking for someone that has this chip and wants to make a driver for Tasmota and also that this request don't get stale.

ascillato2, 2018-09-15:
Hi,

As it is a good idea but there is nobody working on a driver for that chip, this issue is being closed.

If anyone is interested on this, just ask and we can reopen this issue. Thanks.

closed 2018-09-15 by ascillato2

jello1974, 2018-12-01:
It would be nice if someone will make a driver for this chip