arendst/Tasmota #210: do you want help managing/closing issues?

davidelang, 2017-03-16:
If you would like me to help close issues that have been answered, I'm willing to do so but you would need to grant me additional access to the repo.

arendst, 2017-03-16:
Yes please do. I'll just have to find out where to grant access...

davidelang, 2017-03-16:
Ok, that works. I'll give folks a few days to respond to my nags that I just sent out then go through and close them with a note that they can be re-opened if there are still problems/questions

closed 2017-03-16 by davidelang

arendst, 2017-03-16:
Thnx for your support. That way I can focus on expansion/fixing issues.

davidelang, 2017-03-16:
answering questions seems to be the role that I fall into on projects. I do 
code, but it seems that it's most efficient for me to answer as many questions 
as I can to let the core devs code :-)