arendst/Tasmota #213: Need help on calculating color temperature

hamwong, 2017-03-16:
I use below formula and found value is bit offset

val warmwhite = ((((whitedimmer * 255) / 100) * brightness) / 100)
val coldwhite = (((((100 - whitedimmer) * 255) / 100) * brightness) / 100) 

like the formula cal as hex: 1909 and LED return correct value should be Hex:1608, please advise the correct formula,

arendst, 2017-03-17:
You're on the right track. Only keep in mind all variables are integers only and dividing integers will result in rounding down.

You can select warm and cold light values from 0 to 255. This will actually control the intensity of the leds. Intensity is also controlled by dimmer value 0 to 100.

So what happens is that once you select a color for both leds like 1909 the correct dimmer value is calculated. With the calculated dimmer value the (rounded) color is calculated again and returned to the user as what you notice as 1608. If you increase dimmer the color value will change too.

The end result is that the color value might fluctuate a bit caused by rounding but to the eye it makes no difference and it allows for easy control of power (on/off), dimmer (0 to 100) and color (0 to 255).

Oh by the way the color value is made up of two hexadecimal bytes where 19xx are the cold leds and xx09 are the warm leds.

See the code in file xdrv_snfled.ino starting at line 190 for color.

hamwong, 2017-03-17:
Real thanks for detail explain, I will try again to look at that file,
I have tried on yesterday for an hour, your code is bit too hard and complicate for me

closed 2017-03-20 by hamwong