arendst/Tasmota #214: Sonoff Touch: how to change the LED behaviour?

jdpiguet, 2017-03-16:
Hi everybody,

On a Sonoff Touch, I would like to have the LED (or the sensor) be in opposite of the relay:
- If relay (MQTT "POWER") is on, LED is off.
- If relay (MQTT "POWER") is off, LED is on.
Does anybody knows how to do it?

Regards, Jacques-D.

Rudoxxxx, 2017-03-16:
Problem is the sensor light?
If you send cmnd/sonoff/power on, then you LED relay is off?

khcnz, 2017-03-17:
I *think* this could be done just be changing the template in sonoff/sonoff_template.h from
    GPIO_LED1_INV
To
    GPIO_LED1

This should invert the LED behaviour from how it normally works (note I haven't tested this).

khcnz, 2017-03-17:
@arendst I know you don't have much space left - its a pity as otherwise we could have an "admin" or "advanced" mode that allows re-configuration of the non "user" programmable pins in the web ui / mqtt commands.

davidelang, 2017-03-17:
On Thu, 16 Mar 2017, khcnz wrote:

> @arendst I know you don't have much space left - its a pity as otherwise we could have an "admin" or "advanced" mode that allows re-configuration of the non "user" programmable pins in the web ui / mqtt commands.

we are getting closer to that. A lot of the things that are now able to be set 
with GPIO_USER were compile-time only a week or two ago.

khcnz, 2017-03-17:
I have only been around since v4.0, so I just a fresh chicken. You guys have probably been thinking about this for sometime.

Anyway a great platform - if you do have anything like this you could use some grunt work for I am happy to lend a hand, pretty sure I will be using the sonoff's everywhere in my home automation project so only right to give back since I am capable of.

davidelang, 2017-03-17:
aremdst has been accepting to pull requests, so go ahead and look over the code 
and submit improvements that you spot.

If you want to see places to start, look over the issues for enhancement 
requests and see if anyof them are something you can work on.
k

khcnz, 2017-03-17:
Yup I have sent him though a couple already - just don't want to double up on work he may already have mapped out in his mind!

davidelang, 2017-03-19:
to be clear, the enhancement is to have an 'admin mode' that includes all the GPIO functions and all the GPIOs, not just the GPIO_USER ones

khcnz, 2017-03-19:
@arendst how do you feel about such an admin mode? I could start it out just as an undocumented command..?

arendst, 2017-03-19:
@khcnz I prefer an extension to ledstate command. I want to keep the fixed Gpio fixed. They can use the wemos module to play around.

khcnz, 2017-03-19:
Good point - this could be achieved with LedState "Power State Inverted", I also wanted to add "On Always" as a selectable mode - as a way to indicate the unit is running / has power

arendst, 2017-03-19:
There is already an always on command for the led called ledpower on/off.

khcnz, 2017-03-19:
Ahh you are right (of course!) 

I had missed those somehow (I think my only justification is that LEDPOWER is not defined in the user_config). Would it make sense to combine those two options into ledstate? Already Ledstate 0 and ledpower 0 overlap?

arendst, 2017-05-04:
Changed the code to turn led on if power is off for Sonoff Touch only in next release.

kezzab25, 2017-10-26:
Is there any way of changing the led_inv1 to be on all the time or at least when it is connected to WIFI?

Rudoxxxx, 2017-10-26:
I make another programm for termo relay, and working fine with temperatures
without wi-fi network.
Only if you changing, the temp.you need to connect ti devices.

jdpiguet, 2017-11-27:
Running now with 5.9.1
I have changed the the sonoff_template.h file to match my requirement: the switch LED (the circle one) is now ON when the switch is off (the room is dark).
`jdp@amroth:Sonoff-Tasmota-5.x$ diff sonoff/sonoff_template.h*
388,389c388,389
<      GPIO_REL1_INV,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
<      GPIO_LED1,    // GPIO13 Blue Led (0 = On, 1 = Off)
---
>      GPIO_REL1,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
>      GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
jdp@amroth:Sonoff-Tasmota-5.x$ 
`

jdpiguet, 2017-11-27:
Running now with 5.9.1
I have changed the the sonoff_template.h file to match my requirement: the switch LED (the circle one) is now ON when the switch is off (the room is dark).
Lines 388-389:
 GPIO_REL1_INV,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
 GPIO_LED1,    // GPIO13 Blue Led (0 = On, 1 = Off)

niklasfink, 2018-01-27:
Hey @arendst, I have a T1 (which is basically the same as the Touch?) and I'd like to have the same inverted LED functionality. You should probably add this device to that (https://github.com/arendst/Sonoff-Tasmota/commit/77da1dfdc1064fa7b719bbdb84fdebe69e80b982#diff-9a0587e9aa7bed7ae2df63479960e7e0R2091) check as well.

I already installed my device, so is there a way to change that functionality remotely or rebuild the code locally and flash it remotely?

niklasfink, 2018-02-01:
I figured the master branch has the change (https://github.com/arendst/Sonoff-Tasmota/blob/master/sonoff/sonoff.ino#L2210). 
However, my behavior with a T1 is
Switch OFF = Background Light OFF
Switch ON = Background Light ON

How can I invert the behavior?

Mathyass, 2018-04-24:
I dont understand if there is about only small blue LED or about Circles which are in touch places. I need to change behaviour of theese two beckgraound lights... is it possible ?  Have T1 2CH EU.

niklasfink, 2018-04-25:
@Mathyass that's right, I wasn't clear about which lights I'm talking about.
Indeed, I'm also talking about the **touch button background lights** and _not_ the Wifi LED in the middle above the touch buttons.

I still didn't find a way to invert the behaviour of the touch button lights so that they are ON if they are off and OFF if they are on. That way you could find them better when it's dark.

arendst, 2018-04-25:
You can't simply because the led is hardwired to the relay.

Mathyass, 2018-04-26:
Maybe add new smd LED and connect it somwhere but it wants someone who say where :-)

tuga49, 2018-05-22:
@arendst  if I do that, I can get the the circle dots led on when switch is off? (the led I want on is the circle dots, not wifi circle):

Lines 388-389:
GPIO_REL1_INV, // GPIO12 Red Led and Relay (0 = Off, 1 = On)
GPIO_LED1, // GPIO13 Blue Led (0 = On, 1 = Off)

Thanks :)

Rudoxxxx, 2018-05-28:
Ok, but we need to make offline working, if no internet.

Ar Cieņu,

Rūdolfs Jozuus

Tel. +37129454993; 22062201

Aventes@inbox.lv

Mathyass <notifications@github.com> (šajā datumā: Ce, 2018. g. 26. apr.
16:46) rakstīja:

tuga49, 2018-05-28:
What I want to know is if I can get the circle dots led on when switch is off? (the led I want on is the circle dots, not wifi circle), when switch is off. @jdpiguet told if you edit the flashing code, it is possible. I just want to confirm. Because I seen somewhere that the led just can be on when relay is on....

jdpiguet, 2018-05-28:
As @arendst said: the LED is hardwired to the relay. In my case I'm using only the "switch", not the relay.
So I'm simply having the relay in the opposite position than the switch...
My setup is using an external main relay, controlled by MQTT.

tuga49, 2018-05-28:
Oh ok. So if I change this line, with normal sonoff touch, it will not work?

ascillato2, 2018-07-10:
Hi,

This issue seems to be addressed. Closing. Please, reopen if necessary.

closed 2018-07-10 by ascillato2