arendst/Tasmota #215: Temperature relay with DHT

Rudoxxxx, 2017-03-16:
This programm code is working with SONOFF temperature relay with DHT without Mqtt and server?
I will connect my Heatpump with SONOFF temperature relay, but not will use CHINA SERVER.

If I use this programm Sonoff-Tasmota code
1. How can I set temperature for relay?

2. Its possible to set day and night temperatures and how to do it?

davidelang, 2017-03-17:
On Thu, 16 Mar 2017, Rudoxxxx wrote:

> This programm code is working with SONOFF temperature relay with DHT without Mqtt and server?
> I will connect my Heatpump with SONOFF temperature relay, but not will use CHINA SERVER.
>
> If I use this [...]

The Sonoff-Tasmota code does not to this, what it does is reports the temps to a 
central server so that you can have logic there that decides what to do and 
sends on-off commands to the module.

Rudoxxxx, 2017-03-17:
The problem is mobile network. Not stabil work, and my sommer home temperature going up, not switching off if no internet connection.

Tasmota is very good and stabil programm, but
Where I can find code with so good programm for Temperature relay, what can work with deffined values without network, when network in not in range?

davidelang, 2017-03-18:
On Fri, 17 Mar 2017, Rudoxxxx wrote:

> The problem is mobile network. Not stabil work, and my sommer home temperature 
> going up, not switching off if no internet connection.

Why not have a local wifi network and server?

> Tasmota is very good and stabil programm, but
> Where I can find code with so good programm for Temperature relay, what can work with deffined values without network, when network in not in range?

I don't know, write your own? modify tasmota to implement your feature (and 
compile out other things to keep the size down to fit the flash)?

David Lang

Rudoxxxx, 2017-03-19:
I use 4MB chip on board.

I tray changing code with ATOM, but no instruction, how to use:
1. write and read EPROOM
2. Send MQTT command
3. Set new variables in code.
Have somebody introduction in wiki???

davidelang, 2017-03-19:
the sonoff devices don't have eprom, they have flash (not byte writable, must be 
erased in glocks)

the mqtt handling code is in sonoff.ino

If you need help setting variables, then you need to lookup one of the many into 
to programming classes. This firmware is written in C. Support for this level of 
learning is beyond the scope of the support we can provide here.

David Lang

Rudoxxxx, 2017-03-19:
In my programm No problem with writing to flash.

This Code in not easy structure for make changes.

davidelang, 2017-03-19:
we welcome patches to improve things :-)

I'm finding the code fairly reasonable for something it's size. It's not as 
simple as a trivial arduino sketch, but it does a lot more, and with 
functionality comes complexity (and when you add to it the desire to be able to 
compile things out, it gets even worse)
k

decologne, 2017-03-20:
Install Raspberry with openHAB2 on the remote location and write an easy rule with your preferred parameters.

Rudoxxxx, 2017-03-21:
Nice idea, but if raspberry halted in winter time, then all heating system frozen, all sanitary and toilet broken 10 000Eu send to gabage.
Better if Sonoff can working without internet and another server.

davidelang, 2017-03-21:
what makes you think the sonoff is going to be that much more reliable than a 
Pi?

In any case, you can add the logic if you want.

On Tue, 21 Mar 2017, Rudoxxxx wrote:

> Nice idea, but if raspberry halted in winter time, then all heating system frozen, all sanitary and toilet broken 10 000Eu send to gabage.
> Better if Sonoff can working without internet and another server.

coelner, 2017-03-22:
I think the possibility to set ranges for temperature and humidity is a good extension. At least to have a fail safe behaviour.
You need not many parameters, so far I count 7 values. 4 integers to set lower and upper range, and/or combination between temperature and humidity  and if the relay is open/closed if within range.
The values could be easily stored into the RTC Memory and survive a reboot
@Rudoxxxx Day/night is not useful, because you need a RTC or DCF77
However it is not that easy to understand the code, there is some documentation missing.

davidelang, 2017-03-27:
what questions do you have about the code?

coelner, 2017-03-31:
is there an API to add a fragment to the webserver?
where to add this function, include it into the dht file? 
It seems to be a giant blob, relations between the files and at the first view their function is not clear.

Rudoxxxx, 2017-04-10:
1. How to Send my own MQTT command? Can somebody write.
I need change code to send only one temperature value to MQTT server.

davidelang, 2017-04-10:
the code to decode/send a mqtt command is fairly involved, if you are just 
wanting to change the sensor output, then you just need to look at the section 
that generates the tele/ output

changing it to send only one temp instead of multiple temps is going to be a 
matter of splitting up the sensors section of the output into multiple lines of 
output instead of a single line.

If you are wanting to split up multiple sensors of the same type, you may have 
to split up the mqtt_present*() function in the hardware driver as well.

I think you are going to be on your own for this, it's a fair amount of work, 
will break with future updates to the software, and if you aren't extremely 
careful will cause confusion of which temperature is being reported, so it's not 
code that will ever be in the mainline build.

On Sun, 9 Apr 2017, Rudoxxxx wrote:

> Date: Sun, 09 Apr 2017 21:24:57 -0700
> From: Rudoxxxx <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f5a3fad4e52c942a3381d2e4cecbd3028d418937692cf000000011502cb1
>     992a169ce0cce5dfa@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Temperature relay with DHT (#215)
> 
> 1. How to Send my own MQTT command? Can somebody write.
> I need change code to send only one temperature value to MQTT server.
>
>

coelner, 2017-08-04:
#647

ascillato2, 2018-07-10:
Now this can be done using rules with latest firmware.

closed 2018-07-10 by ascillato2