arendst/Tasmota #216: Sonoff Touch: PowerOnState not working

thommycgn, 2017-03-17:
After energy lost or restart of sonoff touch the lights are always on. PowerOnState has no effect.

davidelang, 2017-03-17:
On Thu, 16 Mar 2017, thommycgn wrote:

> After energy lost or restart of sonoff touch the lights are always on. PowerOnState has no effect.

what is the setting of PowerRetain?

thommycgn, 2017-03-17:
On/Off doesn't change anything.

davidelang, 2017-03-17:
you would need to change the power state after making sure it's on to have it 
save it in flash.

If it's still not working, I would suggest changing your logging level to debug 
or more debug and see if you can see what happens when you turn it on or off 
(does it actually write to the flash) and then also look at what happens when it 
boots up (does it complain of any errors)

This is something that is used extensively by others, so it's surprising that 
it's not working for you.

can you dump your full config so we can see if you set anything odd.

Staars, 2017-03-20:
Hi,

do You use "mosquitto" as your mqtt-broker? I think I had the same issue and for me the solition was to set "persistence" to "false" in the config file of "mosquitto".
You can try to restart "mosquitto" and see, if your lights turn on (okay, obviously you should turn them off before ;). If this does not reproduce the problem, then you have some other problem.

davidelang, 2017-03-27:
did this issue get resolved?

steffenmllr, 2017-04-21:
I'm having the opposite problem (PowerOnState=1):

* power on the lamp -> lamp is on
* switch the lamp off (mqtt) -> lamp is off
* power the lamp off and power the lamp on -> lamp is on, after 2-3sek lamp goes off

Might that be also a mosquitto problem ? Shouldn't the lamp stay on with PowerOnState=1? 
Background is I'm using a slampher and the light switch on the wall should always be right. So if the device powers up it should always stay on.

davidelang, 2017-04-21:
what shows up in the logs on the device? enable the most verbose logging and it 
will show you exactly what instructions it's getting.
k

steffenmllr, 2017-04-21:
ok so this happens:

```
08:53:08 Wifi: Connected
08:53:08 mDNS: Initialized
08:53:08 HTTP: Webserver active on mylight-7676.local with IP address 192.168.1.105
08:53:10 MQTT: Attempting connection...
08:53:11 mDNS: Query done with 0 mqtt services found
08:53:11 MQTT: Connected
08:53:11 MQTT: tele/mylight/LWT = Online (retained)
08:53:11 MQTT: cmnd/mylight/POWER = 
08:53:11 RSLT: Receive topic cmnd/mylight/POWER, data size 3, data OFF
08:53:11 RSLT: DataCb Topic mylight, Group 0, Index 1, Type POWER, Data OFF (OFF)
08:53:11 MQTT: stat/mylight/RESULT = {"POWER":"OFF"}
08:53:11 MQTT: stat/mylight/POWER = OFF
08:53:11 MQTT: tele/mylight/INFO1 = {"Module":"Sonoff Basic", "Version":"4.1.3", "FallbackTopic":"DVES_173DFC", "GroupTopic":"sonoffs"}
08:53:11 MQTT: tele/mylight/INFO2 = {"WebserverMode":"Admin", "Hostname":"mylight-7676", "IPaddress":"192.168.1.105"}
08:53:11 MQTT: tele/mylight/INFO3 = {"Started":"Power on"}
08:53:12 Config: Saved configuration (1364 bytes) to flash at F7 and count 20
08:53:12 MQTT: stat/mylight/RESULT = {"POWER":"OFF"}
08:53:12 MQTT: stat/mylight/POWER = OFF
08:53:19 MQTT: tele/mylight/STATE = {"Time":"2017-04-21T08:53:19", "Uptime":0, "Vcc":3.178, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"mywifi", "RSSI":100, "APMac":"XXX"}}
```

I only have mosquito running, no other system. Looks like mosquito retained the value and is sending `cmnd/mylight/POWER` after reconnect right? Is there any way to overwrite it so when the device powers up it sends a power on command or is that better taken care of within mosquito ?

davidelang, 2017-04-21:
try disabling the retained setting, I'm not completely familiar with it, but 
I've seen someone post that that setting tells mosquitto to retain the setting, 
and could be triggering this (doesn't really make sense to me, afaik, a mqtt hub 
like mosquitto should just be passing messages, not sending commands)

but yes, this shows that something is sending the power off command, so it's 
either mosquitto or something else connecting to it. you may want to look at the 
mosquitto logs to double-check things there.
k

steffenmllr, 2017-04-21:
OK I've switched the mosquitto docker image and now it works. Thanks for helping out!

One thing though: If the device has the mqqt state ON and the power is cut and turned back on it is instant. If the mqqt state is off and the power is cut and turned back on it sometimes flashes or there is a delay - is that a know issue?

stefanbode, 2017-04-24:
As far as I can see in the log is that you retained (saved) your message you send to the device /stat/mylight/POWER OFF. Retaining these messages is a bad idea because also the SONOFF Firmware saves the result and power on this value. To get this value deleted you have to post it without a message AND retain it. Then it gets clear and your system is working properly again.

closed 2017-05-04 by arendst

davemcm2, 2017-05-15:
@steffenmllr How exactly did you solve this? I'm having the exact same issue you were having... I'm not using docker, directly installed mosquitto using the included HA script.

If you could give some info it would be much appreciated.

Thanks,

stefanbode, 2017-05-16:
The retaining of the message is not a mosquitto issue. You have to solve this on the client side. You have to set it to "NO RETAIN". For the Tasmoto make sure you did not override:
#define MQTT_BUTTON_RETAIN     0                 // [ButtonRetain] Button may send retain flag (0 = off, 1 = on)
#define MQTT_POWER_RETAIN      0                 // [PowerRetain] Power status message may send retain flag (0 = off, 1 = on)
#define MQTT_SWITCH_RETAIN     0  

If you switch ON/OFF the button with another application like openHAB ensure in the configuration that retains if OFF.

davemcm2, 2017-05-16:
Thanks, so I just have to disable all kinds of retaining on the actual
sonoff tasmota firmware itself? Contrary to the instructions for
HomeAssistant on the Tasmota wiki?

I've spent 12 hours on this switching lights on and having them switch
themselves off again, I think I'm going crazy.

I'll give it a shot and report back.

stefanbode, 2017-05-16:
Please use MQTT Spy (https://github.com/eclipse/paho.mqtt-spy/wiki) to see if there are values set. If so, you have to "retain" and send an EMPTY message do delete the key. I assume there are still stored keys in mosquitto. the MQTT Spy is great for analyzing and setting.

davidelang, 2017-05-16:
the confusing thing is that these settings aren't telling the sonoff to keep 
track of things, they are telling the mqtt server to keep track of the settings.

If these are set, the mqtt server is overriding the poweronstate by setting it 
to what the mqtt server thinks it should be.

davemcm2, 2017-05-16:
Earlier, I ended up just disabling MQTT and it worked fine with the built in Hue emulation and using the wall switch to power on / off, as soon as I enabled MQTT again there it's switching itself off again, all retain options mentioned are set to 0.

Going to try mqtt-spy now and see if I can remove the retain.

stefanbode, 2017-05-16:
you can call me on skype: stefan.bode and we can do screen sharing
Should be a <5min job if MQTT Spy is already connected to your MQTT Mosquitto.

steffenmllr, 2017-05-16:
Retaining the messages by the mqtt server was also my problem. It also drove me mad for a while but makes perfect sense when you think about it:

* Power goes on -> Sonoff-Tasmota is not yet connected to the broker and PowerOnState=1 => light goes on
* Connection to the mqtt -> If the value was sent via a retain flag oder the mqtt server persists the data, it will send the state down to the light: if the mqtt state was off -> light switches off (happens after a few seconds when the light connected to wifi and to the broker)

davemcm2, 2017-05-16:
I managed to fix it sending that one simple command to overwrite the already retained message.

For anyone wondering (using mosquitto):

mosquitto_pub -h localhost -p 1883 -r -t 'cmnd/sonoff3/POWER' -n -u homeassistant -P mosquittopassword

-r sends the retain flag
-n sends a null byte message to overwrite (delete) the retained one already there.

It now worked perfectly for me, thanks to the helpful commends of @steffenmllr 

Much appreciated 👍

davidelang, 2017-05-16:
could one of you do a writeup of how to make this work on the wiki? Either a 
page of it's own or add it to the troubleshooting page (this comes up frequently 
enough)

Please include the specific configs (from earlier) and this mosquitto_pub 
command with explination.

davemcm2, 2017-05-16:
I'll try and get something done later today or tomorrow, it's the least I can do.

stefanbode, 2017-05-16:
done. Feel free to enhance with your experience

https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration

davidelang, 2017-05-16:
There is something non-standard in your config for the commands to be 
hm/cmnd/ESP_0C220C/power, the stock firmware would start with cmnd/ not hm/cmnd

It's probably worth nothing that this is all using mosquitto, other mqtt servers 
may be different (I don't know, I haven't used anything else, so I'm raising the 
possibility in case it needs to be called out)

when you issue the command to sub to the topic, what response do you get if 
mosquitto is retaining the value and will override poweronstate vs if it's not 
set?

stefanbode, 2017-05-16:
good point. changed.

davidelang, 2017-05-16:
when I try the mosquitto_sub command, it just sits there with no response, is 
this what should happen if nothing is happening?
k

stefanbode, 2017-05-16:
yes, in this case, you have no stored messages on the server. Your IN is clear as it should be

asantos15, 2018-03-01:
Hello,

My sonoff touch started turning on at boot. Here is the log:

22:40:13 MQT: Attempting connection...
22:40:14 MQT: Connected
22:40:14 MQT: tele/sonoff/LWT = Online (retained)
22:40:14 MQT: cmnd/sonoff/POWER = 
22:40:14 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
22:40:14 MQT: stat/sonoff/POWER = ON

I got nothing in IN folder, but it keeps turning on as it connects to the wifi/boots.

Any idea?

Thanks!

davidelang, 2018-03-01:
check for retained messages, see the wiki page on poweronstate.

change logging to more detail and you will see where the instruction to turn on 
comes from.

asantos15, 2018-03-06:
Hello @davidelang ,

Here is the output:
22:33:01 MQT: Attempting connection...
22:33:01 MQT: Connected
22:33:01 MQT: tele/sonoff/LWT = Online (retained)
22:33:01 MQT: cmnd/sonoff/POWER = 
22:33:01 MQT: Subscribe to cmnd/sonoff/#
22:33:01 MQT: Subscribe to cmnd/sonoffs/#
22:33:01 MQT: Subscribe to cmnd/homeassistant-1/#
22:33:01 RSL: Group 0, Index 1, Command POWER, Data ON
22:33:01 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
22:33:01 MQT: stat/sonoff/POWER = ON
22:33:02 CFG: Saved to flash at FA, Count 1457, Bytes 1648

I've cleaned mqtt folder by sending the following command:  mosquitto_pub -h xx.xx.xx.xx -p 1883 -r -t 'cmnd/sonoff/POWER' -n -u username -P password
...but still. 
PowerOnState is off, PowerRetain is 0=off as well.

Any ideas would be greatly appreciated.