arendst/Tasmota #218: [New feature]: implement self protection for sonoff-pow

lionhe1966, 2017-03-17:
A batch of Sonoff-Pow switches produced in December 2016 and January 2017 is defective. ITEAD published a recall on their website
(https://www.itead.cc/blog/sonoff-th16-and-pow-recall-notice)
Wouldn't it be possible to implement a simple self-protection rule that switches the relay off if the power exceeds some user defined threshold?

arendst, 2017-03-17:
Take a look at commands PowerHigh, MaxPower, MaxPowerHold and MaxPowerWindow. PowerHigh needs a server to act on it's result (power down Pow). The other three inbuilt commands may be what you need. Experiment with it (no documentation yet).

davidelang, 2017-03-17:
On Fri, 17 Mar 2017, lionhe1966 wrote:

> A batch of Sonoff-Pow switches produced in December 2016 and January 2017 is defective. ITEAD published a recall on their website
> (https://www.itead.cc/blog/sonoff-th16-and-pow-recall-notice) [...]

can you identify what threshold is safe (given that the design is for full 16A 
of power, and the problem is failure to follow the design)

can you identify which devices need to be crippled like this and which are safe 
to use at full power?

lionhe1966, 2017-03-17:
The only clue that we have is the picture on the ITEAD website. Unfortunately they only write that the switch on the picture was exposed to an unspecified high load for an unspecified long time.
My two switches (with a seal date 12-12-2017) amount of tin looked identical to the pictures.
Under the assumption that, by mistake, they used in production the settings for 10A switches, I  think to use the switches with maximum 10A therefore setting a powerlimit of approximately 2.5kW.

It may be interesting to compare pictures of the PCBs.

In any case, I think that these devices are not well designed for 16A. The case is too small. In fact I had some trouble connecting two 16A rated cables to the switches: the holes for holding the two cables are simply too small unless you remove the outside insulator and it is difficult to close the case.

lionhe1966, 2017-03-18:
The command MaxPower sets the power threshold for switching on. If I understand correctly, after MaxPowerHold seconds, the power is switched off, and a warning signal is issued. I think that this is exactly what I was asking for. It would be worth adding that command to the Wiki documentation.
I do not understand what the PowerHigh threshold does. I've tried to set it , but nothing happened.

arendst, 2017-03-18:
PowerHigh should send a (telemetry) message when power is above set level like `stat/sonoff/MARGINS {"PowerHigh","ON"}`.

lionhe1966, 2017-03-18:
Ok, I noticed that message, and now I understand its meaning. It is very convenient functionaly for load shredding. What's the meaning of MaxPowerWindow?

arendst, 2017-03-18:
It's been a while since I wrote it but I think it works as follows:

Once you set MaxPowerWindow to > 0 seconds it will turn on power after MaxPowerWindow seconds and see if the power overload is solved. If not, the sequence of overpower starts again up to MAX_POWER_RETRY (as defined in sonoff.ino line 122)  = 5 times.

davidelang, 2017-03-19:
It sounds as if this is solved (other than possibly needing to have someone update the wiki :-)

can this ticket be closed now?

lionhe1966, 2017-03-19:
I would say that the ticket can be closed. I'll try to update the wiki with the info contained here.

closed 2017-03-27 by davidelang

rmoosa, 2023-04-03:
Apologies to hijack this thread; is there any way to set PowerHigh, MaxPower, MaxPowerHold and MaxPowerWindow in user_config_override?

sfromis, 2023-04-03:
Only as commands in `USER_BACKLOG`

(Beware that changes do not apply when upgrading a binary, unless you do a full reset to the defaults you have there)

rmoosa, 2023-04-04:
> Only as commands in `USER_BACKLOG`
> 
> 
> 
> (Beware that changes do not apply when upgrading a binary, unless you do a full reset to the defaults you have there)

Thanks for the quick response.