arendst/Tasmota #221: Thermal Imaging video of Sonoff and Electrodragon

stoehrmark, 2017-03-17:
First off this is excellent code thanks for all your hard work!!

I took some video showing the temperature the boards get to under a 10A load. Don't know if this is good or bad. But it is what it is.

Sonoff
https://youtu.be/S5ubXyYH0_0
Parts of this board get to 54˚C under normal operation.

Electrodragon
https://youtu.be/fw8nh8uWj3E
Parts of this board get to 77˚C under normal operation.

davidelang, 2017-03-19:
Nothing we can really do about it since we don't have any control over the hardware.

It doesn't surprise me that this code gets hotter than stock (it does a lot more). If you are concerned about the temps, you can enable the sleep function. It's there to reduce the power draw, but that will also reduce the heat generated.

running at a lower clock speed will probably also make things cooler.

closed 2017-03-19 by davidelang

stoehrmark, 2017-03-19:
@davidelang This was never intended to compare the stock code to this one and yes it does do a ton more. Thank you to @arendst and helpers!! 

In fact the Sonoff had stock code and the Electrodragon was without any code when i did the video. It was just to show that under full load 10A the board traces primarily the mains voltage load traces get hot. I'm suggesting that users should be careful. No one wants a fire.

I'm certainly not any expert or engineer just passing on my observations. These could be totally acceptably temperatures. I think these devices are great. I have got quite a few running.

Long live the ESP8266.

davidelang, 2017-03-19:
ahh, I thought you were talking about CPU temps.

As for the trace temps, we have to trust the vendor to get things right.

They have done a recall of the TH series of devices because of exactly this 
reason, there wasn't enough solder on the traces, so they were getting too hot.

stoehrmark, 2017-03-19:
I did see that recall. For myself I never run my devices anywhere near 10A. Just to be safe.

 I don't have any of the Sonoff TH16s to see what they look like under load. I will have to see if i can get one to test.