arendst/Tasmota #222: POW v 2.0

seugenes, 2017-03-17:
After flashing, on power-on the let blinks once. Then only a few seconds button press switches the relay and the led On. Then the led is always on. It is blinking once after each short button press and returns to ON state. if you don't touch the button then periodically (once in 30 seconds or so) you can hear the relay switches on and of by itself.

davidelang, 2017-03-18:
On Fri, 17 Mar 2017, seugenes wrote:

> After flashing, on power-on the let blinks once. Then only a few seconds
> button press switches the relay and the led On. Then the led is always on. It
> is blinking once after each short button [...]

what shows up in the log? (either the web log or syslog)

David Lang

seugenes, 2017-03-18:
Don't have any. No web access and since it's POW unable to attach to serial.
Was just wondering if it could be due to version 2.0 of POW. Just got it 2 days ago.
Works excellent on other devices.

davidelang, 2017-03-18:
I don't know that we have any information about a new version of POW

can you configure a syslog server and configure the pow to log to it?

It sounds as if it's rebooting, but without diagnostics, it's going to be hard to tell

seugenes, 2017-03-18:
It has no wifi and network. 
Will have to build that safe circle to attach to serial debug but it will take me some time.

seugenes, 2017-03-18:
:) What does the quick led blink on power on mean?

arendst, 2017-03-18:
Honesty I don't know. It all depends where the code is at that time. I suggest connecting the serial interface and enable debug logging option 3 to see what's happening at that time.

seugenes, 2017-03-20:
Did anyone manage to run it on POW v 2.0?

khcnz, 2017-03-21:
I thought the only thing different was increased solder on the tracks? that shouldn't impact the firmware....

If you can flash the device you should also have enough power from your FTDI to run the device (again not connected to AC) direct from the FTDI and get the serial output - if you can get that we should be able to help

seugenes, 2017-03-21:
[image]
[image]

Should I use some external power for it? if i just connect it via serial in the same way like uploading the firmware I don't get anything on the Serial Monitor. Am I missing something?

davidelang, 2017-03-21:
You are providing power to it to be able to flash it, that may not be enough 
power to trip the relay, but it's enough to operate the CPU and wifi.

what is serial logging set to? (and do you wait long enough to get the 5 min 
log)

what happens if you try to send something to the device (say the string "power" 
followed by a newline)?

khcnz, 2017-03-21:
Just on David's point the serial logging is disabled after 5mins from boot. However you should get a ton of logs on startup. What serial monitor are you using? Arduino IDE?

seugenes, 2017-03-22:
Yes. Managed to get the logs. Compiled and uploaded with LOG_LEVEL_ALL for  #define SERIAL_LOG_LEVEL but there is not much useful info..
00:00:06 Wifi: Checking connection...
00:00:06 Wifi: Attempting connection...
00:00:09 Wifi: Checking connection...
00:00:09 Wifi: Attempting connection...
00:00:11 Wifi: Checking connection...
00:00:11 Wifi: Attempting connection...
00:00:14 Wifi: Checking connection...
00:00:14 Wifi: Attempting connection...
00:00:16 Wifi: Checking connection...
00:00:16 Wifi: Attempting connection...
00:00:19 Wifi: Checking connection...
00:00:19 Wifi: Attempting connection...
00:00:21 Wifi: Checking connection...
00:00:21 Wifi: Attempting connection...
00:00:24 Wifi: Checking connection...
00:00:24 Wifi: Attempting connection...
00:00:26 Wifi: Checking connection...
00:00:26 Wifi: Attempting connection...
00:00:29 Wifi: Checking connection...
00:00:29 Wifi: Attempting connection...
00:00:31 Wifi: Checking connection...
00:00:31 Wifi: Attempting connection...
00:00:34 Wifi: Checking connection...
00:00:34 Wifi: Attempting connection...
00:00:37 Wifi: Checking connection...
00:00:37 Wifi: Attempting connection...
00:00:39 Wifi: Checking connection...
00:00:39 Wifi: Connect failed with AP timeout
00:00:39 Wifi: Patch issue 2186
00:00:39 Wifi: Connecting to AP255  in mode 11N as �������������������������������...
00:00:41 Config: Saved configuration (1340 bytes) to flash at F7 and count 188
00:00:42 Wifi: Checking connection...
00:00:42 Wifi: Attempting connection...
00:00:44 Wifi: Checking connection...
00:00:44 Wifi: Attempting connection...
00:00:47 Wifi: Checking connection...
00:00:47 Wifi: Attempting connection...
00:00:49 Wifi: Checking connection...
00:00:49 Wifi: Attempting connection...
00:00:52 Wifi: Checking connection...
00:00:52 Wifi: Attempting connection...
00:00:54 Wifi: Checking connection...
00:00:54 Wifi: Attempting connection...
00:00:57 Wifi: Checking connection...
00:00:57 Wifi: Attempting connection...
00:01:00 Wifi: Checking connection...
00:01:00 Wifi: Attempting connection...
00:01:02 Wifi: Checking connection...
00:01:02 Wifi: Attempting connection...
00:01:05 Wifi: Checking connection...
00:01:05 Wifi: Attempting connection...
00:01:07 Wifi: Checking connection...
00:01:07 Wifi: Attempting connection...
00:01:10 Wifi: Checking connection...
00:01:10 Wifi: Attempting connection...
00:01:12 Wifi: Checking connection...
00:01:12 Wifi: Attempting connection...
00:01:15 Wifi: Checking connection...
00:01:15 Wifi: Attempting connection...
00:01:17 Wifi: Checking connection...
00:01:17 Wifi: Connect failed with AP timeout
00:02:30 RSLT: tele/������������������������������������������������������ = {"Time":"1970-01-01T00:02:30", "Uptime":1}

seugenes, 2017-03-22:
And then, indeed it restarts

00:03:55 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
ב
00:00:00 APP: Project sonoff �����������������������������������������������������������������������������������������������������������
00:00:00 Config: Saved configuration (1340 bytes) to flash at F8 and count 189
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:01 Wifi: Patch issue 2186
00:00:01 Wifi: Connecting to AP255  in mode 11N as �������������������������������...
00:00:02 RSLT: ��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������� = {"idx":-1,"nvalue":0,"svalue":""}
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Attempting connection...
00:00:06 Wifi: Checking connection...
00:00:06 Wifi: Attempting connection...
00:00:09 Wifi: Checking connection...
00:00:09 Wifi: Attempting connection...
00:00:11 Wifi: Checking connection...
00:00:11 Wifi: Attempting connection...
00:00:14 Wifi: Checking connection...
00:00:14 Wifi: Attempting connection...
00:00:16 Wifi: Checking connection...
00:00:16 Wifi: Attempting connection...
00:00:19 Wifi: Checking connection...
00:00:19 Wifi: Attempting connection...
00:00:21 Wifi: Checking connection...
00:00:21 Wifi: Attempting connection...
00:00:24 Wifi: Checking connection...
00:00:24 Wifi: Attempting connection...
00:00:26 Wifi: Checking connection...
00:00:26 Wifi: Attempting connection...
00:00:29 Wifi: Checking connection...
00:00:29 Wifi: Attempting connection...
00:00:31 Wifi: Checking connection...
00:00:31 Wifi: Attempting connection...
00:00:34 Wifi: Checking connection...
00:00:34 Wifi: Attempting connection...
00:00:37 Wifi: Checking connection...
00:00:37 Wifi: Attempting connection...
00:00:39 Wifi: Checking connection...
00:00:39 Wifi: Connect failed with AP timeout
00:00:39 Wifi: Patch issue 2186
00:00:39 Wifi: Connecting to AP256  in mode 11N as �������������������������������...

davidelang, 2017-03-22:
well, that tells us that it booted successflly, but that you aren't connecting 
to the network. check the wifi credentials in your build

davidelang, 2017-03-22:
correction, I do see that it did connect. but it doesn't have a valid hostname 
(also visible in the tele/ line)

check what name it's set to (the hostname and topic commands)

seugenes, 2017-03-22:
Is it in user_config.h? what are those setting names? The thing is that I just flashed sonoff touch and then sonoff basic with the exactly same code, and it was good.

seugenes, 2017-03-22:
Also, when I press the button it restarts and I get this:
00:00:37 APP: Multipress 1
00:00:39 RSLT: ��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������� = {"idx":-1,"nvalue":0,"svalue":""}
00:00:39 RSLT: stat/������������������������������������������������������ = {"������������������������������������������������������������

Exception (28):
epc1=0x4000e1e0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff2130 end: 3fff2520 offset: 01a0

>>>stack>>>
3fff22d0:  3fff22f0 3fff22e0 0000000c 4020351f  
3fff22e0:  3fff2312 ffffffff ffffffff 00000001  
3fff22f0:  00000000 3ffe84a4 00000000 00000001  
3fff2300:  00000000 3ffe84a4 00000000 4020855b  
3fff2310:  ffff227b ffffffff ffffffff ffffffff  
3fff2320:  ffffffff ffffffff ffffffff ffffffff  
3fff2330:  ffffffff ffffffff ff01ffff ffffffff  
3fff2340:  ffffffff ffffffff ffffffff 00ffffff  
3fff2350:  74617473 ffffff2f ffffffff ffffffff  
3fff2360:  ffffffff ffffffff ffffffff ffffffff  
3fff2370:  ffffffff ffffffff ffffffff ffff01ff  
3fff2380:  ffffffff ffffffff ffffffff ffffffff  
3fff2390:  0000ffff 00000000 3ffe848c 40207f32  
3fff23a0:  00000000 3fff0a8c 00000001 4020870a  
3fff23b0:  3fff0811 40103616 3fff23f8 40203e2f  
3fff23c0:  000006c7 00000000 00000000 3fff0b98  
3fff23d0:  3fff109c 3fff0804 3fff0c9c 40210cfd  
3fff23e0:  3a505041 6c754d20 72706974 20737365  
3fff23f0:  40210031 00000030 0000001e ffffffff  
3fff2400:  4021c892 00000000 00000000 401006d9  
3fff2410:  00000000 00000000 00000000 fffffffe  
3fff2420:  ffffffff 3fffc6fc 00000058 00000000  
3fff2430:  00000000 3fffdad0 3fff14f8 00000030  
3fff2440:  3fff0bb0 3fff0f4d 3fff0bf8 feefeffe  
3fff2450:  3a505041 6f725020 7463656a 6e6f7320  
3fff2460:  2066666f ffffffff ffffffff ffffffff  
3fff2470:  ffffffff ffffffff ffffffff ffffffff  
3fff2480:  ffffffff ffffffff ffffffff ffffffff  
3fff2490:  40105d96 00b98d98 ffffffff ffffffff  
3fff24a0:  40105e52 3fff3814 00b98d98 4021e87c  
3fff24b0:  40106024 00bac4ec 3fff12d8 00000000  
3fff24c0:  3ffefdf0 3fff12d8 3fff1500 3fff12d8  
3fff24d0:  3fffdad0 3fff14f8 4021e29c 3fff1500  
3fff24e0:  402019f2 000000ff 3fff14f0 3fff14f8  
3fff24f0:  3fffdad0 00000000 3fff109c 402145e9  
3fff2500:  3fffdad0 00000000 3fff14f0 4021e2e8  
3fff2510:  feefeffe feefeffe 3fff1500 40100718  
<<<stack<<<

davidelang, 2017-03-22:
what do you get from the hostname and topic commands?

> 00:00:39 RSLT: [...]
 ????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????? = {"idx":-1,"nvalue":0,"svalue":""}
> 00:00:39 RSLT: stat/?????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????? = [...]

This is showing that there is something very wrong with the hostname/topic.

you can try holding the button down for 4 seconds to force it to reload the 
config from what was set in user_config.h or issue the command 'reset 2' or 
change the value of cfg_version in user_config.h and rebuild/reflash.

David Lang

seugenes, 2017-03-22:
I changed 
#define WIFI_HOSTNAME          "%s-%04d"  to 
#define WIFI_HOSTNAME          "POW1" 
flashed and the result is the same.

Cannot find any cfg_version in user_config.h..

davidelang, 2017-03-22:
Ok, I wasn't quite right in the name, it's CFG_HOLDER

k

seugenes, 2017-03-22:
It did the trick! Thank you very much.

closed 2017-03-22 by seugenes