arendst/Tasmota #223: How is POW calibration working?

jdpiguet, 2017-03-18:
Hi all,

I'm using Tasmota rev. 4.0.6
I have read the wiki about the POW calibration.
The indicated voltage was 220 Volts, the measured one (multimeter) was 235.
I have send the command:
mosquitto_pub -h domoticz.lan -i testPublish -t cmnd/powe8a718/HlwUcal, -m '10'
Now the console log is showing:tele/powe8a718/INFO3 {"Started":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40107345 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x000

What should be the value for the HlwUcal in my case?
How to calculate the other calibration values?
Is it possible to reset the configuration to "factory defaults"?

Regards, Jacques-D.

khcnz, 2017-03-18:
Just had a quick look in source - I _think_ the answer is:

The default scaling value is 
`#define HLW_UREF_PULSE         1950         // was 1666us = 600Hz = 220V`

So, if sonoff is reporting 220v for an actual 235 the scaling would therefore be (1950*235/22) that answers to 2082.

jdpiguet, 2017-03-18:
Thanks @khcnz , it's a good hint.
Actually, after roughly 1 hour of measurement/correction cycles, I have reach the value 2039
Measurements were oscillating between 230 and 238 Volts
I'm still a bit confused on the formula.
I'm waiting for a second POW and will put results here....

arendst, 2017-03-18:
@jdpiguet yes feel free to update the wiki with your valued information.

closed 2017-03-23 by arendst