arendst/Tasmota #224: suggestion: covers

aferraz, 2017-03-18:
@arendst, it's possible to add, for board with 2 relay like electrodragon, a function to trasform it in covers ?
For example:
stat/cover1/STATE: give current state OPEN/CLOSE
cmnd/cover1/set: accepts values OPEN/CLOSE/STOP, when OPEN power off relay2 and power on relay1, otherwise power off relay1 and poweron relay2. For STOP power off both relay.
Setting pulsetime for the relay it's possible to power off the relay when the cover is totally open or close.

Thanks 
 Alessandro

davidelang, 2017-03-18:
in user_config.h you can define what the states are (curently ON/OFF) but you 
cannot have different states for different relays, and there are no commands 
that switch multiple relays at the same time, you would need to send two 
commands.

That should be trivial to do in your home automation software, and the 
complexity of listing all the possible combinations in the firmware would eat up 
too much space and add too much complexity.

On Sat, 18 Mar 2017, Alessandro wrote:

> @arendst, it's possible to add, for board with 2 relay like electrodragon, a function to trasform it in covers ?
> For example:
> stat/cover1/STATE: give current state OPEN/CLOSE
> cmnd/cover1/set: [...]

aferraz, 2017-03-19:
Hi @davidelang if it's a tricky thing there are no problems to do with home automation software. It could be a useful addon to the sotware.  Only an idea ;-)

Regards
 Alessandro

davidelang, 2017-03-27:
I think that what you are looking for is the functionality outlined in #288

closed 2017-03-27 by davidelang