arendst/Tasmota #225: Sonoff combine arduino uno sketch

SayedBaqer, 2017-03-18:
Is there chance to use arduino uno code and combine in easy way with your development code?
Where i dont need any delay in the sketch, and connect i2c lcd and i2c gpio and dht sensor
I will be happy if you help me to implement it

davidelang, 2017-03-18:
I don't think they would be combined, they would be two separate schetches 
installed on two different CPUs/devices. While the two would communicate via 
i2c, the code of the two would be independent.

David Lang

On Sat, 18 Mar 2017, SayedBaqer wrote:

> Is there chance to use arduino uno code and combine in easy way with your development code?
> Where i dont need any delay in the sketch, and connect i2c lcd and i2c gpio and dht sensor
> I will be happy if you help me to implement it

SayedBaqer, 2017-03-19:
No dear, i mean, i want to replace arduino uno by sonoff esp8266, so how i can implement my old code in the esp8266?

arendst, 2017-03-19:
Browse my source and use what ever you like. I2C and DHT are there too so all you need to do is scrap things you do not need. Nice trial and error excercise...:-)

Rudoxxxx, 2017-03-19:
"See wiki how to start."
Can you help Just a little bit.
I have my arduino termorelay code, but Thasoma code is not easy for change.
I will put in your code for local temperature relay.
1. READ settemperature and diference(dead zone) from MQTT and save to EPROOM, if no internet then set default values from programm if no values in EPROOM.
(Problem, how to use EPROOM in Tasmota)
2. Set relay states on/off from temperature and dead zone, and send to MQTT. (I can use default from Thasoma)
3. Send temperature readings to MQTT in one TEMPERATURE TOPIC for easy put in server

davidelang, 2017-03-19:
I think you should look at the sonoff pow related code for examples. It does some similar things with power readings

davidelang, 2017-03-27:
do you still have questions about this?

note that temp data is a subset of sensor data and is sent as part of that.

closed 2017-04-23 by davidelang