arendst/Tasmota #227: MQTT Query for external switches?

dc75, 2017-03-18:
I'm currently running the latest version of Sonoff-Tasmota on a NodeMCU board with 2 external switches attached to GPIO4 & GPIO5.  The switches were configured through the web interface as Switch1 & Switch2.  Everything works fine & I receive an MQTT message when either switch changes state.  In my application the switches don't change state very often, so I was trying to find a way to query the state of each switch periodically via MQTT, just to make sure that I did not miss an MQTT message & end up with the software out of sync with the physical state of the switch.  I see how this can be done for GPIO assigned as a relay with the `cmnd/myesp/power1 status` command, but I can't seem to figure out how to accomplish this for GPIO assigned as a switch.

Perhaps I have overlooked something simple?  Any ideas on how to achieve this would be greatly appreciated!  Thanks!

ionciubotaru, 2017-03-18:
Look at: SwitchRetain and ButtonRetain in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

dc75, 2017-03-18:
Thanks for the suggestion @ionciubotaru!  I've looked at both SwitchRetain & ButtonRetain, but I'm not sure that either of those would work for what I'm trying to achieve; I'm very new to this software & ESPs, so please feel free to correct me if I'm wrong!  

I'm using Node Red & I already retain the last switch state there, so it seems like SwitchRetain/ButtonRetain would be kind of redundant for my application.  What I'd really like to be able to do is to periodically query the GPIO that my switches are connected to.  My switches normally only change state a few times during a 24 hour period; since I only receive an MQTT message on a switch changing state, it can be many hours since I last received the state of the switch.  In this situation, I'd feel much more confident if I could periodically get a confirmation on the state of the switch.   :smiley:

ionciubotaru, 2017-03-19:
I understand your needs, but Tasmota was designed to be used with Sonoffs where the switch is always open, except the short moment when you press it. 
I suggest change the code and subscribe to a different additional topic and handle the received messages.
Tasmota is based on [knolleary MQTT library ](http://pubsubclient.knolleary.net/).
Take a look at [API Documentation](http://pubsubclient.knolleary.net/api.html)

davidelang, 2017-03-19:
take a look at switchmode, specifically follow and inverted follow

dc75, 2017-03-19:
Thanks @ionciubotaru & @davidelang (I am using switchmode FOLLOW).  I understand now that I'm asking for something that the software was not designed to do.  Just wanted to make sure that I wasn't overlooking something simple.  

Thanks for setting me straight!

davidelang, 2017-03-19:
by the way, we got off into the switchmode stuff, but if you are just looking 
for a way to query the state of things, status 0 will return a bunch of things, 
and you can send the appropriate status command to get just one of them.

On Sun, 19 Mar 2017, dc75 wrote:

> Date: Sun, 19 Mar 2017 03:57:23 -0700
> From: dc75 <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24fcf047911bf1f3e1ecdf14c9f1e962cac37d33abd92cf0000000114e6261
>     392a169ce0cd3f75d@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Mention <mention@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] MQTT Query for external switches? (#227)
> 
> Thanks @ionciubotaru & @davidelang (I am using switchmode FOLLOW).  I understand now that I'm asking for something that the software was not designed to do.  Just wanted to make sure that I wasn't overlooking something simple.
>
> Thanks for setting me straight!
>
>

dc75, 2017-03-19:
@davidelang  - I had tried the Status 0 command, & did receive back a lot of information, but the current state of Switch1 & Switch2 did not appear to be in the returned message.  Perhaps I missed something, as my test was rather simple - I issued the Status 0 command & looked at the returned message, then changed the state of Switch1 or Switch2 & issued the Status 0 command again; the message I received was identical to the original message, so I did not attempt to dig further into the message.

davidelang, 2017-03-19:
ahh, so what you are looking for is a status command that reads the state of lastwallswitch[] (and probably lastbutton[]) and spits them out. That shouldn't be hard to do

davidelang, 2017-03-19:
I threw together an update that does this. I don't have any devices with external switches, so I have just compile-tested this and haven't tried running it yet. If you can test it and see that it works, I'll create a PR for it

https://github.com/davidelang/Sonoff-Tasmota/tree/buttonstatus

dc75, 2017-03-19:
@davidelang  -  Thanks!  I have tested your update & it works great!  

Issuing the MQTT command `cmnd/my_esp/status 12` returns an MQTT message `stat/my_esp/STATUS12` containing the state of "Switch1" & "Switch2", which is exactly what I was trying to do.  I'll parse out the message in Node Red to get the values of Switch1 & Switch2.

Thanks so much for your help!

arendst, 2017-03-19:
I would wait just a minute as the new version about to be released does it differently ;-)))

It uses Status 10 and reports ON/OFF taking into account inverted signals.

Hold your horses...

ionciubotaru, 2017-03-19:
Great improvement, please add Switch state in [commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

khcnz, 2017-03-19:
@ionciubotaru could you explain in what scenario that would be useful? 

If I understand you want to receive the current state of the switches/buttons (not relay power) in a status message? I guess that could work for toggle/follow but wouldn't be very useful for momentary

ionciubotaru, 2017-03-19:
Is useful when you use esp8266 in your own project, not in a sonoff device
If sonoff loose mqtt connection and a user modify un input is nice to have a command to get the input state when mqtt recover

dc75, 2017-03-20:
Thanks for the update Theo!

closed 2017-03-20 by dc75