arendst/Tasmota #230: Cant load webserver

prasad7676, 2017-03-18:
after flash firmware successfully i cant open web page using sonoff-4339.local...or ip add.

after change little bit on code i can connect to mqtt but cant get GUI...Please help me...thanks
[image]

arendst, 2017-03-19:
Use another browser.

davidelang, 2017-03-19:
you would need to provide information about what happens when you try to connect to the webserver

ionciubotaru, 2017-03-19:
try to connect from you mobile phone
be sure the ip is the correct one and use the IP to connect

prasad7676, 2017-03-19:
Thanks evry one i try mobile browser and work there. Thank you very much

closed 2017-03-19 by prasad7676