arendst/Tasmota #234: Migration issue from old Tasmoto project

stefanbode, 2017-03-19:
Hi gents,

great work. I installed the outdated version first and now with going to this version is crashes all the time when calling through the webinterface the CONFIGURATION -> CONFIGURE MODULE.

I pretty believe the is invalid data in the flash. I already tried to use the RESET CONFIGURATION but without success. Can someone give me a hint hot to overwrite the COMPLETE config data with the ones from the configuration file. Thanks

davidelang, 2017-03-19:
change CFG_VERSION in user_config.h or hold the button down for 4 seconds and 
the sonoff will do this for you

stefanbode, 2017-03-19:
Thanks. The CFG_VERSION I already checked without success. Also not the 4 second trick reset the device name to sonoff again, but the crash is still there at the http..../md website. I try to paste the stack into Arduino IDE to get information from the exception decoder, but for whatever reason in this configuration the error. No idea whats wrong here. 
ERROR: neither sonoff.ino.elf or sonoff.cpp.elf were found!
Did you forget to compile the sketch?

davidelang, 2017-03-19:
so are you seeing an error message on the serial console when it crashes?

It is possible to write any garbage to the device to overwrite the existing 
config (1M of any garbage) and then flash it again with the current firmware

stefanbode, 2017-03-19:
Ok, further investigation did show the following:
I'm using MQTT with TLS. If I have a valid servername and it can connect it crashes. If I change the servername of the MQTT server to something invalid it does not crash and I can access the configuration. There is no error message beside the crash dump available. Maybe you can reproduce with the one obove

arendst, 2017-03-19:
First update to the latest version (4.0.7) and then:

When it crashes what output do you get? I need exception and stack trace to be able to investigate.

stefanbode, 2017-03-19:
Here we go:
00:00:04 MQTT: Verify TLS fingerprint...
00:00:05 MQTT: Verified
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff-6945.local with IP address 192.168.178.133

Exception (28):
epc1=0x4000e1e0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff3780 end: 3fff3ba0 offset: 01a0

>>>stack>>>
3fff3920:  3fff1eec 0000000c 3fff3970 4021eef3  
3fff3930:  00000008 00001a5b 3fff77c4 00001a5b  
3fff3940:  00000004 3fff1e70 3fff3970 4021ef76  
3fff3950:  3fff1eec 3fff3a68 3fff3a68 00000021  
3fff3960:  3fff1e6f 3fff1e70 3fff3a68 40204826  
3fff3970:  3fff508c 0000000f 0000000c 3fff77c4  
3fff3980:  0000000f 00000004 00000000 00000021  
3fff3990:  00000012 3fff3a51 3fff3a68 40213e93  
3fff39a0:  3fff3a51 3fff9dc4 00000001 3fff9da4  
3fff39b0:  74706f3c 206e6f69 756c6176 33273d65  
3fff39c0:  333e2732 654c2032 3c493464 74706f2f  
3fff39d0:  3e6e6f69 706f2f00 6e6f6974 0000003e  
3fff39e0:  27343167 0000003e 00000ac6 4010020c  
3fff39f0:  00000000 3ffebf84 00000001 3fff2a88  
3fff3a00:  00000010 3fff3a90 00000000 4010053d  
3fff3a10:  402571ce 00000000 3fff3a90 402571e5  
3fff3a20:  3fff9b74 3fff3a90 3fff3a90 3ffeb940  
3fff3a30:  6f6e6f53 42206666 63697361 0d000000  
3fff3a40:  2c2c002c 00000000 11000000 00002c1d  
3fff3a50:  64654c00 49004934 00760000 3f00696e  
3fff3a60:  00000010 3fff3af0 00000000 00001a5f  
3fff3a70:  00001a5b 3fff3ad0 3fff3ad0 4021eea4  
3fff3a80:  3fff1efa 3fff3a3f 3fff3a68 4021eef3  
3fff3a90:  3fff2a88 00000650 00000650 4021a5ec  
3fff3aa0:  00000002 00000001 3fff9424 4021fdba  
3fff3ab0:  00000000 00000000 3fff9424 4021a5e2  
3fff3ac0:  3fff9424 3fff8b84 3fff9424 4021a61e  
3fff3ad0:  00000000 00000000 00000000 4021f054  
3fff3ae0:  3fff9424 3fff8b84 3fff8b44 4021a6b1  
3fff3af0:  3fff9b74 0000000f 00000003 40216e5c  
3fff3b00:  00000000 00000000 00000000 00000001  
3fff3b10:  00000002 40219880 0000000f 402176b9  
3fff3b20:  00000000 00000000 3fff8b44 3fff2b6c  
3fff3b30:  00000001 3fff8b68 3fff8b44 4021a83b  
3fff3b40:  3ffea368 00000000 000003e8 00000000  
3fff3b50:  00000000 3fff5064 4021faf0 3fff2b80  
3fff3b60:  3fffdad0 00000000 3fff2b64 402046ba  
3fff3b70:  00000000 00000000 00000001 40214f2a  
3fff3b80:  3fffdad0 00000000 3fff2b64 4021fb3c  
3fff3b90:  feefeffe feefeffe 3fff2b80 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
á
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback ESP_19FB21, GroupTopic sonoffs) Version 4.0.6
00:00:00 Wifi: Connecting to AP1 BodeNet_Keller in mode 11N as sonoff-6945...
00:00:05 MQTT: Verify TLS fingerprint...
00:00:05 MQTT: Verified
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff-6945.local with IP address 192.168.178.133

stefanbode, 2017-03-19:
Here is the decompiled stack trace. Got it working again:

0x4021f0c3: String::reserve(unsigned int) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4021f146: String::String(char const*) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40204846: showPage(String&) at E:\Tasmoto_Master\sonoff/support.ino line 460
0x40214057: handleModule() at E:\Tasmoto_Master\sonoff/webserver.ino line 568
0x4010020c: _umm_free at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010053d: _umm_realloc at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4025739e: pbuf_free at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 752
0x402573b5: pbuf_free at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 758
0x4021f074: String::changeBuffer(unsigned int) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4021f0c3: String::reserve(unsigned int) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4021a7bc: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4021ff8a: std::_Function_handler ::_M_invoke(std::_Any_data const&) at d:\users\stbode\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4021a7b2: std::function ::operator()() const at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4021a7ee: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4021f224: String::String(String const&) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4021a881: ESP8266WebServer::_handleRequest() at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x402176c8: SSLContext::_readAll() at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
0x40219a50: FunctionRequestHandler::canHandle(HTTPMethod, String) at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40217889: SSLContext::available() at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
0x4021aa0b: ESP8266WebServer::handleClient() at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4021fcc0: esp_yield at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402046da: pollDnsWeb() at E:\Tasmoto_Master\sonoff/support.ino line 460
0x402150f2: loop at E:\Tasmoto_Master\sonoff/sonoff.ino line 2211
0x4021fd0c: loop_wrapper at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at D:\Users\stbode\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

stefanbode, 2017-03-19:
The issue is the same with SONOFF base and the D1 mini that is connected right now. Here some memory information. I thought this might be the issue but does not look like:

Flash size	4096kB
Program flash size	4096kB
Program size	484kB
Free program space	536kB
Free memory	8kB

arendst, 2017-03-20:
I'm afraid it's a memory problem. I receive almost same responses albeit using a serial dump just before webserver takes over at that time there is only 4k left and the webpage still needs to be transferred.

I guess the combination of TLS, webserver and resources has just become to much...

I suggest you select either TLS or Webserver.

stefanbode, 2017-03-23:
Can we put some additional investigations on this topic? Other pages work fine. It is only the configuration page that causes the issue, and this page does not look more complicated than the other ones. I will check, which part of the configuration page causes the extensive memory requirements and maybe can provide a solution

arendst, 2017-03-23:
Please do. The problem is with the amount of drop down boxes; they all have the available options in the source which could end up to 18k which is eating up memory for preparing the web page.

stefanbode, 2017-03-23:
It should work with some JavaScript. Maybe the tradeoff does not work. But I agree the problem is the drop down list that gets added multiple times. My measurement is also 18.5k for the WeMo page. (Far too much). I have found this, that may make it: http://stackoverflow.com/questions/19671540/reusing-single-html-select-drop-down-list-for-multiple-table-rows

davidelang, 2017-03-27:
given the changes in recent releases, can this issue be closed now?

stefanbode, 2017-03-27:
Yes

closed 2017-04-02 by davidelang