arendst/Tasmota #235: firmware flash webUI v4.0x fails

kaaLabs15, 2017-03-19:
Hi,

Updating firmware from the WebUI v4.0x fails. Serial log shows:

`16:27:43 MQTT: stat/sonoff/RESULT = {"OtaUrl":"http://o"}
16:27:43 MQTT: stat/sonoff/RESULT = {"Upgrade":"Version 4.0.7 from http://o"}
16:27:45 HTTP: Webserver stopped
16:27:59 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:28:01 HTTP: Webserver stopped
16:28:14 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:28:16 HTTP: Webserver stopped
16:28:29 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211

16:28:31 HTTP: Webserver stopped
16:28:44 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:28:46 HTTP: Webserver stopped
16:28:59 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:29:01 HTTP: Webserver stopped
16:29:14 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:29:16 HTTP: Webserver stopped
16:29:29 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:29:31 HTTP: Webserver stopped
16:29:45 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:29:46 HTTP: Webserver stopped
16:30:00 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:30:01 HTTP: Webserver stopped
16:30:02 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
16:30:03 MQTT: stat/sonoff/UPGRADE = Failed HTTP error: connection refused

16:30:04 APP: Restarting
`
The OtaUrl shown in the log has been truncated somehow. The URL should be: 'http://odroid-server.local/firmware/sonoff/firmware.bin'. Issue can be replicated consequently. Changing the URL through the web interface does not solve the issue, as does not resetting the whole configuration either through menu or 4-sec pushing the button.

Can I try other scenario's to get this working? Any suggestion is welcome! thanks already

davidelang, 2017-03-19:
from the console, what does otaurl return?

if you set the value via the otaurl command, does it return the correct value 
after that?

upgrade 1 will tell it to upgrade from the console.

I've done several ota upgrades with the 4.x firmware without a problem. I have 
the url hard-coded in user_config.h and have done it both from the web page and 
the web console

arendst, 2017-03-19:
Already found the error. My fault, too much cleaned up. Solved in next release. Workaround: change line 966 in webserver.ino from svalue[16] to svalue[100];

kaaLabs15, 2017-03-19:
It was indeed hardcoded in the userconfig.h and through MQTT refreshing the Otaurl and upgrading went just fine. Just as soon as I proceed upgrade through the web-interface the Otaurl gets messed up I.

kaaLabs15, 2017-03-19:
See you found the problem. Thanks!

kaaLabs15, 2017-03-19:
Will it be possible to upgrade OtA to the new release given the regression?

arendst, 2017-03-19:
Yes. You can also upgrade using the second option on the web page by uploading the file yourself.

davidelang, 2017-03-19:
yes but not through the web page, you would have to do it from the web console 
or via a mqtt command.

did you have to set the otaurl, or just do the upgrade command?

kaaLabs15, 2017-03-19:
Will make the adjustment and try the upload options.

kaaLabs15, 2017-03-19:
Made the code-adjustment and flashed new firmware through FTDI.

Then, re-flash through MQTT still faills (tested, OtaUrl is correct and accessible):

'17:11:07 MQTT: stat/sonoff-0946/RESULT = {"OtaUrl":"http://odroid-server.local:80/firmware/sonoff/firmware.bin"}
17:11:07 MQTT: stat/sonoff-0946/RESULT = {"Upgrade":"Version 4.0.6p from http://odroid-server.local:80/firmware/sonoff/firmware.bin"}
17:11:10 HTTP: Webserver stopped
17:11:24 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:11:26 HTTP: Webserver stopped
17:11:39 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:11:41 HTTP: Webserver stopped
17:11:54 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:11:56 HTTP: Webserver stopped
17:12:09 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:12:11 HTTP: Webserver stopped
17:12:24 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:12:26 HTTP: Webserver stopped

17:12:39 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:12:41 HTTP: Webserver stopped
17:12:54 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:12:56 HTTP: Webserver stopped
17:13:10 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:13:11 HTTP: Webserver stopped
17:13:25 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:13:26 HTTP: Webserver stopped
17:13:27 HTTP: Webserver active on sonoff-0946-0946.local with IP address 192.168.1.211
17:13:28 MQTT: stat/sonoff-0946/UPGRADE = Failed HTTP error: connection refused
17:13:29 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset'

arendst, 2017-03-19:
Right. Another problem as the url is now fine.

Does the ota upgrade work if you initiate it form MQTT with command upgrade 1?

kaaLabs15, 2017-03-19:
It doesn't react anymore on `upgrade 1` through MQTT...

arendst, 2017-03-19:
What happens if you reset the device (power cycle). Does it connect to your broker?

kaaLabs15, 2017-03-19:
Yes it does perfectly:

```
00:00:09 Wifi: Connected
00:00:09 mDNS: Initialized
00:00:09 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:33:35 MQTT: Attempting connection...
17:33:36 mDNS: Query done with 0 mqtt services found
17:33:36 MQTT: Connected
17:33:36 MQTT: tele/sonoff/LWT = Online (retained)
17:33:36 MQTT: cmnd/sonoff/POWER =
17:33:36 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"4.0.6p", "FallbackTopic":"DVES_1BE3B2", "GroupTopic":"sonoffs"}
17:33:36 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-0946", "IPaddress":"192.168.1.211"}
17:33:36 MQTT: tele/sonoff/INFO3 = {"Started":"Power on"}
17:33:37 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
17:33:37 MQTT: stat/sonoff/POWER = ON
```

arendst, 2017-03-19:
And now try command otaurl first and chack it's output.

Then command upgrade 1

kaaLabs15, 2017-03-19:
MQTT or Console?

arendst, 2017-03-19:
Doesn't matter as long as you do not use the web upgrade page

kaaLabs15, 2017-03-19:
This is the output. No luck, I'm afraid:

```
17:41:18 CMND: otaurl
17:41:18 MQTT: stat/sonoff/RESULT = {"OtaUrl":"http://odroid-server.local/firmware/sonoff/firmware.bin"}
```

```
17:42:06 CMND: upgrade 1
17:42:06 MQTT: stat/sonoff/RESULT = {"Upgrade":"Version 4.0.6p from http://odroid-server.local/firmware/sonoff/firmware.bin"}
17:42:09 HTTP: Webserver stopped
17:42:22 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:42:24 HTTP: Webserver stopped
17:42:37 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:42:39 HTTP: Webserver stopped
17:42:52 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:42:54 HTTP: Webserver stopped
17:43:07 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:43:09 HTTP: Webserver stopped
17:43:23 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:43:24 HTTP: Webserver stopped
17:43:38 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:43:39 HTTP: Webserver stopped
17:43:53 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:43:54 HTTP: Webserver stopped

17:44:08 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:44:09 HTTP: Webserver stopped
17:44:23 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:44:24 HTTP: Webserver stopped
17:44:25 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:44:26 MQTT: stat/sonoff/UPGRADE = Failed HTTP error: connection refused
17:44:28 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_1BE3B2, GroupTopic sonoffs) Version 4.0.6p

00:00:00 Wifi: Connecting to AP1 MTB06WMESH in mode 11N as sonoff-0946...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on sonoff-0946.local with IP address 192.168.1.211
17:44:34 MQTT: Attempting connection...
17:44:35 mDNS: Query done with 0 mqtt services found
17:44:35 MQTT: Connected
17:44:35 MQTT: tele/sonoff/LWT = Online (retained)
17:44:35 MQTT: cmnd/sonoff/POWER =
17:44:35 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"4.0.6p", "FallbackTopic":"DVES_1BE3B2", "GroupTopic":"sonoffs"}
17:44:35 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-0946", "IPaddress":"192.168.1.211"}
17:44:35 MQTT: tele/sonoff/INFO3 = {"Started":"Software/System restart"}
17:44:37 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
17:44:37 MQTT: stat/sonoff/POWER = ON
```

arendst, 2017-03-19:
Great!. So you have a problem with your URL. Try again replacing the odroid-server.local name with it's ip address.

kaaLabs15, 2017-03-19:
You're right! That did solve it. Don't know why, I did verify the url through my browser, but it did.

Thanks!

closed 2017-03-19 by arendst