arendst/Tasmota #239: NTP timestamp missing in version 4.0.7

decologne, 2017-03-20:
After upgrade to version 4.0.6-7 timestamp based on NTP is missing.

arendst, 2017-03-20:
Check your NTP servers with commands ntpservers1 to ntpservers3 and make sure there is internet connection.

decologne, 2017-03-20:
If I load version 4.0.1 NTP is working.
The output of version 4.0.7 is
00:30:40 CMND: ntpservers1
00:30:40 RSLT: DataCb Topic sonoff_19495A, Group 0, Index 1, Type NTPSERVERS, Data  ()
00:30:40 MQTT: stat/sonoff_19495A/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
00:30:40 MQTT: stat/sonoff_19495A/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
00:30:40 MQTT: stat/sonoff_19495A/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation"}
00:30:48 MQTT: tele/sonoff_19495A/STATE = {"Time":"1970-01-01T00:30:48", "Uptime":1, "Vcc":2.744, "Wifi":{"AP":1, "SSID":"adoca", "RSSI":50}}
00:30:48 DHT: Received 2C, 00, 18, 00, 44 =? 44
00:30:48 MQTT: tele/sonoff_19495A/SENSOR = {"Time":"1970-01-01T00:30:48", "DHT11":{"Temperature":24.0, "Humidity":44.0}}
00:31:19 CMND: ntpservers2
00:31:19 RSLT: DataCb Topic sonoff_19495A, Group 0, Index 2, Type NTPSERVERS, Data  ()
00:31:19 MQTT: stat/sonoff_19495A/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
00:31:19 MQTT: stat/sonoff_19495A/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
00:31:19 MQTT: stat/sonoff_19495A/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation"}
00:31:31 CMND: ntpservers3
00:31:31 RSLT: DataCb Topic sonoff_19495A, Group 0, Index 3, Type NTPSERVERS, Data  ()
00:31:31 MQTT: stat/sonoff_19495A/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
00:31:31 MQTT: stat/sonoff_19495A/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
00:31:31 MQTT: stat/sonoff_19495A/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation"}

arendst, 2017-03-20:
And in the commands list you see that the correct command is ntpserver without the s. 

Starting with version 4.0.4 ntpservers are configurable so it makes sense that your 4.0.1 still works. Somewhere along the line your ntpservers get lost.

Please try again with the correct commands ntpserver1 to ntpserver3

decologne, 2017-03-20:
Sorry, but you ask to run the commands with s.
Now the webserver console output is (after a restart)
00:00:50 CMND: ntpserver1
00:00:50 RSLT: DataCb Topic sonoff_19495A, Group 0, Index 1, Type NTPSERVER, Data  ()
00:00:50 MQTT: stat/sonoff_19495A/RESULT = {"NTPServer1":""}
After this command the output is frozen.
Using the usb connection of the device the output is:
00:01:06 CMND: ntpserver1
00:01:06 RSLT: Receive topic cmnd/sonoff_19495A/ntpserver1, data size 0, data 
00:01:06 RSLT: DataCb Topic sonoff_19495A, Group 0, Index 1, Type NTPSERVER, Data  ()
00:01:06 MQTT: stat/sonoff_19495A/RESULT = {"NTPServer1":""}
00:01:07 Wifi: Checking connection...
00:01:07 Wifi: Connected
00:01:15 CMND: ntpserver2
00:01:15 RSLT: Receive topic cmnd/sonoff_19495A/ntpserver2, data size 0, data 
00:01:15 RSLT: DataCb Topic sonoff_19495A, Group 0, Index 2, Type NTPSERVER, Data  ()
00:01:15 MQTT: stat/sonoff_19495A/RESULT = {"NTPServer2":"ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ"}
00:01:23 CMND: ntpserver3
00:01:23 RSLT: Receive topic cmnd/sonoff_19495A/ntpserver3, data size 0, data 
00:01:23 RSLT: DataCb Topic sonoff_19495A, Group 0, Index 3, Type NTPSERVER, Data  ()
00:01:23 MQTT: stat/sonoff_19495A/RESULT = {"NTPServer3":"ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ"}
00:01:28 Wifi: Checking connection...
00:01:28 Wifi: Connected

arendst, 2017-03-20:
So for some reason the conversion to user configurable ntpserver failed.

Solution: just enter the three ntpservers using the command `ntpserver1` to `ntpserver3` with your local ntpservers or use the ones in user_config.h

decologne, 2017-03-20:
After the commands:
ntpserver1 pool.ntp.org
etc.
the NTP timestap is now correct.
I did not realise that I change the values before.
Thanks Theo for help.

closed 2017-03-20 by arendst