arendst/Tasmota #240: MQTT crashed with TLS in the "Configuration" Menue

Reinhartd, 2017-03-20:
Hello,
when i tested the TLS function (CloudMQTT) on a Sonoff Dev Modul, i can not call the "Configuration" Menue, but the TLS function is ok. I have the CloudMQTT switched as Bridge and i can receive the MQTT strings on my local Broker.

The Modul crashed with an Exeption:
`Exception (28):
epc1=0x4000e1e0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff3410 end: 3fff3830 offset: 01a0

>>>stack>>>
3fff35b0:  3fff1cec 0000000a 3fff3600 40218ba7  
3fff35c0:  00000006 0000238f 3fff4be4 0000238f  
3fff35d0:  00000004 3fff1c70 3fff3600 40218c2a  
3fff35e0:  3fff1cec 3fff36f8 3fff36f8 00000026  
3fff35f0:  3fff1c6d 3fff1c70 3fff36f8 40203f9a  
3fff3600:  3fff88fc 0000000f 0000000a 3fff4be4  
3fff3610:  0000000f 00000004 00000000 00000026  
3fff3620:  00000012 3fff36e1 3fff36f8 4020fcff  
3fff3630:  3fff36e1 ffffffff 04a1c3a1 00000000  
3fff3640:  74706f3c 206e6f69 756c6176 33273d65  
3fff3650:  333e2737 57502037 2f3c354d 6974706f  
3fff3660:  003e6e6f 706f2f00 6e6f6974 003e003e  
3fff3670:  27343167 0000003e 00000150 4010020c  
3fff3680:  00000000 3ffebc88 00000001 3fff2724  
3fff3690:  00000010 3fff3720 00000000 4010053d  
3fff36a0:  402509fe 00000000 3fff3720 40250a15  
3fff36b0:  3fff4be4 3fff3720 3fff3720 3ffeb644  
3fff36c0:  6f6e6f53 44206666 00007665 0d000000  
3fff36d0:  2c2c002c 0000002c 2c000000 00002c1d  
3fff36e0:  4d57502b 49000035 00760000 3f00696e  
3fff36f0:  00000010 3fff3780 00000000 0000238f  
3fff3700:  0000238f 3fff3760 3fff3760 40218b58  
3fff3710:  3fff1cfa 3fff36cf 3fff36f8 40218ba7  
3fff3720:  3fff2724 00000150 00000150 40216060  
3fff3730:  00000002 00000001 3fff8264 40219a72  
3fff3740:  00000000 00000000 3fff8264 40216056  
3fff3750:  3fff8264 3fff895c 3fff8264 40216092  
3fff3760:  00000000 00000000 00000000 40218d08  
3fff3770:  3fff8264 3fff895c 3fff891c 40216125  
3fff3780:  3fff4b5c 0000000f 00000003 40212f6c  
3fff3790:  00000000 00000000 00000000 00000001  
3fff37a0:  00000002 402152f4 0000000d 4021312d  
3fff37b0:  00000000 00000000 3fff1d3c 3fff2808  
3fff37c0:  00000001 3fff8940 3fff891c 402162af  
3fff37d0:  3ffea130 00000000 3fff2810 00000000  
3fff37e0:  3fffdad0 3fff2808 402197a8 3fff2810  
3fff37f0:  3fffdad0 00000000 3fff2800 40203e2e  
3fff3800:  00000000 00000000 00000001 40210d9e  
3fff3810:  3fffdad0 00000000 3fff2800 402197f4  
3fff3820:  feefeffe feefeffe 3fff2810 40100718  
<<<stack<<<`

Is this a problem with the Memory?

arendst, 2017-03-20:
#234

arendst, 2017-03-20:
I'm afraid it's a memory problem indeed. I receive almost same responses albeit using a serial dump just before webserver takes over at that time there is only 4k left and the webpage still needs to be transferred.

I guess the combination of TLS, webserver and resources has just become to much...

Reinhartd, 2017-03-20:
I have now solved it so that I do the configuration in the console.

`Module 19
Gpio5 3
Gpio14 4` 

But I need the web console.

LG

davidelang, 2017-03-21:
you should be able to issue those commands via mqtt as well

davidelang, 2017-03-27:
with the improvements in recent versions, can this be considered fixed now?

davidelang, 2017-04-02:
closing this due to no response, please reopen if you still have problems.

closed 2017-04-02 by davidelang